kitesurfer1404/WS2812FX #221: Set color

blastdotexe, 2020-05-07:
Hi, do you have a way to change native colors of ws2812fx effects? 
Exemple: I want red (or an other color) running on blue (...) instead one color running on white.
Thanks (I'm French)

moose4lord, 2020-05-07:
You can pass an array of up to three colors to the setSegment() function.
```c++
uint32_t colors[] = { RED, GREEN, BLUE };
ws2812fx.setSegment(0, 0, LED_COUNT-1, FX_MODE_COLOR_WIPE, colors, 1000, NO_OPTIONS);
```
Not all of the effects use all three colors. Some use one, some use two and a few use all three.

blastdotexe, 2020-05-08:
Thanks for your answers, but I want to change de bacground color of FX_MODE_RUNNING_LIGHT. Is it possible ?

moose4lord, 2020-05-08:
Sorry, the current FX_MODE_RUNNING_LIGHTS effect is written to only use one color.

However, I tweaked the mode_running_lights() function in WS2812FX.cpp to allow for two colors.  It will do what you want, if you replace the code with this:
```c++
uint16_t WS2812FX::mode_running_lights(void) {
  uint8_t size = 1 << SIZE_OPTION;
  uint8_t sineIncr = max(1, (256 / SEGMENT_LENGTH) * size);
  for(uint16_t i=0; i < SEGMENT_LENGTH; i++) {
    int lum = (int)sine8(((i + SEGMENT_RUNTIME.counter_mode_step) * sineIncr));
    uint32_t color = color_blend(SEGMENT.colors[0], SEGMENT.colors[1], lum);
    if(IS_REVERSE) {
      setPixelColor(SEGMENT.start + i, color);
    } else {
      setPixelColor(SEGMENT.stop - i,  color);
    }
  }
  SEGMENT_RUNTIME.counter_mode_step = (SEGMENT_RUNTIME.counter_mode_step + 1) % 256;
  return (SEGMENT.speed / SEGMENT_LENGTH);
}
```
Thanks for the good idea. I will make this change in the WS2812FX library the next time I do a release.

blastdotexe, 2020-05-10:
Thanks a lot!!

closed 2020-05-10 by blastdotexe