kitesurfer1404/WS2812FX #223: LED flickering with random color

janhls, 2020-05-21:
Hey there,

i have a strange problem with the WS2812FX, the LEDs are flickering with random color. I am using a ESP32 and LED Strip WS2812B. Does anyone know what is wrong?

Video: [https://photos.app.goo.gl/dvYWKH25gqZrmHNW8](url)

Code:
`#include <WS2812FX.h>

#define LED_COUNT 56
#define LED_PIN 12

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
    ws2812fx.init();
    ws2812fx.setColor(255,0,0);
    ws2812fx.setMode(2);
    ws2812fx.setSpeed(1000);
    ws2812fx.start();
}

void loop() {
    ws2812fx.service();
}`

moose4lord, 2020-05-21:
The ESP32 is notoriously picky when it comes to WS2812 LEDs. You'd think that with two cores, one for your sketch and one to manage WiFi, it would be more robust than an ESP8266, but not so (at least in my experience). When developing LED sketches on the ESP32, I've never been able to pin a task to one core and have it work properly. Even without monkeying with pinning tasks to cores, you get the weird LED flashing issue as discussed in issue #177.

I would recommend you start with the ws2812fx_esp32 example sketch, which uses the ESP32's RMT hardware (instead of bit-banging) to fix the weird flashing LED issue.

janhls, 2020-05-21:
Thx, it is working now

moose4lord, 2020-05-21:
I'm going to leave this issue open for a while, since this issue regarding the ESP32 comes up pretty regularly.

mriksman, 2020-06-08:
I'm using ESP-IDF, and pinning the task that runs ws2812fx.service() to Core 1 fixed my issue.

moose4lord, 2020-06-08:
I'd be interested in seeing your code, if you don't mind posting it.

mriksman, 2020-06-09:
I'm using ESP-IDF, so I'm not sure if it helps. But I create a task with;
```
xTaskCreatePinnedToCore(&animation_task, "anim", 2048, NULL, 10, NULL, 1);
```
And then within that task, I just call `service()` on a loop with a small task delay to allow the IDLE task to run.
```
    while(1) {
        ws2812fx.service();
        vTaskDelay(5);
```

silardg, 2021-07-30:
This is still a problem. If you have multiple things going on (multiple xtaskcreates) it wont fix it if you have it pinned.

Buddhika93n, 2022-12-13:
Dear all,
I am experiencing the same issue..In my code I have 5 tasks. one of them is main_task... when I include main_task to the code led flickering issue occurs...

 xTaskCreate(&blink_led_task, "blink_led_task", 1024 * 10, NULL, 5, NULL);

I am using above function to create the task...

 for (j = 0; j < (CONFIG_EXAMPLE_STRIP_LED_NUMBER - 4); j++)
    {
        led_strip_hsv2rgb(hue, saturation, (brightness / 20), &red, &green, &blue);
        // Write RGB values to strip driver
        ESP_ERROR_CHECK(strip->set_pixel(strip, j + 4, red, green, blue));
        led_strip_hsv2rgb(hue, saturation, (brightness / 16), &red, &green, &blue);
        // Write RGB values to strip driver
        ESP_ERROR_CHECK(strip->set_pixel(strip, j + 3, red, green, blue));
        led_strip_hsv2rgb(hue, saturation, (brightness / 8), &red, &green, &blue);
        // Write RGB values to strip driver
        ESP_ERROR_CHECK(strip->set_pixel(strip, j + 2, red, green, blue));
        led_strip_hsv2rgb(hue, saturation, (brightness / 4), &red, &green, &blue);
        // Write RGB values to strip driver
        ESP_ERROR_CHECK(strip->set_pixel(strip, (j + 1), red, green, blue));
        led_strip_hsv2rgb(hue, saturation, (brightness), &red, &green, &blue);
        // Write RGB values to strip driver
        ESP_ERROR_CHECK(strip->set_pixel(strip, (j), red, green, blue));
        ESP_ERROR_CHECK(strip->refresh(strip, 100));
        vTaskDelay(delay/portTICK_RATE_MS);
        // ESP_LOGI(TAG, "j:%d", j);
    }

when above code executes...led flickering issue with random colors occurs...

moose4lord, 2022-12-15:
Hmmmm...it doesn't look like you're using the WS2812FX lib. You might have better luck posting your question to a forum that deals with ESP32 RMT issues in general.

Buddhika93n, 2022-12-15:
ok thanks for the guidance...

moose4lord, 2025-01-21:
The ESP32 RMT driver is in common use today, so closing this issue.

closed 2025-01-21 by moose4lord