kitesurfer1404/WS2812FX #224: Way to stop

Devilscave, 2020-05-22:
I have a question about "Loop stop", but I am not talking about the loop of the µC but about the continuous repetition of the color change. In my example I'll take the "Comet_Mode". Is there a way to let it go through only once when an array is triggered, so that it is not permanent? As an example, I have an LED strip with 20 leds when an action is triggered, whether by pressing a button or something else that only runs a "Comet" once. Is there a function for, or do I have to come up with this myself? Otherwise it might be useful for people to insert such a function for all effects.

moose4lord, 2020-05-22:
You can get feedback about how an effect is progressing by using the isFrame() and isCycle() functions. See issue #199.

Devilscave, 2020-05-22:
I have continued to write on it since I opened the issue's and have now come up with a solution, I have linked a stop to the interval of the strip speed and declared this using integers, now I have exactly the effect I wanted.