kitesurfer1404/WS2812FX #227: Latest Update reset/restart issue

henmel, 2020-05-24:
I am running into an issue with the latest update of the WS2812FX Library(1.2.4), along with the latest Arduino ESP8266 board library(2.7.1) update. I have included 
[esp8266_webinterface_testota-200524a.zip](https://github.com/kitesurfer1404/WS2812FX/files/4674070/esp8266_webinterface_testota-200524a.zip)
a copy of Web Interface example with Arduino OTA added. When the sketch is uploaded via Arduino OTA, my NodeMCU ESP8266 continuously keeps resetting with:
```
12:14:02.732 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
12:14:02.732 -> 
12:14:02.732 -> load 0x4010f000, len 3456, room 16 
12:14:02.732 -> tail 0
12:14:02.732 -> chksum 0x84
12:14:02.732 -> csum 0x84
12:14:02.732 -> va5432625
12:14:02.732 -> ~ld

```
until I press the reset button on the NodeMCU, then it restarts normally.  I also added a restart function to the web interface for testing, which appears to do the same thing as uploading over OTA. After doing a lot of testing, trying to narrow down the cause, it appears that the latest changes in the mode_running_lights, may have something to do with it, as changing the code to the old version of the effect resolves the issue.

Thanks in advance!

moose4lord, 2020-05-25:
Your sketch works fine on my NodeMCU. Looks like you're a victim of this XMC flash bug:
[https://github.com/esp8266/Arduino/issues/7307](https://github.com/esp8266/Arduino/issues/7307)

henmel, 2020-05-25:
Thank you for the quick response and reference links, but I am a bit confused about the fix. How do I install the Pull Request(PR)? Do I just uninstall then reinstall esp8266 board, using the board manager? I am sorry to be a bother, but this is still a bit new to me.

moose4lord, 2020-05-25:
Unless you have a burning need for the 2.7.1 release, I would drop back to the 2.6.3 release. I would guess that the PR would go out in the 2.7.2 release, but not sure when that will be.

henmel, 2020-05-26:
Thanks again for your quick help and a great library!

closed 2020-05-26 by henmel