kitesurfer1404/WS2812FX #230: Segments in Auto Cycle Example

sprewl, 2020-06-04:
Hi, I'm sorry to put this as an issue but I'm not able to use Segments inside the Setup Section of the code here's the code, The strip just stays off (Using Auto Cycle Example Sketch)

`#include <WS2812FX.h>

#define LED_COUNT 10
#define LED_PIN 5

#define TIMER_MS 10000

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_RGB + NEO_KHZ800);

unsigned long last_change = 0;
unsigned long now = 0;

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(70);
  ws2812fx.setSpeed(1000);
  ws2812fx.setColor(0x007BFF);
  ws2812fx.setMode(FX_MODE_STATIC);
  ws2812fx.start();
  ws2812fx.setSegment(0,  11,  19, FX_MODE_MULTI_STROBE, 0xFF0000, 1000, false);
}

void loop() {
  now = millis();
 
  ws2812fx.service();

  if(now - last_change > TIMER_MS) {
    ws2812fx.setMode((ws2812fx.getMode() + 1) % ws2812fx.getModeCount());
    last_change = now;
  }
}`

moose4lord, 2020-06-04:
You've configured your strip to have 10 LEDs (```#define LED_COUNT 10```), but you've configured the segment to use LEDs 11-19 (```ws2812fx.setSegment(0, 11, 19,...```). Change your setSegment statement to use LEDs 0 - 9 and you should be good to go.

P.S. the setSegment() function is meant to replace the setColor(), setMode() and setSpeed() functions, so you don't need both. SetSegment() sets those three parameters all at once, instead of one at a time.

sprewl, 2020-06-04:
Thanks a bunch got it working ! makes sense!

closed 2020-06-04 by sprewl