kitesurfer1404/WS2812FX #231: Fade Up/Down?

mriksman, 2020-06-08:
Hey,

I can't work out how to do a simple 'fade to colour'. I don't need it to fade in and out (`FX_MODE_FADE`). And I can't work out how to use `fade_out()`

Thoughts?

moose4lord, 2020-06-08:
WS2812FX isn't really geared toward creating custom effects. It's more about choosing from the built-in effects to create your lighting project. That said, you can create a custom effect if you have a good understanding of how to manipulate the LED data. It can get quite complicated though. If you tell me more about your project, I might be able to help.

If creating your own custom effect is what you're after, you might want to look at the [FastLED](https://github.com/FastLED/FastLED) library. It's got a ton of utility functions that can manipulate LED data to create lots of different effects.

mriksman, 2020-06-09:
Thanks for the reply. 
The animations are great, but I also need it to handle basic, static colours. Think; Apple HomeKit and setting a light’s brightness and colour. `FX_MODE_STATIC` does that, but I’d like it to have a subtle ‘fade to’ instead of a hard change. 

I’m pretty well versed with NeoPixelBus (which is actually what I have WS2812FX using to manipulate the pixel data using a custom Adafruit_NeoPixel.h), so maybe I’ll stick with that and migrate some of WS2812FX’s animations over. 
See https://github.com/mriksman/WS2812FX-idf

closed 2021-03-01 by moose4lord