kitesurfer1404/WS2812FX #233: Animation cycle

craigrobbo, 2020-06-17:
Hi again

Sorry I know you proberbly shudder when you see me crop up

Im making greathead way with my code.

In the CPP file I am wondering if its possible to make an edit so that effects like Color wipe do not continue their cycle but instead would just complete one cycle only?

For example with color wipe all LED would light up one after the other, then they would turn off one after the other...then would just stay off?

Thank you :)

moose4lord, 2020-06-21:
You're looking for the isCycle() function. It returns "true" when an animation has completed one cycle. It is discussed in issue #199.

Something like this
```c++
void loop() {
  ws2812fx.service();

  if(ws2812fx.isCycle()) { // test if animation cycle is complete
    ws2812fx.stop();
  }
}
```

craigrobbo, 2020-06-21:
> 
> 
> You're looking for the isCycle() function. It returns "true" when an animation has completed one cycle. It is discussed in issue #199.
> 
> Something like this
> 
> ```c++
> void loop() {
>   ws2812fx.service();
> 
>   if(ws2812fx.isCycle()) { // test if animation cycle is complete
>     ws2812fx.stop();
>   }
> }
> ```

Thanks Keith.

I was thinking more is it possible to permenantly edit the Library CPP file so that the Color wipe animation never  repeats and/or so it only compleates half of its cycle?

moose4lord, 2020-06-21:
It's possible, but that's not something I want to get into. Sorry.

craigrobbo, 2020-06-21:
no worries mate thank you :)

closed 2020-08-08 by moose4lord