kitesurfer1404/WS2812FX #234: Latest Library Version Issue

henmel, 2020-06-20:
I recently updated the library from 1.2.4 to 1.3.1 and am running into an issue related getting and using the Segment variables. I am using a customized version of esp8266_webinterface, which includes an On/Off function.  If the "ws2812fx.start()" is not executed in the void setup() function(start in the Off position), with the new library, when I try to get segment information using:
```
    WS2812FX::Segment* seg = ws2812fx.getSegment(); // get the current segment
    int seglen = seg->stop - seg->start + 1;
    Serial.print("seg->mode: "); Serial.println(ws2812fx.getModeName(seg->mode));
    Serial.print("seg->speed: "); Serial.println(seg->speed);

```
I get soft reset. This did not happen with the older library version. I have setup and attached a simplified version of my code, to illustrate the issue. If I compile the code with v1.2.4, no soft reset occurs, when the above code is executed valid values are displayed for mode and speed, but when compiled with v1.3.1, I get soft reset when the above code is executed.

I did notice if I execute the following code first, before "ws2812fx.stop();", no soft reset occurs with the new library:

```
  ws2812fx.start();
  ws2812fx.service();

```

[esp8266_webinterface_newlibvtest-200620a.zip](https://github.com/kitesurfer1404/WS2812FX/files/4808624/esp8266_webinterface_newlibvtest-200620a.zip)

Thanks for a Great Library!!!

moose4lord, 2020-06-21:
Your use of getSegment() is incorrect. That function "gets" the segment currently being processed by the service() routine. It's really only useful when used in code for a custom effect.

You can use getSegment(0) to explicitly get the first segment's config, which should give you what you want.

Your use of getSegment() used to work in previous versions of the lib, just by happenstance. The old getSegment() code used an index parameter to access the segment data, and the index parameter happened to be initialized to zero. That is no longer the case.

You're right, if you allow the service() routine to run at least once (by clicking your "Power On" button or by executing service() in the setup() function) before the "Checking WiFi... " code runs, then it will work. But that's a bad hack. Unless you're writing a custom effect, stay away from getSegment(). Use getSegment(0) instead.

henmel, 2020-06-21:
Thank You for a quick response and the explanation.  I guess the use of "WS2812FX::segment* segments = ws2812fx.getSegments();" will still work as before, when getting multiple segments, since the array will return the segment counter?

moose4lord, 2020-06-21:
Yes, getSegments() works the same as before. It returns the entire array of segment data. You would just pick and choose the array elements you're interested in. In your code you could just as easily do this
```c++
WS2812FX::segment* segments = ws2812fx.getSegments();
Serial.print("seg->mode: "); Serial.println(ws2812fx.getModeName(segments[0].mode));
```
I'm not sure what you mean by "return the segment counter". There is no segment counter anymore (what used to be called _segment_index in the old code). The segment code was refactored to use pointers instead of array indexes.

henmel, 2020-06-21:
Thanks again! Really appreciate all of your help and quick responses.

Stay Safe,

closed 2020-06-21 by henmel

henmel, 2020-06-23:
Quick Question...

Tried to setup and test the refactored Matrix effect, using the ws2812fx_custom_effect2 example, but get an error when compiling:

"
Arduino\libraries\WS2812FX\src/custom/Matrix.h:73:18: error: invalid conversion from 'uint32_t* {aka unsigned int*}' to 'int32_t* {aka int*}' [-fpermissive]

   _matrix.colors = colors;
"
I tried to change int32_t* colors to uint32_t* colors, it compiled, but strip is blank.