kitesurfer1404/WS2812FX #235: Multiple segments and multiple strips running at the same time ...

chrisgimson, 2020-06-21:
I have an ESP8266 setup running (successfully) with two strips and up to 32 segments per strip potentially running different colours, animations and options.  I wish to be able to further monitor/modify animations "mid animation" for extra effect.  Eg. simply stop COMET or COLOR_WIPE after a number of  sweeps.  I see solutions from similar questions but they all seem to be limited to one segment and/or strip.  I do not see a way to be able to direct select which strip and segment to address to be able to use the functions such as isFrame, isCycle and others.  I would appreciate any advice you can give me.

moose4lord, 2020-06-21:
You pass the segment number you're interested in to the isFrame() and isCycle() functions.

```c++
boolean WS2812FX::isFrame(uint8_t seg);
boolean WS2812FX::isCycle(uint8_t seg);
```
They are member functions, so are associated with a particular strip when you call the function.
```c++
ws2812fx.isCycle(0)
```

chrisgimson, 2020-06-21:
Brilliant.  Many thanks. So just for clarity on my part.   int x = strip1.isFrame(5) would give me how many completed frames segment 5 of strip 1 had completed?

moose4lord, 2020-06-21:
isFrame() returns a boolean. "true" if the animation has advanced to the next frame, "false" otherwise. So you'll have to do the counting yourself. Something like this:
```c++
void loop() {
  static int frameCount = 0;

  ws2812fx.service();

  if(ws2812fx.isFrame(5)) { // new frame for segment 5?
    frameCount++;
    if(frameCount == 100) { // time to do something special
      ...
      frameCount = 0; // reset the frame counter
    }
  }
}
```

chrisgimson, 2020-06-21:
Strange.  This is a simple test that I have added to loop ...
static byte Frame;   // ????
   if(LED_StripLeft.isFrame(0) == true) 
   {
    Serial.println(Frame); 
    Frame ++; 
    if (Frame > 10) Frame = 0;  
   } 

else where in the loop I have a function that sets segment 0 to a STATIC animation of one colour on pressing a button and then  STATIC to another colour on button release - this works fine but my isFrame prints 0 to 10 repeatedly per loop pass  suggesting that isFrame is always returning true regardless of the button/segment state.  I was expecting Frame to increment every time I pressed and released the button.  I would appeciate your insight in what I am missing here?  (I also tries COMET when pressed and STATIC when released with exactly the same result.

moose4lord, 2020-06-22:
An animation frame is when the library pushes new data out to the LED strip based on the effect that's running. Take the BLINK effect for example. All LEDs on is one frame, and all LEDs off is a second frame. So if you wanted to change the color after 5 blinks, you'd do something like this:
```c++
void loop() {
  static int frameCount = 0;

  ws2812fx.service();

  if(ws2812fx.isFrame(0)) { // new frame for segment 0?
    frameCount++;
    if(frameCount == 10) { // 10 frames == 5 blinks
      uint32_t color = ws2812fx.color_wheel(ws2812fx.random8()); // get a random color
      ws2812fx.setColor(0, color); // change segment 0's color
      frameCount = 0; // reset the frame counter
    }
  }
}
```
Setting isFrame()==true when a "new frame" is drawn is strictly a function of the effect that's being run. It really has nothing to do with updating the segment color when your button state changes.

It's probably easier if your use the BLINK animation during your debug, since it's easy to count BLINK frames.

Hope that helps.

chrisgimson, 2020-06-22:
Thanks that helped considerably and I managed to get a test case that allowed me to see how it functions.  Is there a document anywhere that outlines the dependency of isFrame on each animation?

I stop an dynamic animation by writing a STATIC mode to it with a particular colour.  This works fine for my application but it does always leave isFrame() for those particular segments as true.  Is there a way to force reset isFrame() to false.  My workaround would be to only examine isFrame() after my target animation is started and during its run time but it would be "nice" if I could reset it in code.
Many thanks for your support - it is much appreciated

moose4lord, 2020-06-22:
Unfortunately, there's very little documentation about the library's inner workings. Sorry. You can peruse the source code if you really need that kind of detail.

I think you misunderstand how isFrame() is working. It is never just always true. Keep in mind that your ESP8266 is super fast and the LEDs are updated very slowly by comparison. The loop() function is iterated several dozen (or hundreds) of times while the LEDs are waiting to be updated to the next frame. Take BLINK for example. If you're blinking the LEDs every second, then loop() is being executed hundreds of times between each blink. For the vast majority of the time, the service() routine is not updating the LEDs and isFrame() returns false. Every second the LEDs are updated and isFrame() returns true.

The STATIC mode is a bit of a special case. Even though it presents a static color, in reality it does periodically refresh the LEDs with that color. Specifically it refreshes the LEDs based on the segment's speed setting. If you set speed to 1000, then the LEDs get refreshed every 1000ms, and isFrame() would return true every 1000ms. I get what you're saying, that's arguably incorrect behavior. A purist would say the LEDs should be set once and then left alone forever. But the service() routine is built with idea that the LEDs are constantly changing, so rather then make STATIC an exception, we tolerate this quirk. I think you can work around it by adding a test for the STATIC effect when you're counting frames.
```c++
if(ws2812fx.isFrame(0) && ws2812fx.getSegment(0)->mode != FX_MODE_STATIC) { // new frame for segment 0?
```

chrisgimson, 2020-06-22:
Shame about the documentation but I can manage.  OK, I totally understand your explanation so thank you.  My statements were based on me setting & resetting a hardware port pin immediately after the depending on the state of isFrame and looking at it on an oscillosope.    So I understand that my STATIC commands are also influencing the state.  I can certainly work around it and was just trying to "cement" my understanding.

closed 2020-06-22 by chrisgimson