kitesurfer1404/WS2812FX #236: Build error after update

nparodi1, 2020-06-30:
Hi you.
Thanks a lot for your work. I have been using the library in some  projects.
Today I updated your library and Adafruit Neopixel. After that i could not Build my programs again.
 
This is the error I get trying to build the first library example in a Digispark and from PlatformIO:

|-- <Adafruit NeoPixel> 1.5.0
|-- <WS2812FX> 1.3.1
|   |-- <Adafruit NeoPixel> 1.5.0
Building in release mode
Compiling .pio/build/digispark-tiny/src/main.cpp.o
In file included from /home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/Stream.h:24:0,
                 from /home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/TinyDebugSerial.h:31,
                 from /home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/WProgram.h:18,
                 from /home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/Arduino.h:4,
                 from .pio/libdeps/digispark-tiny/Adafruit NeoPixel_ID28/Adafruit_NeoPixel.h:41,
                 from .pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:42,
                 from src/main.cpp:1:
/home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/Print.h:37:0: warning: "BIN" redefined
 #define BIN 2
 ^
In file included from /home/bonzo/.platformio/packages/toolchain-atmelavr/avr/include/avr/iotn85.h:38:0,
                 from /home/bonzo/.platformio/packages/toolchain-atmelavr/avr/include/avr/io.h:428,
                 from /home/bonzo/.platformio/packages/toolchain-atmelavr/avr/include/avr/interrupt.h:38,
                 from /home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/WProgram.h:8,
                 from /home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/Arduino.h:4,
                 from .pio/libdeps/digispark-tiny/Adafruit NeoPixel_ID28/Adafruit_NeoPixel.h:41,
                 from .pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:42,
                 from src/main.cpp:1:
/home/bonzo/.platformio/packages/toolchain-atmelavr/avr/include/avr/iotnx5.h:55:0: note: this is the location of the previous definition
 #define BIN     7
 ^
Compiling .pio/build/digispark-tiny/lib37f/WS2812FX_ID1647/WS2812FX.cpp.o
In file included from src/main.cpp:1:0:
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:257:14: error: '__FlashStringHelper' does not name a type
 static const __FlashStringHelper* _names[] = {
              ^
In file included from src/main.cpp:1:0:
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:436:27: error: '__FlashStringHelper' does not name a type
       setCustomMode(const __FlashStringHelper* name, uint16_t (*p)()),
                           ^
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:437:38: error: '__FlashStringHelper' does not name a type
       setCustomMode(uint8_t i, const __FlashStringHelper* name, uint16_t (*p)()),
                                      ^
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:461:11: error: '__FlashStringHelper' does not name a type
     const __FlashStringHelper* getModeName(uint8_t m);
           ^
In file included from src/main.cpp:1:0:
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:123:28: error: '_names' was not declared in this scope
 #define MODE_COUNT (sizeof(_names)/sizeof(_names[0]))
                            ^
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:593:34: note: in expansion of macro 'MODE_COUNT'
 static WS2812FX::mode_ptr _modes[MODE_COUNT] = {
                                  ^
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:123:43: error: '_names' was not declared in this scope
 #define MODE_COUNT (sizeof(_names)/sizeof(_names[0]))
                                           ^
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:593:34: note: in expansion of macro 'MODE_COUNT'
 static WS2812FX::mode_ptr _modes[MODE_COUNT] = {
                                  ^
In file included from /home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/Stream.h:24:0,
                 from /home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/TinyDebugSerial.h:31,
                 from /home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/WProgram.h:18,
                 from /home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/Arduino.h:4,
                 from .pio/libdeps/digispark-tiny/Adafruit NeoPixel_ID28/Adafruit_NeoPixel.h:41,
                 from .pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:42,
                 from .pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.cpp:55:
/home/bonzo/.platformio/packages/framework-arduino-avr-digistump/cores/dtiny/Print.h:37:0: warning: "BIN" redefined
 #define BIN 2
 ^
[... 62 lines cut ...]
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.cpp:320:10: note: in expansion of macro 'MODE_COUNT'
   return MODE_COUNT;
          ^
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.cpp: At global scope:
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.cpp:373:7: error: '__FlashStringHelper' does not name a type
 const __FlashStringHelper* WS2812FX::getModeName(uint8_t m) {
       ^
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.cpp:1641:39: error: '__FlashStringHelper' does not name a type
 uint8_t WS2812FX::setCustomMode(const __FlashStringHelper* name, uint16_t (*p)()) {
                                       ^
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.cpp:1646:54: error: '__FlashStringHelper' does not name a type
 uint8_t WS2812FX::setCustomMode(uint8_t index, const __FlashStringHelper* name, uint16_t (*p)()) {
                                                      ^
In file included from .pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.cpp:55:0:
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.cpp: In member function 'uint8_t WS2812FX::setCustomMode(uint8_t, const int*, uint16_t (*)())':
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.h:123:28: error: '_names' was not declared in this scope
 #define MODE_COUNT (sizeof(_names)/sizeof(_names[0]))
                            ^
.pio/libdeps/digispark-tiny/WS2812FX_ID1647/src/WS2812FX.cpp:1647:44: note: in expansion of macro 'MODE_COUNT'
   if((uint8_t)(FX_MODE_CUSTOM_0 + index) < MODE_COUNT) {
                                            ^
*** [.pio/build/digispark-tiny/src/main.cpp.o] Error 1
*** [.pio/build/digispark-tiny/lib37f/WS2812FX_ID1647/WS2812FX.cpp.o] Error 1

Thanks Thanks ThankS!

moose4lord, 2020-06-30:
I don't think this error has to do with the WS3812FX lib. See https://forum.arduino.cc/index.php?topic=397627.0

nparodi1, 2020-07-06:
Thanks Keith for the information!

closed 2020-07-06 by nparodi1