kitesurfer1404/WS2812FX #237: ws2812fx_esp32 2nd zone issue

henmel, 2020-06-30:
I have been using this library on ESP8266 for a while and aside from some WiFi related issues it has been working great. I decided to try upgrading to ESP32, but am running into some issues. Using example ws2812fx_esp32, zone 1 appears to work great, but zone 2 the colors appear to be random. I have 2 ESP32's, 30 pin and 38 pin, rev1, and they both do the same thing. I have tested with 10-30 LED's and get the same result. When I try to use the other WS2812FX examples, without RMT driver, I just get white LED's, full brightness, for the whole segment width. I also tried to use FastLED library examples, without RMT and aside from some random flickering and colors, it works ok.

Any ideas of what may be going on or how to resolve?

henmel, 2020-06-30:
By the way, regarding the comment about the bright white LEDs, I deleted, I identified the cause when uploading the regular examples to esp32. The cause was the latest update in the AdafruitNeoPixel library. I downgraded to 1.4.0 and the WS2812FX examples now work as intended. This however did not fix the RMT issues with Zone 2, described above, in the ws2812fx_esp32 example. I still get random colors in the effects.

moose4lord, 2020-06-30:
I ran the ws2812fx_esp32 example sketch with my ESP32 and it worked fine.

Note in the example sketch zone 2 is configured as if it was driving **RGBW** LEDs.
```c++
WS2812FX ws2812fx2 = WS2812FX( 60, 13, NEO_GRBW + NEO_KHZ800); //  60 RGBW LEDs driven by GPIO_13
```

You probably want to change that to normal RGB LEDs.

henmel, 2020-07-01:
That was it!!! Thank you so much!

closed 2020-07-01 by henmel