kitesurfer1404/WS2812FX #238: 2 segments with only 2 leds?

huge-11, 2020-07-08:
Hello,
i have a project that only requires 2 leds and and 2 segments, so one led per segment.
How can i do that? 
my current attempt:

                ws2812fx.setSegment(0,  0,  1, modeBLE, colors, StripSpeed, false);
                ws2812fx.setSegment(1, 1, 1, modeBLE,  colors, StripSpeed, false);

modeBLE and StripSpeed are variables int
and colors[] uint32_t

i get a strange behaviour on the second led? the first one is working fine.

i change the value of a variable and call ws2812fx.setSegment(1, 1, 1, modeBLE,  colors, StripSpeed, false);

moose4lord, 2020-07-08:
The setSegment() function takes the first and last LED index as parameters. In your case, with one LED per segment, it would look like this:
```c++
  uint32_t colors[] = {RED, GREEN, BLUE};
  ws2812fx.setSegment(0, 0, 0, FX_MODE_STATIC, colors, 1000, false);
  ws2812fx.setSegment(1, 1, 1, FX_MODE_BLINK,  colors, 1000, false);
```

huge-11, 2020-07-09:
Ok, perfect thanks.

closed 2020-08-08 by moose4lord