kitesurfer1404/WS2812FX #240: welcome

anto168, 2020-08-13:
good night, sorry to disturb everyone, i'm learning to try, led ws2812bFX, who wants how to combine several animations, so that they can loop .. for example animation effect 1 then animation effect 2, then animation 3, so looping without pause, sorry if the question I'm weird? I'm learning now ..

terimakasih,

kitesurfer1404, 2020-08-13:
Hi! Thanks for your comment and have fun, learning programming and letting LEDs blink. I think WS2812FX is a good starting point for that.

Take a look at the "Auto Mode Cycle" in the examples. It does exactly what you want to do and it is only about 20 lines of code, so easy to start and experiment with.
https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/auto_mode_cycle/auto_mode_cycle.ino

anto168, 2020-08-14:
is there another example, for example I only want animation mode 10 and animation mode 20, in the sample segment I see there are about 50 more modes, and I only want to use a few modes.

moose4lord, 2020-08-16:
The **esp8266_webinterface** example sketch has a code snippet to do what you're asking. To modify the **auto_mod_cycle** sketch, first create an array of modes/effects that you want to cycle:
```c++
uint8_t myModes[] = {FX_MODE_STATIC, FX_MODE_LARSON_SCANNER};
```
Then change the loop() function to cycle the effects in your new array:
```c++
void loop() {
  now = millis();

  ws2812fx.service();

  if(now - last_change > TIMER_MS) {
    for(uint8_t i=0; i < sizeof(myModes); i++) {
      if(myModes[i] == ws2812fx.getMode()) {
        uint8_t next_mode = ((i + 1) < sizeof(myModes)) ? myModes[i + 1] : myModes[0];
        ws2812fx.setMode(next_mode);
        break;
      }
    }
    last_change = now;
  }
}
```

anto168, 2020-08-16:
I found this sketch, sketch as intended in the original question and I tried it, it worked, but when I tried to change the color, speed, and direction of each mode, I couldn't ... can anyone help, I know very little about the program.

#include <WS2812FX.h>

#define LED_COUNT 30
#define LED_PIN D1

#define TIMER_MS 5000

#define NUM_SEGMENTS 2
#define NUM_MODES 4
uint8_t myModes[][NUM_MODES] = { // list of modes for each segment
  {FX_MODE_STATIC, FX_MODE_BLINK, FX_MODE_SCAN,      FX_MODE_RAINBOW_CYCLE}, // modes for first segment
  {FX_MODE_FADE,   FX_MODE_COMET, FX_MODE_DUAL_SCAN, FX_MODE_CHASE_COLOR}    // modes for second segment
};
uint8_t modeIndex = 0;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(255);

  // init default segments
  const uint32_t colors[] ={RED, GREEN, BLUE};
  ws2812fx.setSegment(0,           0,  (LED_COUNT/2)-1, FX_MODE_STATIC, colors, 1000, false);
  ws2812fx.setSegment(1, LED_COUNT/2,      LED_COUNT-1, FX_MODE_STATIC, colors, 1000, false);

  ws2812fx.start();
}

void loop() {
  static unsigned long then = 0;
  unsigned long now = millis();

  ws2812fx.service();

  if(now > (then + TIMER_MS)) {
    ws2812fx.resetSegmentRuntimes();
    for(uint8_t i=0; i < NUM_SEGMENTS; i++) { // update the mode for all segments
      ws2812fx.setMode(i, myModes[i][modeIndex]);
    }
    modeIndex = (modeIndex + 1) % NUM_MODES; // increment the mode index
    then = now;
  }
}

anto168, 2020-08-16:
how do I change each mode, I want to try changing the color, speed, and direction of each mode, such as setting the segment () set (index, star, stop, mode, speed, true / false)

moose4lord, 2020-08-16:
You can run setSegment() at any time to change a segment's mode, color, speed or reverse. Just remember to run resetSegment() before your setSegment() statements.
```c++
ws2812fx.resetSegments();                             
ws2812fx.setSegment(0, 0,           (LED_COUNT/2)-1, FX_MODE_COMET, colors, 2000, true);
ws2812fx.setSegment(1, LED_COUNT/2, LED_COUNT-1,     FX_MODE_FADE,  colors, 3000, true);
```

Or if you just want to change one aspect of a segment, you can use the setColor(), setSpeed() or setOptions() functions, similar to what's being done with the setMode() function.
```c++
ws2812fx.setColor(i, myColors[i][colorIndex]);
ws2812fx.setSpeed(i, mySpeeds[i][speedIndex]);
ws2812fx.setOptions(i, myOptions[i][optionIndex]);
```

anto168, 2020-08-17:
if I change the color, speed, and direction in this script, then the result is that all modes in [NUM_MODES] will change, not every mode that I want to change the color, speed, direction I want.

ws2812fx.resetSegments();                             
ws2812fx.setSegment(0, 0,           (LED_COUNT/2)-1, FX_MODE_COMET, colors, 2000, true);
ws2812fx.setSegment(1, LED_COUNT/2, LED_COUNT-1,     FX_MODE_FADE,  colors,

moose4lord, 2020-08-17:
If you want to change more then the mode dynamically, you might want to just run setSegment() statements instead of setMode(). That way you'd have independent control of all the segment parameters. Maybe this:
```c++
#include <WS2812FX.h>

#define LED_COUNT 30
#define LED_PIN D1

#define SEG0_START 0
#define SEG0_END   ((LED_COUNT/2) - 1)
#define SEG1_START (LED_COUNT/2)
#define SEG1_END   (LED_COUNT - 1)

#define TIMER_MS 5000
#define NUM_MODES 4

uint8_t modeIndex = 0;
const uint32_t colors[] ={RED, GREEN, BLUE};

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(255);

  // init default segments
  ws2812fx.setSegment(0, SEG0_START, SEG0_END, FX_MODE_STATIC, colors, 1000, false);
  ws2812fx.setSegment(1, SEG1_START, SEG1_END, FX_MODE_STATIC, colors, 1000, false);

  ws2812fx.start();
}

void loop() {
  static unsigned long then = 0;
  unsigned long now = millis();

  ws2812fx.service();

  if(now > (then + TIMER_MS)) {
    ws2812fx.resetSegments();
    if(modeIndex == 0) {
      ws2812fx.setSegment(0, SEG0_START, SEG0_END, FX_MODE_STATIC, colors, 1000, false);
      ws2812fx.setSegment(1, SEG1_START, SEG1_END, FX_MODE_FADE,   colors, 1000, false);
    } else if(modeIndex == 1) {
      ws2812fx.setSegment(0, SEG0_START, SEG0_END, FX_MODE_BLINK, colors, 1000, false);
      ws2812fx.setSegment(1, SEG1_START, SEG1_END, FX_MODE_COMET, colors, 1000, false);
    } else if(modeIndex == 2) {
      ws2812fx.setSegment(0, SEG0_START, SEG0_END, FX_MODE_SCAN,      colors, 1000, false);
      ws2812fx.setSegment(1, SEG1_START, SEG1_END, FX_MODE_DUAL_SCAN, colors, 1000, false);
    } else if(modeIndex == 3) {
      ws2812fx.setSegment(0, SEG0_START, SEG0_END, FX_MODE_RAINBOW_CYCLE, colors, 1000, false);
      ws2812fx.setSegment(1, SEG1_START, SEG1_END, FX_MODE_CHASE_COLOR,   colors, 1000, false);
    }

    modeIndex = (modeIndex + 1) % NUM_MODES; // increment the mode index
    then = now;
  }
}
```

anto168, 2020-08-17:
thank you for helping me learn this, 

I want to ask again, in this script, the beginning was like this ..

ws2812fx.setSegment(0, SEG0_START, SEG0_END, FX_MODE_SCAN,      colors, 1000, false);
ws2812fx.setSegment(1, SEG1_START, SEG1_END, FX_MODE_DUAL_SCAN, colors, 1000, false);

then I try to change, to be like this, try to change the starting point of the led start,

ws2812fx.setSegment(0, SEG0_START, SEG0_END, FX_MODE_SCAN,      colors, 1000, false);
ws2812fx.setSegment(0, SEG5_START, SEG0_END, FX_MODE_SCAN,      colors, 1000, false);
ws2812fx.setSegment(1, SEG1_START, SEG1_END, FX_MODE_DUAL_SCAN, colors, 1000, false);

but it doesn't work, is it wrong to be like that? hehehehe😁😁

moose4lord, 2020-08-17:
The LED start and end macros are setup at the top of the sketch. 
```c++
#define SEG0_START 0
#define SEG0_END   ((LED_COUNT/2) - 1)
#define SEG1_START (LED_COUNT/2)
#define SEG1_END   (LED_COUNT - 1)
```
You'll have to change them if you want to change the start/end LED in each segment.

anto168, 2020-08-18:
#define SEG0_START 0
#define SEG0_END   ((LED_COUNT/2) - 1)
#define SEG1_START (LED_COUNT/2)
#define SEG1_END   (LED_COUNT - 1)

I have tried to change this in this section, but the results change the overall effect mode, not a particular effect. For example, I want the FX_COMET mode which I change the initial position.

anto168, 2020-08-18:
I want to change it to a certain effect2, but it always gets an error.

moose4lord, 2020-08-18:
The LED start/end macros are useful if the segments are always the same size. If you're going to have segments with different sizes, you can just hardcode the start LED and end LED in the setSegment() statement:
```c++
ws2812fx.setSegment(0,  0, 17, FX_MODE_SCAN, colors, 1000, false);
ws2812fx.setSegment(1, 18, 29, FX_MODE_DUAL_SCAN, colors, 1000, false);
```

Also, I'll mention, you don't always need to have two segments. You can setup a different number of segments for each effect.
```c++
if(now > (then + TIMER_MS)) {
    ws2812fx.resetSegments();
    if(modeIndex == 0) {
      ws2812fx.setSegment(0,  0, 14, FX_MODE_STATIC, colors, 1000, false);
      ws2812fx.setSegment(1, 15, 29, FX_MODE_FADE,   colors, 1000, false);
    } else if(modeIndex == 1) {
      ws2812fx.setSegment(0, 0, 29, FX_MODE_BLINK, colors, 1000, false);
    } else if(modeIndex == 2) {
      ws2812fx.setSegment(0,  0,  9, FX_MODE_SCAN,      colors, 1000, false);
      ws2812fx.setSegment(1, 10, 19, FX_MODE_DUAL_SCAN, colors, 1000, false);
      ws2812fx.setSegment(2, 20, 29, FX_MODE_COMET, colors, 1000, false);
    }
    ...
```

anto168, 2020-08-18:
thank you for helping me a lot,

I will try to change the program that you provide, if I have trouble I will ask again here

anto168, 2020-08-18:
i have 60 leds, i divide by 2.

I write scripts like this, but in the last segment the led is all silent, it does not immediately loop back to the start mode,

am I wrong if I divide into 6 segments like this?

if(now > (then + TIMER_MS)) {
    ws2812fx.resetSegments();
    if(modeIndex == 0) {
      ws2812fx.setSegment(0,  0, 30, FX_MODE_COMET, colors, 500, false);
      ws2812fx.setSegment(1, 31, 60, FX_MODE_COMET, colors, 500, true);
    } else if(modeIndex == 1) {
      ws2812fx.setSegment(0,  0, 30, FX_MODE_COMET, PURPLE, 500,true);
      ws2812fx.setSegment(1, 31, 60, FX_MODE_COMET, PURPLE, 500,false);
    } else if(modeIndex == 2) {
      ws2812fx.setSegment(0,  0,  9, FX_MODE_SCAN, colors, 100, false);
      ws2812fx.setSegment(1, 10, 19, FX_MODE_BLINK,colors, 100, false);
      ws2812fx.setSegment(2, 20, 29, FX_MODE_SCAN, colors, 100, false);
      ws2812fx.setSegment(3, 30, 39, FX_MODE_SCAN, colors, 100, false);
      ws2812fx.setSegment(4, 40, 49, FX_MODE_BLINK,colors, 100, false);
      ws2812fx.setSegment(5, 50, 59, FX_MODE_SCAN, colors, 100, false);
    }

moose4lord, 2020-08-18:
The TIMER_MS macro sets the delay (in milliseconds) between mode changes.
```c++
#define TIMER_MS 5000
```

anto168, 2020-08-19:
if I set it on that line, 

#define TIMER_MS 5000

then each mode has the same delay, 

how do I set the delay for each mode to be different, because there are some modes that don't fit if the delay is too short or too long ..

anto168, 2020-08-19:
For example, I want to set FX_MODE_RAINBOW_CYCLE mode to be around 5 seconds longer, and set FX_MODE_COMET to 2 SECONDS ... please help, thanks ...

moose4lord, 2020-08-19:
You're right. TIMER_MS is a constant (used in the loop() function to trigger the effect changes), so each effect is shown for the same duration.
```c++
if(now > (then + TIMER_MS)) {
```
But instead of a constant, you could use an integer variable, and set the variable to a different duration every time you use setSegment() to change the effect. Here's a nice explanation of variables on the Arduino web site:
[https://www.arduino.cc/en/Reference/VariableDeclaration](https://www.arduino.cc/en/Reference/VariableDeclaration)

anto168, 2020-08-19:
hehehehe, sorry to interrupt, I have tried to change the delay function change, but it doesn't work, try to put it in several lines but it can't ... can you help me, sorry for my disturbance ..  😁😁😁

moose4lord, 2020-08-19:
Show me what you've done and I'll try to get you back on track.

anto168, 2020-08-19:
example that I made, last I added like this delay (1000); no error message however, the led display got messy.
😁😁

example that I made, last I added like this delay (500); no error message however, the led display got messy. if i try to clear the TIMER_MS code the led doesn't work.

if(now > (then + TIMER_MS)) {
    ws2812fx.resetSegments();
    if(modeIndex == 0) {
      ws2812fx.setSegment(0,  0, 14, FX_MODE_STATIC, colors, 1000, false);
      ws2812fx.setSegment(1, 15, 29, FX_MODE_FADE,   colors, 1000, false);
    }
delay(1000);

moose4lord, 2020-08-19:
Using the delay() function is not allowed when using the Ws2812FX library. What you want to do is create a global variable that will control the effect duration.
```c++
int duration = 5000; // effect duration in milliseconds
```
Then when you execute the setSegment() functions to change the effect, update the duration variable to however long you want the effect to last.
```c++
if(now > (then + duration)) { // use the duration variable instead of the TIMER_MS constant
    ws2812fx.resetSegments();
    if(modeIndex == 0) {
      ws2812fx.setSegment(0,  0, 14, FX_MODE_STATIC, colors, 1000, false);
      ws2812fx.setSegment(1, 15, 29, FX_MODE_FADE,   colors, 1000, false);
      duration = 10000; // run the effect for 10 seconds
    } else if(modeIndex == 1) {
    ...
```

anto168, 2020-08-21:
thank you for your help, I have successfully run, the program you are teaching ...

 I have something to ask again, at se brightness (255) the blue color appears normal, if I reduce the brightness (200) from blue to purple, the problem is in My hardware or settings are wrong ..

anto168, 2020-08-27:
hello, sorry to interrupt, I want to ask, can 2 modes be combined in one segment? For example, dual scan mode and color wip

moose4lord, 2020-08-27:
If you mean can two effects be overlayed in the same segment, then no, that's not possible. At least not easily. I suppose you could create a custom effect using the code for the dual_scan and color_wipe effects, but it wouldn't be easy.

anto168, 2020-08-29:
sorry to bother, I tried to increase the speed, but it doesn't work, is the default only 100ms ...

moose4lord, 2020-08-29:
The default id 1000ms, but you should be able to change the speed of the animation with the setSpeed() function. Or the setSegment() function and passing a speed parameter. Can you paste your code?

anto168, 2020-08-29:
ws2812fx.setSegment(0, 0, 59, FX_MODE_RAINBOW_CYCLE, colors, 1000, false);

I use set segment (), I try to change the speed to 50ms but the speed is the same as 100ms.

anto168, 2020-08-29:
the other effect is the same

moose4lord, 2020-08-29:
The speed for the FX_MODE_RAINBOW_CYCLE effect is limited by the amount of time it takes to update the 60 LEDs in your strip (approximately 2ms). Setting speed to anything less than about 500 will not make the animation go any faster.

closed 2021-03-01 by moose4lord