kitesurfer1404/WS2812FX #241: Errors whit "setSegment" amount

Devilscave, 2020-08-16:
First: this is a translated text.
I have problems with the "setSegment" function.
1. It doesn't go to put more than 10 segments (need 24)
2. In effects that are changed afterwards, the segment string "0" is taken over as the LEDs,
    I was able to fix it by giving it back the entire length.
3. The segment effect suddenly flashes through in other effects, in a 1 second delay (but i use not delay) 
4. Other effects begin to mix with it (40/41/42) all "running" effects are now pushing these segments in front of them, but these fades are re-inserted every second and are pushed again. Very strange!!!

I made a test sketch for you, to show the error.
[https://pastebin.com/ty0w8KU6](url)

It runs on a ESP8266 (NODE_MCU V3) whit 306 LEDs
D1 for the LEDs
D2 for a Button (normal High, pressed Low)

Is there a way to delete the segment division after the effect has been used?

ANNOTATION / IDEA
can you add the Fade_Mode (15) as a Fade_Random! So it change the color after every fadeout.

moose4lord, 2020-08-16:
To increase the number of segments, add two parameters to the WS2812FX constructor:
```c++
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800, 24, 24);
```
This will increase the maximum number of segments to 24, and allow all 24 to be active at the same time.

When you change segments dynamically, you need to call resetSegments() before changing the segments.
```c++
    switch(menu) {                 
      case 1:                                                   
        Serial.println("Fire-Flicker 50");
        ws2812fx.resetSegments(); // add this                             
        ws2812fx.setSegment(0,  0, LED_COUNT -1, 50, 0xB22222, 7000, false);
```

I'll have to look at the fade mode. A random fade seems like a good idea.

Devilscave, 2020-08-16:
Ahh OK, thanks now it's working.
But I had a strange combilier bug.
I added a "Case 13" which contains this segment code but it always gave me this error:
"jump to case label [-fpermissive]"
and referred to the 13 of case.
Then I did nothing but swap case 12 and 13 in the order and off I went.