kitesurfer1404/WS2812FX #243: Webinterface custom mode show of out order at the webpage

AlejandroSantosG, 2020-08-17:
Hello Guys, I am trying to made a custom mode:

uint8_t myModes[] = {FX_MODE_STATIC, FX_MODE_RAINBOW, FX_MODE_LARSON_SCANNER};

And the webpage show, the right names for the custom mode, but the efect  activated it´s just the order as the number mode at the library:

#define FX_MODE_STATIC         0
#define FX_MODE_BLINK           1
#define FX_MODE_BREATH        2

Custom Mode:

FX_MODE_STATIC                              Activate        FX_MODE_STATIC  0
FX_MODE_RAINBOW                         Activate        FX_MODE_BLINK    1
FX_MODE_LARSON_SCANNER       Actiavte        FX_MODE_BREATH  2

Could you help to fix it please, thanks

moose4lord, 2020-08-17:
What sketch are you working with?

moose4lord, 2020-08-17:
There is indeed a bug in the esp8266_webinterface sketch that expresses itself when myModes[] is used. To fix, go to line 231 in the esp8266_webinterface.ino file and change it from this:
```c++
ws2812fx.setMode(tmp % ws2812fx.getModeCount());
```
to this:
```c++
uint8_t new_mode = sizeof(myModes) > 0 ? myModes[tmp % sizeof(myModes)] : tmp % ws2812fx.getModeCount();
ws2812fx.setMode(new_mode);
```
Let me know if that works for you.

AlejandroSantosG, 2020-08-18:
Thanks, now it's working perfectly, Thank you very much!

closed 2020-08-18 by AlejandroSantosG