kitesurfer1404/WS2812FX #246: Should it be possible to use this library on the LGT8F328?

seanspotatobusiness, 2020-08-27:
I tried to compile a sketch using this library for the LGT8F328 and got a lot of errors. Does anyone know whether it's reasonable to expect to be able to make this work on the LGT8F328? Thanks.

I'm using Arduino IDE 1.8.13 on Windows 10 with variant 328P-LQFP32 of the LGT8F328 clone of the ATMEGA328P.

`E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'void WS2812FX::decreaseLength(uint16_t)':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:249:3: error: 'fill' was not declared in this scope
   fill(BLACK, _segments[0].start, seglen);
   ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:249:3: note: suggested alternative: 'ftell'
   fill(BLACK, _segments[0].start, seglen);
   ^~~~
   ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::mode_static()':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:593:3: error: 'fill' was not declared in this scope
   fill(_seg->colors[0], _seg->start, _seg_len);
   ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:593:3: note: suggested alternative: 'ftell'
   fill(_seg->colors[0], _seg->start, _seg_len);
   ^~~~
   ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::blink(uint32_t, uint32_t, bool)':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:607:5: error: 'fill' was not declared in this scope
     fill(color, _seg->start, _seg_len);
     ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:607:5: note: suggested alternative: 'ftell'
     fill(color, _seg->start, _seg_len);
     ^~~~
     ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:612:5: error: 'fill' was not declared in this scope
     fill(color, _seg->start, _seg_len);
     ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:612:5: note: suggested alternative: 'ftell'
     fill(color, _seg->start, _seg_len);
     ^~~~
     ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::mode_random_color()':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:731:3: error: 'fill' was not declared in this scope
   fill(color, _seg->start, _seg_len);
   ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:731:3: note: suggested alternative: 'ftell'
   fill(color, _seg->start, _seg_len);
   ^~~~
   ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::mode_breath()':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:786:3: error: 'fill' was not declared in this scope
   fill(color, _seg->start, _seg_len);
   ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:786:3: note: suggested alternative: 'ftell'
   fill(color, _seg->start, _seg_len);
   ^~~~
   ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::mode_fade()':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:805:3: error: 'fill' was not declared in this scope
   fill(color, _seg->start, _seg_len);
   ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:805:3: note: suggested alternative: 'ftell'
   fill(color, _seg->start, _seg_len);
   ^~~~
   ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::scan(uint32_t, uint32_t, bool)':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:823:3: error: 'fill' was not declared in this scope
   fill(color2, _seg->start, _seg_len);
   ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:823:3: note: suggested alternative: 'ftell'
   fill(color2, _seg->start, _seg_len);
   ^~~~
   ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::mode_rainbow()':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:866:3: error: 'fill' was not declared in this scope
   fill(color, _seg->start, _seg_len);
   ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:866:3: note: suggested alternative: 'ftell'
   fill(color, _seg->start, _seg_len);
   ^~~~
   ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::twinkle(uint32_t, uint32_t)':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:984:5: error: 'fill' was not declared in this scope
     fill(color2, _seg->start, _seg_len);
     ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:984:5: note: suggested alternative: 'ftell'
     fill(color2, _seg->start, _seg_len);
     ^~~~
     ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::twinkle_fade(uint32_t)':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1098:5: error: 'fill' was not declared in this scope
     fill(color, index, size);
     ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1098:5: note: suggested alternative: 'ftell'
     fill(color, index, size);
     ^~~~
     ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::sparkle(uint32_t, uint32_t)':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1127:5: error: 'fill' was not declared in this scope
     fill(color1, _seg->start, _seg_len);
     ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1127:5: note: suggested alternative: 'ftell'
     fill(color1, _seg->start, _seg_len);
     ^~~~
     ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1131:3: error: 'fill' was not declared in this scope
   fill(color1, _seg->start + _seg_rt->aux_param3, size);
   ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1131:3: note: suggested alternative: 'ftell'
   fill(color1, _seg->start + _seg_rt->aux_param3, size);
   ^~~~
   ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::mode_hyper_sparkle()':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1164:3: error: 'fill' was not declared in this scope
   fill(_seg->colors[0], _seg->start, _seg_len);
   ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1164:3: note: suggested alternative: 'ftell'
   fill(_seg->colors[0], _seg->start, _seg_len);
   ^~~~
   ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::mode_multi_strobe()':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1180:3: error: 'fill' was not declared in this scope
   fill(_seg->colors[1], _seg->start, _seg_len);
   ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1180:3: note: suggested alternative: 'ftell'
   fill(_seg->colors[1], _seg->start, _seg_len);
   ^~~~
   ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::fireworks(uint32_t)':
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1495:9: error: 'fill' was not declared in this scope
         fill(color, index, size);
         ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1495:9: note: suggested alternative: 'ftell'
         fill(color, index, size);
         ^~~~
         ftell
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1502:7: error: 'fill' was not declared in this scope
       fill(color, index, size);
       ^~~~
E:\Projects\Electronics\Arduino sketches\libraries\WS2812FX\src\WS2812FX.cpp:1502:7: note: suggested alternative: 'ftell'
       fill(color, index, size);
       ^~~~
       ftell
exit status 1
Error compiling for board LGT8F328.
`

moose4lord, 2020-08-27:
It looks like you're using an old version of the Adafruit_Neopixel library. Try updating to the latest.

seanspotatobusiness, 2020-08-27:
Thanks very much, Keith; that seems to have completely resolved all
the errors for WS2812FX version 1.3.1!

closed 2020-08-29 by seanspotatobusiness