kitesurfer1404/WS2812FX #249: FX_MODE_STATIC fucks my ESP up

Devilscave, 2020-08-31:
Hello, i am again...
I recently asked why only 10 segments work with an LED segment division. I got the answer to that I should append (, 24,24) to the config line
Like: `WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800, **24 ,24**);

So the sketch ran OK so far, even if I had to write all the other effects in a complete config line because otherwise they wouldn't work. like this: 
`case 12:`                                                   
`                     ws2812fx.resetSegments();`
`                     ws2812fx.setBrightness(80);`
`                     ws2812fx.setSegment(0, 0, LED_COUNT - 1, 10, 0x007BFF, 1100, NO_OPTIONS);`
`break;`

Then I added the sketch WebInterface to my sketch and changed it and created a single effect call that looks like this:
`case 13:`     
`                     ws2812fx.setSegment(0,  0, 15,     FX_MODE_STATIC, RED, 100, false);`
`                     ws2812fx.setSegment(1, 16, 27,    FX_MODE_STATIC, GREEN, 100, false);`
`                     ws2812fx.setSegment(2, 28, 41,    FX_MODE_STATIC, BLUE, 100, false);`
`                     ws2812fx.setSegment(3, 42, 51,    FX_MODE_STATIC, YELLOW, 100, false);`
`                     ws2812fx.setSegment(4, 52, 63,    FX_MODE_STATIC, MAGENTA, 100, false);`
`                     ws2812fx.setSegment(5, 64, 79,    FX_MODE_STATIC, CYAN, 100, false);`
`                     ws2812fx.setSegment(6, 80, 93,    FX_MODE_STATIC, WHITE, 100, false);`
`                     ws2812fx.setSegment(7, 94, 103,   FX_MODE_STATIC, RED, 100, false);`
`                     ws2812fx.setSegment(8, 104, 117,  FX_MODE_STATIC, ORANGE, 100, false);`
`                     ws2812fx.setSegment(9, 118, 129,  FX_MODE_STATIC, CYAN, 100, false);`
`                     ws2812fx.setSegment(10, 130, 139, FX_MODE_STATIC, GREEN, 100, false);`
`                     ws2812fx.setSegment(11, 140, 155, FX_MODE_STATIC, YELLOW, 100, false);`
`                     ws2812fx.setSegment(12, 156, 169, FX_MODE_STATIC, PURPLE, 100, false);`
`                     ws2812fx.setSegment(13, 170, 179, FX_MODE_STATIC, RED, 100, false);`
`                     ws2812fx.setSegment(14, 180, 191, FX_MODE_STATIC, BLUE, 100, false);`
`                     ws2812fx.setSegment(15, 192, 201, FX_MODE_STATIC, ORANGE, 100, false);`
`                     ws2812fx.setSegment(16, 202, 215, FX_MODE_STATIC, GREEN, 100, false);`
`                     ws2812fx.setSegment(17, 216, 231, FX_MODE_STATIC, WHITE, 100, false);`
`                     ws2812fx.setSegment(18, 232, 243, FX_MODE_STATIC, MAGENTA, 100, false);`
`                     ws2812fx.setSegment(19, 244, 257, FX_MODE_STATIC, CYAN, 100, false);`
`                     ws2812fx.setSegment(20, 258, 267, FX_MODE_STATIC, RED, 100, false);`
`                     ws2812fx.setSegment(21, 268, 279, FX_MODE_STATIC, ORANGE, 100, false);`
`                     ws2812fx.setSegment(22, 280, 289, FX_MODE_STATIC, BLUE, 100, false);`
`                     ws2812fx.setSegment(23, 290, 305, FX_MODE_STATIC, WHITE, 100, false);`
`break;`

This was still running in the WebInterface, but now I have unwritten it as an Access_Point so that I can use a direct connection, this is also possible. Only when I call up this effect above do you notice how the ESP suddenly becomes slow, other effects are no longer adopted so quickly and suddenly no longer react, in such a way that the ESP starts a self-start with this error: "Exception (29)"
Can i write this effect somehow "more elegantly"? With the basic segment "0-306" and still divide the colors as desired?

moose4lord, 2020-09-01:
I created 24 segments as you describe using the esp8266_webinterface example sketch and I didn't see any problem. The ESP did not run slow and didn't throw an exception. Sorry, I'm not sure why it's not working for you.

Devilscave, 2020-09-02:
I think it has something to do with how the website transmits your data, as a web server I have no problems at all, the connection always works, is fast, almost no delay and very stable with a long range. As an access point, it repeatedly causes problems with the connection, it finds it and it simply breaks off, it doesn't just seem to be due to this effect, but rather to how often or quickly I change effects or values on the website. As if the memory was overflowing.

Exception 29 = A store referenced a page mapped with an attribute that does not permit stores
[https://links2004.github.io/Arduino/dc/deb/md_esp8266_doc_exception_causes.html](url)

I do not know exactly what this error means, especially since I have not changed the code of the website or the effects between WebServer and AccessPoint. And it never happens immediately but always after a few changes of attributes or effects, but especially often directly after selecting the above.

Today I will try to relieve the ESP of some work by releasing it from drawing the LEDs again after selecting the static effect, the WS2812 remember the color anyway, so you don't need new data in each run.

I hope the text is understandable, I had it translated (from German)

Devilscave, 2020-09-08:
Today I found time and changed your web interface to access point function. But this also shows me the same mistakes. If I play it and connect to the ESP, it loses the connection shortly afterwards. I attached the sketch and only commented out the original unused parts of the webinterface. so that you can keep track of it. Maybe you will find a bug or a better solution.
However, I noticed that since the change in the myModes indicator #243 is now always on the website.
(And thanks for the title change, I couldn't think of anything better at the moment.)
[https://pastebin.com/kwbC1b8c](url)

moose4lord, 2020-09-09:
I tried your sketch and I also see random network disconnects and ESP crashes. SoftAP mode just doesn't seem to be as robust as station mode. I added some decoupling caps on the ESP's 3.3V supply, which seemed to help. I also compiled with the old SDK 2.2.1 option and that seemed to help too. But I never got the problem to go away completely. Sorry, I don't think I can be much help.

Devilscave, 2020-09-09:
When testing, I use a Lolin NodeMCU V3, my ESP in the finished construction has a 100µF capacitor on the +/- and the ws2812 Leds hav a 1000µF.  I also have the main problem that after a reset the network appears and logging in is possible but the connection is terminated immediately when i will opened the website. Other AccessPoint sketches are also running. I suspect the site is sending something wrong to the ESP. To do this, I tried to find out exactly what the problem was with the "ESP Exception Decoder" tool, but I can't really handle its output. It seems to me that the memory is running full or that parts of the program are accessing the wrong memory blocks. But which!

Devilscave, 2020-09-09:
here is my Exception result
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x4000e1c3
EXCVADDR: 0x00000018

> Decoding stack results
0x40100ac4: malloc(size_t) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x40100d22: calloc(size_t, size_t) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 908
0x40100d14: calloc(size_t, size_t) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 902
0x4020b8e4: loop_wrapper() at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 189
0x40215b40: lwip_cyclic_timer at core/timeouts.c line 233
0x40100ac4: malloc(size_t) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x40215ae5: sys_timeout_abs at core/timeouts.c line 189
0x4020b760: loop_task(ETSEvent*) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 205
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x40207c97: NeoPixelPainterCanvas::addColorHSV(int, unsigned char, unsigned char, unsigned char) at C:\Users\Devilscave\Documents\Arduino\libraries\NeoPixel_Painter\src\NeoPixelPainter.cpp line 193
0x40207c97: NeoPixelPainterCanvas::addColorHSV(int, unsigned char, unsigned char, unsigned char) at C:\Users\Devilscave\Documents\Arduino\libraries\NeoPixel_Painter\src\NeoPixelPainter.cpp line 193
0x402077fc: Adafruit_NeoPixel::show() at C:\Users\Devilscave\Documents\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp line 2207
0x40209f7c: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266/HardwareSerial.h line 164
0x4020a24d: Print::write(char const*) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266/Print.h line 62
0x40207b2c: NeoPixelPainterCanvas::addColorRGB(int, RGB) at C:\Users\Devilscave\Documents\Arduino\libraries\NeoPixel_Painter\src\NeoPixelPainter.cpp line 172
0x4020c6fd: uart_write(uart_t*, char const*, size_t) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\uart.cpp line 509
0x40207cb1: NeoPixelPainterCanvas::addColorHSV(int, unsigned char, unsigned char, unsigned char) at C:\Users\Devilscave\Documents\Arduino\libraries\NeoPixel_Painter\src\NeoPixelPainter.cpp line 195
0x40207cef: NeoPixelPainterCanvas::transfer() at C:\Users\Devilscave\Documents\Arduino\libraries\NeoPixel_Painter\src\NeoPixelPainter.cpp line 55
0x40207805: Adafruit_NeoPixel::show() at C:\Users\Devilscave\Documents\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp line 2210
0x40201b0a: zapfen_rush() at E:\Arduino Zeug\Projekte\Zapfen_lichterkette\eis_test5_AP_MODE/eis_test5_AP_MODE.ino line 451
0x4020411f: loop() at E:\Arduino Zeug\Projekte\Zapfen_lichterkette\eis_test5_AP_MODE/eis_test5_AP_MODE.ino line 172
0x4020d400: tune_timeshift64(uint64_t) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\time.cpp line 53
0x4020d454: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\devilscave\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x40100241: esp_schedule() at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 125
0x4020b90d: loop_wrapper() at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 199

closed 2021-03-01 by moose4lord