kitesurfer1404/WS2812FX #250: Speed up animations more?

craigrobbo, 2020-09-01:
Hi there, I've been using the colour wipe function in my setup but I seem to be running into a limitation of how fast it can run, I have tried running at speed "500" and lower down to "100" but it's not quick enough over a long strip.

Is there any way at all to speed it up more?

moose4lord, 2020-09-01:
Are you using an Arduino of ESP processor? The ESP can run much faster frame rates.

How many LEDs in your setup? The color wipe effect , although changing only one pixel for each animation frame, has to update ALL the LEDs in your strip for each frame, just because of the serial nature of the signal protocol. For very long LED strips this "update" time can be quite long.

Having said that, there is an ugly hack to get things to run faster. In the service() function of the library's WS2812FX.cpp file (line 81) there's this line:
```c++
uint16_t delay = (this->*_modes[_seg->mode])();
```
That line creates one animation frame, so if you repeat the line, then two animation frames will be generated (only the second one will actually be shown). Effectively the animation will run twice as fast.
```c++
uint16_t delay = (this->*_modes[_seg->mode])();
         delay = (this->*_modes[_seg->mode])();
```
ALL effects will run twice as fast, not just the color wipe effect. With very long strips of LEDs that may be what you want, or it may not.

craigrobbo, 2020-09-05:
That worked Keith thank you!

Am I right in assuming the more of this line hi  add the faster they will go (skip frames technically)

moose4lord, 2020-09-05:
That's right, three lines will triple the animation speed, four lines will quadruple the speed, and so on. If you crank up the speed too much, you'll start to notice the skipped frames.

craigrobbo, 2020-09-05:
Thank you thats perfect, and yes youre right there is a bit of reduced smoothness but as the sped increases it seems less noticeable 

Thanksa agin keith

closed 2021-03-01 by moose4lord