kitesurfer1404/WS2812FX #251: Idea: Add a notification effect

SamueldaCostaAraujoNunes, 2020-09-04:
The idea consists of a special mode, which takes a color as a parameter (Each different trigger would send a different color), and would generate a pulse, similar to the Comet Effect, with the specified color. However, the comet would act on the current effect, adding it.

moose4lord, 2020-09-04:
What kind of trigger were you thinking about? A button press or some kind of network message?

SamueldaCostaAraujoNunes, 2020-09-04:
A network trigger, before using the WS2812FX, I used a led strip, very simple. But, I created a library, which had a Notification function(https://github.com/SamueldaCostaAraujoNunes/LedStripLibrary), the effect is similar to that of breathing, in which I received different colors, by IFTTT triggers through a WebHook, I would like to reproduce a similar effect using WS2812FX !!

moose4lord, 2020-09-09:
Ok, I had a few minutes to look at this. It seems there are two parts to your question. Using IFTTT to trigger effect changes, and blending effects to create a composite effect (I think that's what you mean when you mention adding the Comet effect to the current effect). Is it the IFTTT part you're having trouble with, or the blending part? Or both?

SamueldaCostaAraujoNunes, 2020-09-09:
My problem is in the compound effect! Example: when there is a change, in a specific repository on my Github, it triggers a trigger, which sends a color to my led strip. I wanted to insert a single comet (with the color I received, so I can differentiate the notifications), within the current effect, and finally this comet, the effect that was working, continues to be reproduced normally!

moose4lord, 2020-09-10:
This topic of combining effects comes up every once in a while. I think the best way to accomplish this is with a custom show() function that merges the data from two "virtual" strips into one "physical" strip. This is what I came up with:
```c++
#include <WS2812FX.h>

#define LED_PIN     4
#define LED_COUNT 144

// create instances of two virtual strips and one physical strip.
// the animations will each run independently in the virtual strips,
// but the LED data will be merged and sent to the physical strip's LEDs.
WS2812FX ws2812fx_v1 = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_v2 = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_p  = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  // initialize the virtual strips
  ws2812fx_v1.init();
  ws2812fx_v1.setBrightness(255);
  ws2812fx_v1.setSegment(0, 0, LED_COUNT-1, FX_MODE_BREATH, RED, 1000, NO_OPTIONS);
  ws2812fx_v1.start();

  ws2812fx_v2.init();
  ws2812fx_v2.setBrightness(255);
  ws2812fx_v2.setSegment(0, 0, LED_COUNT-1, FX_MODE_COMET, BLUE, 3000, NO_OPTIONS);
  ws2812fx_v2.start();

  // initialize the physical strip
  ws2812fx_p.init();

  // config custom show() functions for the virtual strips, so their
  // pixel data gets merged and sent to the physical strip's LEDs
  ws2812fx_v1.setCustomShow(myCustomShow);
  ws2812fx_v2.setCustomShow(myCustomShow);
}

void loop() {
  ws2812fx_v1.service();
  ws2812fx_v2.service();
}

// update the physical strip's LEDs and run it's show() function
void myCustomShow(void) {
  uint8_t *pixels_v1 = ws2812fx_v1.getPixels();
  uint8_t *pixels_v2 = ws2812fx_v2.getPixels();
  uint8_t *pixels_p  = ws2812fx_p.getPixels();
  for (int i=0; i < ws2812fx_p.getNumBytes(); i++) {
    pixels_p[i] = (pixels_v1[i] / 2) + (pixels_v2[i] / 2);
  }
  ws2812fx_p.Adafruit_NeoPixel::show();
}
```

SamueldaCostaAraujoNunes, 2020-09-10:
This is exactly what I wanted, amazing! I will study this code, but from what I tested here, it will suit me perfectly! Grateful!

SamueldaCostaAraujoNunes, 2020-09-10:
I really liked the solution, but when I apply this average to the colors, with the Comet effect, all colors have their brightness reduced by half, I would like to know, if there is no way to overwrite on top of the effects, only the Comet itself. Maintaining the color integrity of the two effects!

SamueldaCostaAraujoNunes, 2020-09-10:
As each notification has a specific color, merging the colors of the effects ends up losing the main function of the notification!

moose4lord, 2020-09-10:
You could turn down the brightness of the "v1" strip, so that the color of "v2" stands out more.
```c++
ws2812fx_v1.setBrightness(64);
```

Or if you really don't like blending the colors, instead of averaging the two pixel colors, you could test the value of the "v2" pixel (in the strip running the COMET effect). If it's BLACK, copy the pixel color from the "v1" strip to the "p" strip, otherwise copy the "v2" pixel color.
```c++
void myCustomShow(void) {
  for (int i=0; i < ws2812fx_p.numPixels(); i++) {
    if(ws2812fx_v2.getPixelColor(i) == BLACK) {
      ws2812fx_p.setPixelColor(i, ws2812fx_v1.getPixelColor(i));
    } else {
      ws2812fx_p.setPixelColor(i, ws2812fx_v2.getPixelColor(i));
    }
  }
  ws2812fx_p.Adafruit_NeoPixel::show();
}
```
Note for this to work you have to completely initialize the "p" strip and call it's service() routine at least once.
```c++
  // initialize the physical strip
  ws2812fx_p.init();
  ws2812fx_p.setBrightness(255);
  ws2812fx_p.setSegment(0, 0, LED_COUNT-1, FX_MODE_STATIC, BLACK, 1000, NO_OPTIONS);
  ws2812fx_p.start();
  ws2812fx_p.service();
```

SamueldaCostaAraujoNunes, 2020-09-15:
Thank you!! With your help I managed to finish my project!
https://github.com/SamueldaCostaAraujoNunes/My-Strip-Led-With-Google-Home-And-Blynk

closed 2020-09-15 by SamueldaCostaAraujoNunes