kitesurfer1404/WS2812FX #256: lack of documentation

AliTavak, 2020-10-01:
Dear friends

Thanks for great library. I'm using this library for about 2 years. I've tried many different configuration using segments, but it seems there is a great lake of documentation about many useful functions and variables inside the lib. I didn't find any using guide or documents except "readme" provided in main Github's page.
For example I didn't find any relative information about "isCycle" or "setCycle" functions or about "frames" except one usage in "ws2812fx_segment_sequence" example.

It will be great if someone can update readme or add some documentation.

Thanks a lot.
Sincerely
Ali

moose4lord, 2020-10-05:
HI Ali, there is a User Guide in the "extras" folder, but it's geared more for beginners. I agree, there should be more documentation of the lib's more advanced features. I'll add that to my to-do list.

moose4lord, 2021-08-30:
I have started working on a W2812FX API document to go along with the Users Guide. It's include in the extras folder of v1.3.4. Still much to do, but it's a start.

closed 2021-08-30 by moose4lord