kitesurfer1404/WS2812FX #257: setPixel in certain segment whilst other segments keep running

jokke009, 2020-10-12:
Hi, 
Love the library, compliments to the designer(s). 

Just needed some pointers for the following design: 
-is it possible to set a pixel (manually, lets say on a callback from serial input), whilst the other segments continue their effect mode. 

Do I first need to take the segment (which I wish to set pixel ) out of effect mode? Like set it to static? Or do i need to reconfigure that setgment ? Exclude the pixels (segment) from ws2812fx and use the strip.SetPixel directly. 

The idea seems simple, yet I'm unsure the best way to accomplish this. 

Thanks for any advice.

tobi01001, 2020-10-12:
Hi,

It certainly depends on what you would like to achieve but I think the one
good option (without digging too much into the library itself) is to use a
"customShow()" function in which you could set your individual pixel(s)
just before writing the data to the Leds.

Regards,
Toby

moose4lord, 2020-10-12:
You can dynamically adjust the segments to accomplish what you're after, but an unfortunate side effect of this technique is the segment animation resets every time setSegment() is executed. So the transition wouldn't be smooth. I think Toby has the right idea. Create a customShow() function and do the pixel manipulation outside of the libraries domain. As long as you're just doing simple, static manipulation of the LEDs, this should work well.

jokke009, 2020-10-12:
Thanks for the feedback. 
I haven't been able to fully comprehend the customShow concept yet. I was however trying to solve it with the 'customMode' method. 
Up until now it seems to work. 

For those interested: 
```
//call back
  switch (buffer[0])
            {
            case 4 :
            strandPixelPosition = buffer[1];
            rRaw = buffer[2];
            gRaw = buffer[3];
            bRaw = buffer[4];
            break;
}

//my custom function.
uint16_t raw(void) { //  drive the pixel in a strand with raw byte data
  WS2812FX::Segment* seg = ws2812fx.getSegment(); // get the current segment
  ws2812fx.setPixelColor(seg->start + strandPixelPosition, rRaw, gRaw, bRaw);
  return seg->speed; // return the delay until the next animation step (in msec)
}

// then in setup
uint8_t rawMode  = ws2812fx.setCustomMode(F("Raw"), raw);

// and ofcoarse the strand: 
ws2812fx.setSegment(2, 33, 41, rawMode, RED, 50, false);

```
Not quite sure what I should set the speed to: 50? I'm guessing setting this value to high will cause callbacks to go unupdated.

closed 2020-10-12 by jokke009