kitesurfer1404/WS2812FX #258: save last program

Tschurl, 2020-10-23:
Hi, this ws2812fx is awesome!! It works really great. 
If I want to start the esp8266 automatic it will always start with "static", "yellow". Is there a way to save the last program, color and speed used? Is there also a way to stop the light (switch it off like changing the program)?

kind regards 👍

SamueldaCostaAraujoNunes, 2020-10-23:
Olá @Tschurl, olha você pode usar a biblioteca SPIFFS, para gravar algumas informações na memoria do ESP8266, e no setup você recupera as informações e joga para a fita de led. 
A docs: https://tttapa.github.io/ESP8266/Chap11%20-%20SPIFFS.html

SamueldaCostaAraujoNunes, 2020-10-23:
Quanto a segunda questão, você pode usar os métodos ```ws2812fx.start()``` para ligar e o ```ws2812fx.stop()``` para desligar. Para pausar ou continuar um efeito da fita, pode usar ```ws2812fx.resume()``` ou ```ws2812fx.pause()```, respectivamente.

moose4lord, 2020-10-24:
The **ws2812_segments_web** example sketch has code to save/load the state of the strip using EEPROM. Look at the saveToEEPROM() and restoreFromEEPROM() functions.

Tschurl, 2020-10-29:
Thank you all, finally everything works great (switch on / off and save the last state). 
Now (I think it will be the last) another question: how and where do I implement the custom codes? What do I have to do to get to the custom codes? If possible, a little example would be useful for me - the section of the code where I have to implement this.

moose4lord, 2020-10-29:
If you're referring to custom effects, the ws2812fx_custom_effect sketch is an example of creating your own custom effect. The ws2812fx_custom_effect2 sketch is an example of using the pre-made custom effects found in the custom folder.

Tschurl, 2020-10-29:
@moose4lord In your programmed webinterface there are 8 buttons for custom effects. How do I connect the buttons from the webinterface with the eight custom effects?

This is the code, that sets the mode
`    if(server.argName(i) == "m") {
      uint8_t tmp = (uint8_t) strtol(server.arg(i).c_str(), NULL, 10);
      uint8_t new_mode = sizeof(myModes) > 0 ? myModes[tmp % sizeof(myModes)] : tmp % ws2812fx.getModeCount();
      ws2812fx.setMode(new_mode);
      Serial.print("mode is "); Serial.println(ws2812fx.getModeName(ws2812fx.getMode()));`

but where do I code the custom codes? Is this a "void" in the code of the esp8266_webinterface.ino file? Is this like 
`void CUSTOM_EFFECT_1(){
....
....
}`

kind regards

moose4lord, 2020-10-30:
You first need to include the code for the custom effect near the top of your sketch. (See the WS2812FX libraries src/custom folder for a list of pre-made custom effects.) For example, here's the line for using the Dual Larson custom effect:
```c++
#include "custom/DualLarson.h"
```
Then, near the top of the setup() function, use setCustomMode() to reference the custom code and give it a name:
```c++
void setup(){
  Serial.begin(115200);
  delay(500);
  Serial.println("\n\nStarting...");

  // add the custom effect
  ws2812fx.setCustomMode(F("Dual Larson"), dualLarson);
  
  modes.reserve(5000);
  modes_setup();

```
Make sure to call the ```setCustomEffect()``` function **before** the call to ```modes_setup()```, otherwise the custom effect will not show in the web interface.

That's it. You should see "Dual Larson" as one of the eight custom effects at the bottom of the web interface's effects list. You can repeat this seven more times for a total of eight custom effects.

Tschurl, 2020-10-30:
UUuarrrrrrghhhh!! I did exactly what you wrote me, but I get the following error:
`esp8266_webinterface:58:33: fatal error: custom/weihnachten2.h: No such file or directory
 #include "custom/weihnachten2.h"
                                 ^
compilation terminated.
exit status 1
custom/weihnachten2.h: No such file or directory
`

I implemented this 
`  ws2812fx.setCustomMode(F("C7 twinkleFox"), twinkleFox);
  ws2812fx.setCustomMode(F("C8 Christmas"), Weihnachten2);`

and my custom file is in the \custom\weihanchten2.h located with this 

`
#ifndef weihnachten2_h
#define weihnachten2_h

#include <WS2812FX.h>

extern WS2812FX ws2812fx;

uint16_t weihnachten2(void) {
  for (uint16_t j=0;j<12;j++){
   for (uint16_t i=0;i<LED_COUNT+12;i+=12){
	ws2812fx.setPixelColor(j+i, RED);
	ws2812fx.setPixelColor(j+i-1, RED);
	ws2812fx.setPixelColor(j+i-2, RED);
	ws2812fx.setPixelColor(j+i-3, RED);
	ws2812fx.setPixelColor(j+i-4, WHITE);
	ws2812fx.setPixelColor(j+i-5, WHITE);
	ws2812fx.setPixelColor(j+i-6, WHITE);
	ws2812fx.setPixelColor(j+i-7, WHITE);
	ws2812fx.setPixelColor(j+i-8, GREEN);
	ws2812fx.setPixelColor(j+i-9, GREEN);
	ws2812fx.setPixelColor(j+i-10, GREEN);
	ws2812fx.setPixelColor(j+i-11, GREEN);
   }
   delay(ws2812fx.getSpeed);
  }
}

#endif
`

And at least I just open the BlockDissolve custom file and deleted its content and pasted my content - and uaargh it made the dissolve????? I don't know why?!

Tschurl, 2020-10-30:
BTW the implemented custom effects work great

moose4lord, 2020-10-30:
There's a couple things wrong with your implementation.

First, you should not be creating your custom effects in the WS2812FX's custom folder. If the library is ever updated, all your hard work will be gone. Instead create a new tab in the Arduino IDE and put your custom effect code there.

Second, custom effects work on LED segments, so you should use the segment's start and stop LED index in your "for" loop that sets the LED colors. Since you seem to only use one segment, this isn't strictly necessary, but it's a good programming practice.
```c++
  WS2812FX::Segment* seg = ws2812fx.getSegment();
  
  for (uint16_t i=seg->start; i<=seg->stop - 12 + 1; i+=12) {
```

Third, your custom effect must return an unsigned 16-bit integer. The easiest thing to do is just return the segment's speed parameter.
```c++
return seg->speed;
```

[image]

Tschurl, 2020-10-31:
[image]

Tschurl, 2020-10-31:
[image]

But the file is in the correct directory. I don't know what to do now :(

moose4lord, 2020-11-01:
It doesn't look like your folder structure is correct. The Arduino IDE isn't great about handling subfolder, so to get started I would not use subfolders. Your folder structure should look like this:
```
Documents
    Arduino
        esp8266_webinterface
            esp8266_webinterface.ino
            main.js.cpp
            index.html.cpp
            weihnachten2.h
        libraries
            WS2812FX
```
And include your custom effect like so:
```c++
#include "weihnachten2.h"
```
The Arduino IDE expects this folder layout and you'll be able to edit and compile your sketches.

After you get this working, and start creating many more custom effects, you can organize them into subfolders like this:
```
Documents
    Arduino
        esp8266_webinterface
            esp8266_webinterface.ino
            main.js.cpp
            index.html.cpp
            src
                mycustom
                    weihnachten2.h
                    weihnachten3.h
```
And include your custom effects like so:
```c++
#include "src/mycustom/weihnachten3.h"
```
Be warned the Arduino IDE will compile your sketch with this folder structure, BUT you will not be able to edit your custom effect code from within the IDE. The IDE expects the files to be in a flat folder structure, so simply won't show the files in the src subfolder. You would need to use Notepad or some other text editor to modify your custom effect code.

closed 2021-03-01 by moose4lord