kitesurfer1404/WS2812FX #264: Create a new WS2812FX Object for new led settings

Miky94x, 2020-11-23:
Hi, is possible to create a new WS2812FX object at runtime?

i've tried this way, when i receive serial data, using a switch case i select between two or more led configuration but the strip respond only at the first ws2812fx object change with some problem (some blink random), on the second try the arduino don't respond to the serial and all the leds turn off

code:
`    switch (LEDTYPE) {
      case 0:
        ws2812fx.stop();
        ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_RGB + NEO_KHZ800); //NEO_RGB 
        ws2812fx.init();
        ws2812fx.setBrightness(BRIGHTNESS);
        ws2812fx.setSpeed(SPEED);
        ws2812fx.setMode(MODE);
        ws2812fx.setColor(Red, Green, Blue); //CYAN
        ws2812fx.start();
        break;
      case 1:
        ws2812fx.stop();
        ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800); //NEO_GRB 
        ws2812fx.init();
        ws2812fx.setBrightness(BRIGHTNESS);
        ws2812fx.setSpeed(SPEED);
        ws2812fx.setMode(MODE);
        ws2812fx.setColor(Red, Green, Blue); //CYAN
        ws2812fx.start();
        break;
    }`

is possible to change the led type at runtime?
thanks

moose4lord, 2020-11-23:
The Adafruit_Neopixel lib provides the updateType() function to change the LED type. Something like this:
```c++
ws2812fx.updateType(NEO_GRB + NEO_KHZ800);
```

Miky94x, 2020-11-24:
I've read all the neopixel function but as i see without the needed attention
Thanks and have a good day

closed 2020-11-24 by Miky94x