kitesurfer1404/WS2812FX #268: LED Start with an offset

Ashram56, 2020-12-10:
Good morning,

I'm trying to write a code that would have two "comets" chasing each other on a led ring. To that end I have created two segments, but I have not seen yet how to have the second segment start it's animation with an offset (ie half of the ring).

Is this possible ?

Regards

moose4lord, 2020-12-11:
If I'm understanding what you're trying to do, you shouldn't need to start the second comet with an offset. Assigning each half of the ring to a segment should be enough. Something like this:
```c++
ws2812fx.setSegment(0, 0,           LED_COUNT/2-1, FX_MODE_COMET, RED, 5000, NO_OPTIONS);
ws2812fx.setSegment(1, LED_COUNT/2, LED_COUNT-1,   FX_MODE_COMET, RED, 5000, NO_OPTIONS);
```

Ashram56, 2020-12-11:
Thanks, that indeed did the trick.

Now I have another issue linked to segment management.

I have the following segments defined

"
ws2812fx.setSegment(0,0,LED_COUNT-1,DEFAULT_MODE,WHITE,DEFAULT_SPEED,true);
ws2812fx.setSegment(1,0,LED_COUNT-1,DEFAULT_MODE,WHITE,DEFAULT_SPEED,false);

ws2812fx.setIdleSegment(2,0,LED_COUNT/2-1,DEFAULT_MODE,RED,DEFAULT_SPEED,true);
ws2812fx.setIdleSegment(3,LED_COUNT/2,LED_COUNT-1,DEFAULT_MODE,RED,DEFAULT_SPEED,true);

ws2812fx.setIdleSegment(4,0,LED_COUNT/2-1,DEFAULT_MODE,BLUE,DEFAULT_SPEED,false);
ws2812fx.setIdleSegment(5,LED_COUNT/2,LED_COUNT-1,DEFAULT_MODE,BLUE,DEFAULT_SPEED,false);
"

Only one pair can be active at any time.
First pair is fully working, no issue

Second pair and third pair however are causing problem:
In this code, the second and third pair should not be rotating in the same direction, correct ?

However when I activate them using a code like this:

if (ws2812fx.isActiveSegment(2))
						{
						ws2812fx.swapActiveSegment(2,4);
						ws2812fx.swapActiveSegment(3,5);
						}

It works fine with regards to one comet chasing the other, with the correct color, however the orientation statement does not have any effect (it's always the same direction).

Did I miss something with regards to segment definition ?

moose4lord, 2020-12-11:
Hmmm...interesting. There is no setIdleSegment() function call that accepts a boolean value as the last parameter. These are the two allowed function signatures:
```c++
setIdleSegment(uint8_t n, uint16_t start, uint16_t stop, uint8_t mode, uint32_t color,          uint16_t speed, uint8_t options)
setIdleSegment(uint8_t n, uint16_t start, uint16_t stop, uint8_t mode, const uint32_t colors[], uint16_t speed, uint8_t options)
```
It's odd that the compiler doesn't throw an error when you use true or false as the last parameter. Weird.

Ashram56, 2020-12-30:
I'm using Sloeber Arduino, so that might explains.
Note that I had not realized reading the documentation, that true/false only applied to the SetSegment function and not SetIdleSegment

closed 2021-03-01 by moose4lord