kitesurfer1404/WS2812FX #269: SCROLLING RAINBOW TEXT

Elpapu23, 2020-12-12:
Hi, i have a question. How can i make a code to run a text in rainbow mode and scrolling it? Sorry for my bad english . Anyway, Can you help me to do this?

moose4lord, 2020-12-12:
You have the LEDs arranged in a matrix, correct? How many rows and columns are your matrix?

Elpapu23, 2020-12-13:
i have a neopixel 32x8 panel

moose4lord, 2020-12-13:
Have you looked at the [Adafruit Neomatrix](https://learn.adafruit.com/adafruit-neopixel-uberguide/neomatrix-library) library? That's a better fit for what you're trying to do. The **matrixTest** example sketch is a good staring point for a scrolling text project.

Elpapu23, 2020-12-13:
yes, i already tried it, but i don't know how to do the rainbow effect in that library. can you say to me how can i do that?

moose4lord, 2020-12-13:
That depends on what you mean by "rainbow effect". I can think of two rainbow effects for your project:

1) As the text scrolls across the matrix, the letters sweep across the range of rainbow colors. Each LED displays the same color for each frame of animation, but that color is continually changing.

2) The left side of the matrix is red, the right side of the matrix is violet, and the LEDs in the middle of the matrix are a blend of colors ordered like the colors of the rainbow. So each LED displays a different color for each frame of animation, but the display is always showing red on the left, blue on the right, and a rainbow blend in the middle.

Hope that makes sense.

The first scenario isn't too difficult. Here's my attempt:
```c++
Adafruit_NeoMatrix matrix = Adafruit_NeoMatrix(5, 8, PIN,
  NEO_MATRIX_TOP     + NEO_MATRIX_RIGHT +
  NEO_MATRIX_COLUMNS + NEO_MATRIX_PROGRESSIVE,
  NEO_GRB            + NEO_KHZ800);

int16_t x = matrix.width();
uint8_t color_index = 0;
char text[] = "Howdy"; // the text to display
int scroll_limit = sizeof(text) * 6; // each text character is 6 pixels wide

void setup() {
  matrix.begin();
  matrix.setTextWrap(false);
  matrix.setBrightness(40);
  matrix.setTextColor(matrix.Color(255, 0, 0)); // default color is red
}

void loop() {
  matrix.fillScreen(0);
  matrix.setCursor(x, 0);
  matrix.print(text);

  if(--x < -scroll_limit) {
    x = matrix.width();
    color_index = 0; // reset the rainbow color index
  }
  matrix.setTextColor(color_wheel(color_index));
  color_index += max(1, 256 / scroll_limit);
  matrix.show();
  delay(100);
}

uint16_t color_wheel(uint8_t pos) {
  pos = 255 - pos;
  if(pos < 85) {
//  return ((uint32_t)(255 - pos * 3) << 16) | ((uint32_t)(0) << 8) | (pos * 3);
    return matrix.Color((uint16_t)(255 - pos * 3), 0, (pos * 3));
  } else if(pos < 170) {
    pos -= 85;
//  return ((uint32_t)(0) << 16) | ((uint32_t)(pos * 3) << 8) | (255 - pos * 3);
    return matrix.Color(0, (uint32_t)(pos * 3), (255 - pos * 3));
  } else {
    pos -= 170;
//  return ((uint32_t)(pos * 3) << 16) | ((uint32_t)(255 - pos * 3) << 8) | (0);
    return matrix.Color((uint16_t)(pos * 3), (uint32_t)(255 - pos * 3), 0);
  }
}
```
It borrows the _color_wheel()_ function from the WS2812FX library to produce all the colors of the rainbow. Then it's a matter of using matrix.setTextColor() to cycle through the colors.

Note, I don't have a matrix of LEDs to try this on, but I think it will get you close.

closed 2021-03-01 by moose4lord