kitesurfer1404/WS2812FX #272: External trigger/Sound to light interfere with webserver?

charitha95, 2020-12-28:
First of all, let me thank you for this amazing library which saves our time a lot 😇.

The issue um facing is really strange for me.  When I use ` ws2812fx.trigger();` with a web server, the server stop working. 

So I created a small sketch to re-produce the issue. (code looks a little big but you just have to change the pins and wifi credentials and hit the local wifi IP which displays on the serial monitor, then the web page will serve).

Scenario:
Basically, it has 3 buttons. The first two buttons to change the `FX_MODE` via connecting to the server and getting a response back, which works.

3rd button which is "Sound to light", will change the effect but the server will stop working after the execution. It will never get a response back as well.

Any help would be much appreciated. I have been stuck in this for about 2 days and still, I don't have any clue  🤷‍♂️😓😓.

Please replace your pins and wifi credentials when testing.

```
#include <WS2812FX.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

// Replace with your pins
#define LED_COUNT 15
#define LED_PIN 5
#define ANALOG_PIN A0

// Replace with your network credentials
const char *ssid = "***";
const char *password = "***";

#define ANALOG_THRESHOLD 100
#define TIMER_MS 3000

bool isRhythm = false;
unsigned long last_trigger = 0;
unsigned long now = 0;

ESP8266WebServer server(80);
String webSite = "";

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup(void)
{
  Serial.begin(115200);
  webSite += "<meta http-equiv='refresh' content='3'><title>ESP8266 Demo</title>\n";
  webSite += "<style>body{ background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }</style>\n";
  webSite += "<h2 style=\"text-align: left;\">Test server</h2>\n";
  webSite += "<h1>ESP8266 Web Server</h1><p>Rainbow <a href=\"rainbow\"><button>Change</button></a></p>\n";
  webSite += "<p>Color sweep <a href=\"sweep\"><button>Change</button></a></p>\n";
  webSite += "<p>Sound to light <a href=\"soundToLight\"><button>Change</button></a></p>\n";

  ws2812fx.init();
  ws2812fx.setBrightness(100);
  ws2812fx.setSpeed(240);
  ws2812fx.setColor(BLUE);
  ws2812fx.setMode(FX_MODE_BLINK);
  ws2812fx.start();

  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", []() {
    server.send(200, "text/html", webSite);
  });

  server.on("/soundToLight", []() {
    ws2812fx.stop();
    isRhythm = true;
    ws2812fx.setMode(FX_MODE_MULTI_DYNAMIC);
    server.send(200, "text/html", webSite);
  });

  server.on("/rainbow", []() {
    ws2812fx.setMode(FX_MODE_RAINBOW_CYCLE);
    server.send(200, "text/html", webSite);
  });

  server.on("/sweep", []() {
    ws2812fx.setMode(FX_MODE_COLOR_SWEEP_RANDOM);
    server.send(200, "text/html", webSite);
  });

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void)
{

  server.handleClient();
  ws2812fx.service();

  if (isRhythm)
  {
    now = millis();
    if (now - last_trigger > TIMER_MS)
    {
      ws2812fx.trigger();
      last_trigger = now;
    }
    if (analogRead(ANALOG_PIN) > ANALOG_THRESHOLD)
    {
      ws2812fx.trigger();
    }
  }
}

```

moose4lord, 2020-12-28:
Your problem lies with the ```analogRead(ANALOG_PIN)``` statement. The ESP8266's ADC is quite slow and continually sampling the analog pin in the loop() function will muck with the WiFi. Try adding a timer delay so that AnalogRead() only runs every 10ms.

charitha95, 2020-12-28:
delaying by 5ms did the trick. Thank you.

closed 2020-12-28 by charitha95