kitesurfer1404/WS2812FX #273: Multiple strip problem

Miky94x, 2020-12-29:
Hi, i have a problem when trying to use strips on multiple pin on an arduino uno, if i use more then 3 pin or i use too much led for each strip all the animation stop, i think it is for too much ram usage, but i need to use all the 6 pwm pin 3, 5, 6, 9, 10, 11, and strip of different led number

this is a test i've made

```C++
#include <WS2812FX.h>

#define LED_COUNT 50  // number of LEDs on the strip

WS2812FX ws2812fx1 = WS2812FX(LED_COUNT, 3, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx2 = WS2812FX(LED_COUNT, 5, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx3 = WS2812FX(LED_COUNT, 6, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  delay(500);

  ws2812fx1.init();
  ws2812fx2.init();
  ws2812fx3.init();

  // parameters: index, start, stop, mode, color, speed, reverse
  ws2812fx1.setSegment(0,  0, LED_COUNT - 1, FX_MODE_BLINK, RED,   1000, NO_OPTIONS);
  ws2812fx2.setSegment(0,  0, LED_COUNT - 1, FX_MODE_BLINK, GREEN, 1000, NO_OPTIONS);
  ws2812fx3.setSegment(0,  0, LED_COUNT - 1, FX_MODE_BLINK, BLUE, 1000, NO_OPTIONS);

  ws2812fx1.start();
  ws2812fx2.start();
  ws2812fx3.start();
}

void loop() {
  ws2812fx1.service();
  ws2812fx2.service();
  ws2812fx3.service();
}
```

there is a code that don't stop the execution or fill the ram using this library?
have a good day

moose4lord, 2020-12-29:
You are correct, your sketch is using too much RAM. If you are only going to use one segment for each LED strip, you can reduce the amount of RAM that each WS2812FX instance uses by specifying the maximum number of active segments. Like so:
```c++
WS2812FX ws2812fx1 = WS2812FX(LED_COUNT, 3, NEO_GRB + NEO_KHZ800,1 ,1); // allow only one active segment
```
I was able to get 6 strips running on my Arduino Nano with this:
```c++
WS2812FX ws2812fx1 = WS2812FX(LED_COUNT, 3, NEO_GRB + NEO_KHZ800,1 ,1);
WS2812FX ws2812fx2 = WS2812FX(LED_COUNT, 5, NEO_GRB + NEO_KHZ800,1 ,1);
WS2812FX ws2812fx3 = WS2812FX(LED_COUNT, 6, NEO_GRB + NEO_KHZ800,1 ,1);
WS2812FX ws2812fx4 = WS2812FX(LED_COUNT, 9, NEO_GRB + NEO_KHZ800,1 ,1);
WS2812FX ws2812fx5 = WS2812FX(LED_COUNT,10, NEO_GRB + NEO_KHZ800,1 ,1);
WS2812FX ws2812fx6 = WS2812FX(LED_COUNT,11, NEO_GRB + NEO_KHZ800,1 ,1);
```

Miky94x, 2021-01-01:
ok thanks but unfortunately i need to have active effects on all outputs

I am thinking of switching to esp8266 or esp32, do you think that in this way I can drive several outputs at the same time? and with how many LEDs for each output? pretending that it is rgbw led with the heaviest effect for the ram
i don't have any of that board in this moment to try it

moose4lord, 2021-01-01:
With my code above you can have active, and independent,  effects on all outputs. You just can't partition each LED strip into logical segments, and have different effects in each segment.

I have used ESP8266s and ESP32s. They both have more than enough RAM for hundreds of LEDs and work very well. I would recommend starting with a NodeMCU version of the ESP8266. They are inexpensive and easy to program.

Miky94x, 2021-01-01:
thank you very much for your help, I'll explain the project better
at this moment I made a software that looks for arduinos in the usb ports with the sketch loaded above, when it finds them it gives me the possibility to change any aspect of the connected strips, the only problem is that an atmega328 or atmega32u4 cannot support many leds on multiple outputs

so I need to have a different effect with strips of different lengths on each output, the code is already written and working, I just need a microprocessor that can support the amount of work

moose4lord, 2021-01-01:
I see. A NodeMCU should be able to handle your project's requirements. Just remember an ESP8266 is a 3.3V device, so you should use a level shifter (like a 74HCT245) to drive 5V RGB LEDs.

closed 2021-03-01 by moose4lord