kitesurfer1404/WS2812FX #278: No issue.. code request / custom button function

d4rr3n87, 2021-02-25:
Hi.. I have been searching for a way to add audio reactive l.e.d mode to a custom button on the user ws2812fx control web interface for the "esp8266_webinterface"  found in examples. 
So the user could use the web interface to select the pre defined button modes (such as.. static.. rainbow.. strobe.. running random etc etc) found when opening the ws2812fx control web interface and one of the desired custom buttons (custom 0 through to custom 7) to enable the led's to respond to audio.. ideally have the leds respond to audio playback via a pc over the wifi connection although microphone input is an option..  

This would be a great addition to the kitesurfer1404/ws2812fx release..

Many thanks for any replys, suggestions, info or code uploaded/updated

moose4lord, 2021-02-26:
Getting WS2812FX to respond to audio streaming over WiFi would be quite a feat. There are sketches for steaming mp3s to an ESP (see @earlephilhower's amazing **ESP8266Audio** library for examples), but I could only get them to work reliably at very low bit rates.

Adding audio reactivity with a microphone is much easier. There's lots of audio detector gadgets (see [this article](https://randomnerdtutorials.com/guide-for-microphone-sound-sensor-with-arduino/)) that'll do the trick. See the WS2812FX _external_trigger_ example sketch.

d4rr3n87, 2021-03-01:
Impressed with speedy response and really appreciate the time taken to respond... that's awesome I'll do some digging around.. while I'm doing some digging 1 question while I'm looking into the above.. I would be doing some work to using the buttons 0 to 7 on the web interface  but I can not find the lines where I would input the commands that would take place when those buttons 0 to 7 are triggered.. I could not even find the lines to edit the function, speed ,color etc of the predefined buttons such as static, strobe, chase etc etc 😂😂

P.s. close on reply comment or I'll close in around a week

moose4lord, 2021-03-01:
@d4rr3n87, I got an email with a followup question, but I don't see it in this thread. Strange. Did you delete your followup question?

moose4lord, 2021-04-18:
Just FYI, the latest version of WS2812FX (v1.3.3) includes an example sketch that demos audio reactivity using a microphone.

defnotkevin92, 2021-08-12:
Is there any way to assign the audio reactive sketch to the "Custom 0" button in the web interface? I've combed through the issues and I see that you can create a custom effect using the source effects provided in the library  and assign it to a custom button, and I also see that there is an audio sketch, but I havent seen exactly how to assign the audio reactivity sketch to a custom button in the interface.

moose4lord, 2021-08-13:
It turns out running an audio responsive program with WiFi enabled (for a web interface) is not trivial. The problem lies with the ESP8266's WiFi subsystem, which periodically uses the on-chip ADC to fine tune the radio's RF power output. So if you use the ADC to constantly sample the microphone, the WiFi subsystem will become unstable and eventually crash. See esp8266/Arduino core issue [#1634](https://github.com/esp8266/Arduino/issues/1634).

The secret is to not constantly sample the A0 pin, but rather sample it at a regular interval (say every 5ms). This allows some time for the WiFi subsystem to use the ADC and hopefully avoid contention for that resource.

I took a stab at it by merging the **esp82666_webinterface** and **ws2812fx_audio_reactive** example sketches, tweaked the audio sampling code, and added an Auxiliary button to the web interface to turn audio reactivity on and off. That should get you started.

[audio_reactive_webinterface.zip](https://github.com/kitesurfer1404/WS2812FX/files/6984608/audio_reactive_webinterface.zip)

closed 2021-11-19 by moose4lord