kitesurfer1404/WS2812FX #279: Current limiting with DMA?

sprior, 2021-03-13:
I've been using DMA to eliminate the ESP8266 reset issue and my 93 Neopixel Christmas tree has been very stable for a couple of years.  Now I'm looking to play with some neopixel matrix boards and want to incorporate current limiting.

First question: is the ws2812fx.intensitySum() very "expensive" to run in terms of performance?

Second question: The ws2812fx_limit_current example doesn't use DMA.  Do the Adafruit library calls still work when using a DMA based customShow()?  To use the NeopixelBus library in this example would I just replace the call to ws2812fx.Adafruit_NeoPixel::show(); with:

if(strip.CanShow()) {
    // copy the WS2812FX pixel data to the NeoPixelBus instance
    memcpy(strip.Pixels(), ws2812fx.getPixels(), strip.PixelsSize());
    strip.Dirty();
    strip.Show();
  }

?

moose4lord, 2021-03-13:
I timed intensitySum() on my ESP8266 and it took about 40 microseconds for a strip of 100 LEDs. That's a tiny fraction of the time allotted to calculating the next frame, so I don't think it puts much of a burden on the processor.

Replacing `ws2812fx.Adafruit_NeoPixel::show();` with the code snippet you show from the DMA example sketch should work OK.

sprior, 2021-03-13:
OK thanks, I've got it basically working.

Unless I'm missing something it looks like the code can only reduce the brightness when it exceeds the max_current, it can't restore the original set brightness when the strip is no longer exceeding the threshold, because the original set brightness isn't stored anywhere.  I'm thinking if you set the brightness, set a mode like static, then changed the mode to larson scanner without resetting the brightness.

moose4lord, 2021-03-14:
You're right. The ws2812fx_limit_current example sketch adjusts the overall brightness downward until the maximum intensity, for a given mode/color, draws MAX_CURRENT amps. If you change the mode or color, I suppose you should reset the brightness to some nominal value and let the code re-adjust the brightness again.

sprior, 2021-03-18:
I guess I'm set for now, but I do have another question.  I just ordered some RGBW rings to play with, assuming that works with ws2812fx at all, how does it work - is the white LED simply ignored, or do the RGB values get mapped across all 4 LEDs per pixel?

moose4lord, 2021-03-18:
WS2812FX supports RGBW LEDs. The extra white LED's intensity is set by adding a fourth byte to color values. So instead of specifying a color with a three byte value (0xRRGGBB), you specify a four byte value (0xWWRRGGBB). Don't forget to instantiate your WS2812FX object using a third parameter that is compatible with WRGB LEDs (like `NEO_WRGB + NEO_KHZ800`).

sprior, 2021-03-18:
Thanks, I look forward to playing with them when they get here.

closed 2021-03-18 by sprior