kitesurfer1404/WS2812FX #282: Dumb question - is there a simple setPixelColor function?

IraSch, 2021-04-13:
I really like this library to use with ESP32 - very straightforward and simple.

But a dumb question - is there a function similar to 

` strip.setPixelColor(2, strip.Color(0, 0, 255));`
  
to just set a neopixel to a specific color - no special effects, just simple pixel commands for ESP32

Thanks.

kitesurfer1404, 2021-04-13:
Yes, you can use the setPixelColor function from the Adafruit NeoPixel library. Just stop all effects, so they don't overwrite your manually set pixels. Or do I get something wrong here?

IraSch, 2021-04-14:
Yes, that was my question. I wanted to confirm that the regular neopixel function could be used before starting to use your library. Your library seems much "cleaner" than alternatives when it comes to running on ESP32. Thank you.

closed 2021-04-14 by IraSch