kitesurfer1404/WS2812FX #285: WS2811 not working with ESP32 RMT example

silardg, 2021-05-20:
Hi!
I've been testing the WS2811 with the esp32 rmt example, but I'm getting random colors.
When I set it to red it will show sometimes blue, cyan, orange ect.

Does this mean that I have to tweak the timing? Can you show me how to, because it doesn't include the 0h,0l,1h,1l parameters, but something combined.

Thanks

moose4lord, 2021-05-20:
As you may have seen, the RMT driver hardcodes the pulse timing in this snippet:
```c++
// timing parameters for WS2812B LEDs. you may need to
// tweek these if you're using a different kind of LED
#define T1_TICKS      250 / RMT_TICK // 250ns
#define T2_TICKS      625 / RMT_TICK // 625ns
#define T3_TICKS      375 / RMT_TICK // 375ns
#define RESET_TICKS 50000 / RMT_TICK // 50us
```
Each clock "tick" is 25ns, so that's the timing resolution you have to work with. The sum of T1, T2 and T3 should be 1.25us, the bit stream pulse width.

The T1, T2 and T3 parameters correspond to the values in this diagram:
[image]

Good luck.

silardg, 2021-05-22:
Will try and get back to you. The funny thing is that other ws2811 work fine, but mine not. It's like they are fake ones.

moose4lord, 2021-05-22:
I don't have any WS2811s. They are 12V parts, right? Are you using a level shifter to drive the 12V Din pin? The [Adafruit Uberguide](https://learn.adafruit.com/adafruit-neopixel-uberguide) has some good tips about driving and decoupling the LED strips, which can fix flakey behavior.

silardg, 2021-05-22:
Hmm, it works now, it's interesting, because I have unscrewed everything and screwed back on. Probably the placement of the LEDs from the powersupply is very important too. Now it works well! 

Will close this!

Thanks!

closed 2021-05-22 by silardg