kitesurfer1404/WS2812FX #287: Effects not working

multiltech, 2021-06-01:
Hello, 
so I am wanting to make a strip of LEDs change the light pattern every 10 seconds without having to interact with the Arduino.
None of the effects is working, only the firework one does. They switch every 10 secs but a few LEDs are lit with colors but nothing is changing its colors. I am using an Arduino Mega and the Arduino IDE (Version 1.8.15). The librarys are all up to date.

`
#include <WS2812FX.h>

#define LED_COUNT 10 //155
#define LED_PIN 12

int a = 0;

const unsigned long Interval = 10000;
unsigned long alteZeit = 0;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() { 
  
  pinMode(12, OUTPUT);
  Serial.begin(9600);
  ws2812fx.init();
  ws2812fx.setBrightness(50);
  ws2812fx.setSpeed(200);
  ws2812fx.setColor(GREEN);
  ws2812fx.setMode(FX_MODE_STATIC);
  ws2812fx.start();
  Serial.println("HELLO WORLD!");
}

void loop() {
  ws2812fx.service();
  unsigned long Zeit = millis();

  if (a>5){
    a = 1;
  }
  
  switch (a) {
    case 1:
      ws2812fx.setMode(FX_MODE_RAINBOW_CYCLE);
      break;
    case 2:
      ws2812fx.setMode(FX_MODE_LARSON_SCANNER);
      break;
    case 3:
      ws2812fx.setMode(FX_MODE_FIREWORKS);
      break;
    case 4:
      ws2812fx.setMode(FX_MODE_THEATER_CHASE_RAINBOW);
      break;
    case 5:
      ws2812fx.setMode(FX_MODE_COLOR_SWEEP_RANDOM);
      break;
  }

  if (Zeit - alteZeit >= Interval){
    a = a+1;
    Serial.println("Wechsel zu Programm ");
    Serial.println(a);
    alteZeit = Zeit;
  }
}`

moose4lord, 2021-06-01:
You are executing the setMode() function for every iteration of the loop() function. You only want to execute setMode() when you want to change the effect. Put your switch block inside the ```if (Zeit - alteZeit >= Interval){``` block.

multiltech, 2021-06-01:
Thanks, worked just fine

closed 2021-06-01 by multiltech