kitesurfer1404/WS2812FX #289: play Effects one after the other

ddinternetagentur, 2021-06-12:
is it possible to play effects one after the other .. like an animation? would like to build an animation for opening and closing my garage.

To do this, of course, every animation would have to stop at some point ?

All White
Fade Out
Fade in RED
Larsonscanner RED
Fade out
Fade In White

or is it too time-consuming with this library?

moose4lord, 2021-06-14:
The WS2812FX library isn't well suited to your project, since it doesn't provide functions to chain effects. It can be done, as mentioned in issue #127, but it's not pretty. You may have better luck, just using the base Adafruit_Neopixel library and implement the effect and state change code yourself.

moose4lord, 2021-08-30:
I have implemented some effect transition code in the latest v1.3.4 release, which allows you to fade from one effect to another. Please see the ws2812fx_transitions example sketch.

closed 2021-11-19 by moose4lord