kitesurfer1404/WS2812FX #292: src/WS2812FX.h:96:31: warning: 'B00000000' is deprecated: use 0b00000000 instead....??

serget2, 2021-09-05:
I am fairly new to Arduino so if I am doin this wrong please inform me. I am trying to make the Disco Helmet but I ran in to quite a few problems here and now I am at wits end. https://www.electromaker.io/project/view/disco-helmet#project-info
First off I found out the hard way that my Arduino Nano is actually a Nano Every, so I needed to flash it with different board and emulator, but when I had Nano selected I did not get these error messages., Here is the full listing:

Arduino:1.8.15 (Windows 10), Board:"Arduino Nano Every, ATMEGA328"

In file included from C:\Users\serge\Documents\Arduino\sketch_sep04b\sketch_sep04b.ino:4:0:

C:\Users\serge\Documents\Arduino\libraries\WS2812FX-master\src/WS2812FX.h: In constructor 'WS2812FX::WS2812FX(uint16_t, uint8_t, neoPixelType, uint8_t, uint8_t)':

C:\Users\serge\Documents\Arduino\libraries\WS2812FX-master\src/WS2812FX.h:96:31: warning: 'B00000000' is deprecated: use 0b00000000 instead [-Wdeprecated-declarations]

 #define NO_OPTIONS   (uint8_t)B00000000

                               ^

C:\Users\serge\Documents\Arduino\libraries\WS2812FX-master\src/WS2812FX.h:373:82: note: in expansion of macro 'NO_OPTIONS'

       setSegment(0, 0, num_leds - 1, DEFAULT_MODE, DEFAULT_COLOR, DEFAULT_SPEED, NO_OPTIONS);

                                                                                  ^~~~~~~~~~

In file included from C:\Users\serge\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/api/ArduinoAPI.h:26:0,

                 from C:\Users\serge\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/Arduino.h:23,

                 from sketch\sketch_sep04b.ino.cpp:1:

C:\Users\serge\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/api/Binary.h:45:3: note: declared here

   B00000000 DEPRECATED(0b00000000) = 0,

   ^~~~~~~~~

De schets gebruikt 20276 bytes (41%)  programma-opslagruimte. Maximum is 49152 bytes.

Globale variabelen gebruiken 448 bytes (7%) van het dynamisch geheugen. Resteren 5696 bytes voor lokale variabelen. Maximum is 6144 bytes.

Er trad een error op gedurende laden van de schets

Even now I get an error during loading, I do not know if this is fixable, or even if you can spare a bit of time helping me out here, but it I am to buy a new Nano, just tell me

moose4lord, 2021-09-05:
Strange. I do not see the warning/error message that you see when I compile for the Nano Every. I downloaded ElectroMaker's sketch and (after fixing a minor typo) it compiles fine for me.

The only odd thing I can see from your compile output is that you installed the WS2812FX library from GitHub instead of using the library manager. (ElectroMaker's instructions say to install WS2812FX from GitHub, although I'm not sure why.) I don't think that this is really an issue, but it couldn't hurt to delete the WS2812FX_master folder from your Documents/Arduino/libraries directory, then install WS2812FX using the Library Manager (just like you did with the Adafruit_Neopixel library). Give that a try and let me know if it helps.

serget2, 2021-09-05:
Woaw didn't expect help on a sunday so fast, so thank you very much, I will try this tomorrow, to see if it fixes it. Good to know that it does work for you, at least i know it is not a code error. I didn't get this error when I tried to write it with Arduino Nano and the atpmegar328p old bootloader selected but it wouldn't load anyway, Also my nano every has a green led that is allways on and an orange led that is allways blinking, if I press the reset nothing changes and they say that it might be an indication of a corrupted bootloader, to fix it I would need another arduino which I do not have at the moment. Also I tried several other from the examples menu like blink en the bare essentials, but even those won't load , so I reccon they are right about the bootloader, any way I can fix that without having to wait for another Arduino?

moose4lord, 2021-09-06:
I have to apologize. After looking at it again, I realized I had compiler warnings turned off (in the Arduino IDE's Preferences panel). Doh! After enabling compiler warnings, I do indeed see the warning message you mentioned. But it's just that...a warning, not an error. The sketch still compiles fine.

So I think your problem is just that the upload isn't working. Double check you're selecting the proper COM port in the IDE. I don't have a Nano Every, so I'm not sure I can help debug an upload problem.

serget2, 2021-09-06:
It happens, too many variables equals too many things going wrong, I have tried several approaches, I only Have 1 comort though, but unlike with others it is just marked as COM1, (Not Arduino.....com 1). Too much cloned crap on the market as well. I will order a few others, see if I can get this one fixed. Again thx for helping, at least thanks to you I know the code is fine, which leaves the hardware as being faulty, I may try another USB cable though just to be sure

moose4lord, 2021-11-19:
This issue has been fixed in the newest version of WS2812FX (v1.3.5).

closed 2021-11-19 by moose4lord