kitesurfer1404/WS2812FX #293: How to set every segment CustomMode

qinzheng, 2021-09-28:
I have 8 segments on 64 leds, every segment has 8 leds, and I want  each segment  has its own custommode, how can I do this? Sorry for my poor English.

moose4lord, 2021-09-28:
Include the custom effects you're interested in:
```c++
#include "custom/BlockDissolve.h"
#include "custom/DualLarson.h"
#include "custom/MultiComet.h"
#include "custom/Oscillate.h"
#include "custom/RainbowLarson.h"
#include "custom/RandomChase.h"
#include "custom/TriFade.h"
#include "custom/Heartbeat.h"
```
Setup each custom effect:
```c++
  uint8_t blockDissolveMode  = ws2812fx.setCustomMode(0, F("Block Dissolve"), blockDissolve);
  uint8_t dualLarsonMode     = ws2812fx.setCustomMode(1, F("Dual Larson"),    dualLarson);
  uint8_t multiCometMode     = ws2812fx.setCustomMode(2, F("Multi Comet"),    multiComet);
  uint8_t oscillateMode      = ws2812fx.setCustomMode(3, F("Oscillate"),      oscillate);
  uint8_t rainbowLarsonMode  = ws2812fx.setCustomMode(4, F("Rainbow Larson"), rainbowLarson);
  uint8_t randomChaseMode    = ws2812fx.setCustomMode(5, F("Random Chase"),   randomChase);
  uint8_t triFadeMode        = ws2812fx.setCustomMode(6, F("TriFade"),        triFade);
  uint8_t heartbeatMode      = ws2812fx.setCustomMode(7, F("Heartbeat"),      heartbeat);
```
Assign each custom effect to a segment:
```c++
  ws2812fx.setSegment(0,  0,  7, blockDissolveMode, RED, 1000);
  ws2812fx.setSegment(1,  8, 15, dualLarsonMode,    RED, 1000);
  ws2812fx.setSegment(2, 16, 23, multiCometMode,    RED, 1000);
  ws2812fx.setSegment(3, 24, 31, oscillateMode,     RED, 1000);
  ws2812fx.setSegment(4, 32, 39, rainbowLarsonMode, RED, 1000);
  ws2812fx.setSegment(5, 40, 47, randomChaseMode,   RED, 1000);
  ws2812fx.setSegment(6, 48, 55, triFadeMode,       RED, 1000);
  ws2812fx.setSegment(7, 56, 63, heartbeatMode,     RED, 1000);
```

qinzheng, 2021-09-28:
@moose4lord  Thank you! I'v tried and it works fine!

closed 2021-09-28 by qinzheng