kitesurfer1404/WS2812FX #298: Increase chasing effects speed on LED count increment

aleksandravojinovic-htec, 2021-11-16:
Hi, I was wondering whether it is possible (and how) to increase the speed on the chasing effects (scan, dual scan, Larson scanner, comment).
What's happening right now is that on the LED count increment the effect speed decreases.

For example, when the number of LEDs is 144, the speed is fine, but the speed decreases significantly when the number of LEDs is incremented to 1440.

moose4lord, 2021-11-16:
I'm not sure I understand. Are you dynamically changing the number of LEDs? If so, and you want to maintain a constant animation speed, you'll need to use the setSpeed() function to change the speed parameter when you change the number of LEDs.
```c++
ws2812fx.setSpeed(1000); // set the speed parameter
```
The scan and Larson effects use this function to calculate the animation speed:

> _animation_speed = speed_parameter / (num_leds * 2)_

So if num_leds=144 and speed_parameter=1000, then animation_speed = 3.47 ms/frame
To get the same animation_speed with 1440 LEDs, you would need to set the speed_parameter to 3.47 * (1440 * 2) = 10000

Make sense?

aleksandravojinovic-htec, 2021-11-16:
Do you mean this function?
[image]

moose4lord, 2021-11-17:
Yes, that's it.

closed 2021-11-17 by aleksandravojinovic-htec