kitesurfer1404/WS2812FX #304: Mode_rainbow_cycle doesn't support reverse option

DominikGrudzinski, 2021-12-03:
I noticed that mode_rainbow_cycle doesn't support reverse option. The fix is very simple:
just replace: (line 926)
`   setPixelColor(_seg->start + i, color);`
to
```
    if(IS_REVERSE) {
          setPixelColor(_seg->start + i, color);
      } else {
          setPixelColor(_seg->stop - i, color);
      }
```

moose4lord, 2021-12-05:
You're right. Thanks for pointing this out. However, I think your solution is backwards. It would have the rainbow direction reversed if REVERSE is _not_ set. Shouldn't it be this:
```c++
if(IS_REVERSE) {
  setPixelColor(_seg->stop - i, color);
} else {
  setPixelColor(_seg->start + i, color);
}
```

DominikGrudzinski, 2021-12-05:
Yes you are right. I just copied the code block from the next available function and didn't notice this error because it works fine (the direction of movement is reverse).
By the way, this means there is the same error in the functions:
`tricolor_chase (row ~ 944)`
and
`mode_running_lights (line ~ 1009)`

moose4lord, 2021-12-05:
Actually, the tricolor_chase and running_lights modes are correct. They do use the code snippet from your first post, but their algorithm works a little different and does produce animations going in the right direction. (I assume "forward" means running an animation so lights move from led.start to led.stop, and "reverse" means moving from led.stop to led.start.)

I looked through the modes and there are three that run backwards: rainbow_cycle, theater_chase_rainbow and rain. So I think I'll use your original code snippet to fix the rainbow_cycle direction, and implement fixes for the other two as well. That way animation direction will be consistent for all the modes.

Thanks for your help with this.

closed 2022-11-10 by moose4lord