kitesurfer1404/WS2812FX #307: BUG - Loose Wifi connection by using "Color Wipe Random"

Eisbaeeer, 2022-01-09:
I will loose Wifi connection to ESP8266 by using the "Color Wipe Random" effect. It takes about 1 minute, then the ESP will not be pingable. It is reproducable.
If you need more debug infos, let me know.

moose4lord, 2022-01-10:
I could not reproduce this issue with the esp8266_webinterface example sketch. Can you show your sketch?

Eisbaeeer, 2022-01-10:
I will try some yield() calls in loop. I think this would help. The source of issue is 100% neopixel or ws2812fx part, because if it is disabled, the CPU is availlable all the time. This issue is also only in some animations. The "no blinking, just plain" working perfect and blinking 50%50% also. I will report. Sorry, the source code of project is not public.

moose4lord, 2022-01-10:
I see. It does sound like you may be overtaxing the processor, so that not enough processing time is available to the WiFi stack. If you are only driving one strip of LEDs, you may want to consider using the esp8266's DMA and I2S hardware to drive the LEDs. It has the advantage that interrupts are not disabled while the LEDs are being updated, leaving the CPU more time to manage WiFi. The disadvantage of using DMA/I2S is that it uses much more memory, which may or may not be a problem with your application

The ws2812fx_dma example sketch shows how to implement DMA/I2S. It borrows the DMA/I2S driver from the NeoPixelBus library.

Good luck.

Eisbaeeer, 2022-01-10:
Yes, I solved the issue by creating a TASK with 10ms. I call the service from this task and all worked fine. Thanks a lot!

closed 2022-01-10 by Eisbaeeer

Eisbaeeer, 2022-01-12:
For other users. You will find something more here: Issue: #159
It seems the ESP8266 arduino core have som wifi issues.