kitesurfer1404/WS2812FX #313: How to lengthen the Comet-Tail?

EherXtrem, 2022-04-07:
I would like to lengthen the tail of the Comet (#44) effect a bit. Is that possible and if so, how?

EherXtrem, 2022-04-07:
Wow ! It works preaty nice ! THX a lot !

Now I will try to discover the options for Fireworks Random (#46) by my own :)

moose4lord, 2022-04-07:
You're welcome. Good luck!

closed 2022-04-07 by moose4lord

EherXtrem, 2022-04-07:
Is there a backgroud color option instead of black in Fireworks Random mode too?

moose4lord, 2022-04-07:
You can use a fade option to get a non-black background color even with the FIREWORKS_RANDOM effect. This works for me:
```c++
  ws2812fx.setSegment(0,  0, LED_COUNT - 1, FX_MODE_FIREWORKS_RANDOM, COLORS(BLUE, DARK(RED)), 1000, FADE_MEDIUM);
```
It gets a little tricky with FIREWORKS_RANDOM (and other effects) if you slow down the fade too much. The background color can easily wash out the effect if the LEDs fade too slowly. For best effect set the brightness to 255 and set the background color to something very dim to produce good contrast.  If you want bigger fireworks, you can combine a FADE and SIZE option like so:
```
ws2812fx.setSegment(0,  0, LED_COUNT - 1, FX_MODE_FIREWORKS_RANDOM, COLORS(BLUE, DARK(RED)), 1000, (uint8_t)(FADE_MEDIUM | SIZE_LARGE));
```

EherXtrem, 2022-04-08:
**Big THX - both code lines work very well. Awesome support :)**