kitesurfer1404/WS2812FX #314: I can't use more WS2812FX

chenmaker, 2022-05-13:
Today I need to use 10 pins but I use two most WS2812FX
If I use the third the  WS2812FX is can not wor
attach my code
[WS2812.zip](https://github.com/kitesurfer1404/WS2812FX/files/8683623/WS2812.zip)
k

moose4lord, 2022-05-13:
Your sketch is constantly updating the segments every time through the main loop(). This does not leave any time to actually run the animation. You should only update the segments when a button change is detected.
Maybe this?
```c++
void loop() {

  bool buttonChanged = bouncer.update(); // call update only once in loop() function
  
  if (buttonChanged == true && bouncer.read() == HIGH) {
    if (count >= 4) {
      count = 0;
    } else if (count <= 3) {
      count++;
    }
  } else if (buttonChanged == false && bouncer.read() == LOW) {
    if (count >= 4) {
      count = 0;
    }
  }

  Serial.println(count);
  if(buttonChanged) { // if button change detected, update the segments
    switch (count) {
      case 0:
        ws2812fx0.setSegment(0, 0, (LED_COUNT0 - 1), 5, COLORS(ORANGE, PURPLE), 2000, false);
        ws2812fx1.setSegment(0, 0, (LED_COUNT1 - 1), 6, COLORS(ORANGE, PURPLE), 2000, false);
.
.
.
```

chenmaker, 2022-05-14:
I removed the button
But the same can not add a third
By the way I use NANO

#include <WS2812FX.h>

#define LED_COUNT0 8
#define LED_COUNT1 10
#define LED_COUNT2 4

#define LED_PIN0 2
#define LED_PIN1 3
#define LED_PIN2 4

//燈數,腳位,參數
WS2812FX ws2812fx0 = WS2812FX(LED_COUNT0, LED_PIN0, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx1 = WS2812FX(LED_COUNT1, LED_PIN1, NEO_GRB + NEO_KHZ800);
//WS2812FX ws2812fx2 = WS2812FX(LED_COUNT2, LED_PIN2, NEO_GRB + NEO_KHZ800);
void setup() {
  //ws2812fx.setSegment(0,0,(LED_COUNT-1),5, COLORS(ORANGE, PURPLE), 2000,false);
  //陣列,起始燈位,結束燈位,模式,顏色,速度,是否反向
  Serial.begin(9600);
  ws2812fx0.init();
  ws2812fx0.setBrightness(100);
  ws2812fx0.setSegment(0, 0, (LED_COUNT0 - 1), 5, COLORS(ORANGE, PURPLE), 2000, false);
  ws2812fx0.start();
  Serial.println("OK1");
  ws2812fx1.init();
  ws2812fx1.setBrightness(100);
  ws2812fx1.setSegment(1, 0, (LED_COUNT1 - 1), 6, COLORS(ORANGE, PURPLE), 1500, false);
  ws2812fx1.start();
  Serial.println("OK2");
  /*ws2812fx2.init();
  ws2812fx2.setBrightness(50);
  ws2812fx2.setSegment(2, 0, (LED_COUNT2 - 1), 8, COLORS(ORANGE, PURPLE), 1000, false);
  ws2812fx2.start();
  Serial.println("OK3");*/
}
void loop() {

 // ws2812fx0.setSegment(0, 0, (LED_COUNT0 - 1), 5, COLORS(ORANGE, PURPLE), 2000, false);
 // Serial.println("OK3");
 // ws2812fx1.setSegment(1, 0, (LED_COUNT1 - 1), 6, COLORS(ORANGE, PURPLE), 1500, false);
 // Serial.println("OK4");
 // ws2812fx2.setSegment(2, 0, (LED_COUNT2 - 1), 8, COLORS(ORANGE, PURPLE), 1000, false);
 // Serial.println("OK5");
  ws2812fx0.service();
  ws2812fx1.service();
  //ws2812fx2.service();
  Serial.println("OK");
}

moose4lord, 2022-05-14:
Ah, I see. The Arduino Nano has a very limited amount of SRAM memory (only 2048 bytes). Trying to use more than two LED strips is causing the sketch to fail, because it is trying to allocate more memory than is available on a Nano. There is a workaround however. If you are only going to program one segment in each LED strip, you can reduce WS2812FX's memory requirements by using this constructor:
WS2812FX ws2812fx0 = WS2812FX(LED_COUNT0, LED_PIN0, NEO_GRB + NEO_KHZ800, 1, 1);

Note the "1,1" at the end. This instructs WS2812FX to only allocate memory for one segment per LED strip. This dramatically reduces the memory used by each strip.

I ran the following code on my Nano and it worked fine, so you should be able to drive at least five strips with your Nano using the above modification.

```c++
#include <WS2812FX.h>

#define LED_COUNT 10

WS2812FX ws2812fx0 = WS2812FX(LED_COUNT,  2, NEO_GRB + NEO_KHZ800, 1, 1);
WS2812FX ws2812fx1 = WS2812FX(LED_COUNT,  3, NEO_GRB + NEO_KHZ800, 1, 1);
WS2812FX ws2812fx2 = WS2812FX(LED_COUNT,  4, NEO_GRB + NEO_KHZ800, 1, 1);
WS2812FX ws2812fx3 = WS2812FX(LED_COUNT,  5, NEO_GRB + NEO_KHZ800, 1, 1);
WS2812FX ws2812fx4 = WS2812FX(LED_COUNT,  6, NEO_GRB + NEO_KHZ800, 1, 1);

void setup() {
  Serial.begin(115200);
  delay(500);

  ws2812fx0.init();
  ws2812fx0.setBrightness(100);
  ws2812fx0.setSegment(0, 0, (LED_COUNT - 1), 5, COLORS(ORANGE, PURPLE), 2000);
  ws2812fx0.start();
  Serial.println("OK0");

  ws2812fx1.init();
  ws2812fx1.setBrightness(100);
  ws2812fx1.setSegment(0, 0, (LED_COUNT - 1), 6, COLORS(ORANGE, PURPLE), 1500);
  ws2812fx1.start();
  Serial.println("OK1");

  ws2812fx2.init();
  ws2812fx2.setBrightness(50);
  ws2812fx2.setSegment(0, 0, (LED_COUNT - 1), 8, COLORS(ORANGE, PURPLE), 1000);
  ws2812fx2.start();
  Serial.println("OK2");

  ws2812fx3.init();
  ws2812fx3.setBrightness(50);
  ws2812fx3.setSegment(0, 0, (LED_COUNT - 1), 8, COLORS(ORANGE, PURPLE), 1000);
  ws2812fx3.start();
  Serial.println("OK3");

  ws2812fx4.init();
  ws2812fx4.setBrightness(50);
  ws2812fx4.setSegment(0, 0, (LED_COUNT - 1), 8, COLORS(ORANGE, PURPLE), 1000);
  ws2812fx4.start();
  Serial.println("OK4");

  Serial.print("Free RAM:"); Serial.println(freeRam());
}

void loop() {
  ws2812fx0.service();
  ws2812fx1.service();
  ws2812fx2.service();
  ws2812fx3.service();
  ws2812fx4.service();
}

int freeRam () {
  extern int __heap_start, *__brkval;
  int v;
  return (int) &v - (__brkval == 0 ? (int) &__heap_start : (int) __brkval);
}
```

chenmaker, 2022-05-15:
Okay, I understand
But I need to control many groups
So I changed to MEGA2560
thanks for your help

closed 2022-05-15 by chenmaker