kitesurfer1404/WS2812FX #315: Any way to use WS2812FX with a character generator and DMA with Neopixelbus?

sprior, 2022-06-28:
Now that I've closed another issue...
Now that I've got my firmware working with various Neopixel devices I'm playing with the idea of running a Neopixel rectangular matrix (8x32 arrangement) with a character generator in one area with a WS2812FX animation in another part.  I'm also using the custom show with DMA technique to avoid wdt reset issues and implement current limiting.  In a perfect world I'd even be able to do scrolling of the character section.

It seems the key to this would involve the memcpy below:

      if(dma.CanShow()) {
         // copy the WS2812FX pixel data to the NeoPixelBus instance
         memcpy(dma.Pixels(), ws2812fx.getPixels(), dma.PixelsSize());
         dma.Dirty();
         dma.Show();
      }

Any ideas?

moose4lord, 2022-06-28:
I don't have a lot of experience with 2D displays, but for me they have been a pain in the keister. The LEDs are typically hardwired in a serpentine pattern, which is probably the LEAST convenient arrangement and never what you want. I have always had to create a mapping function to translate a logical order of LEDs into their physical positions in the strip (which is discussed in issue #284).

For text you should look at the Adafruit_NeoMatrix library. It includes a "matrixtest" example sketch that demonstrates how to create scrolling text. The [Adafruit Uber Guide](https://learn.adafruit.com/adafruit-neopixel-uberguide/neomatrix-library) also has a section on using the library to create text effects.

sprior, 2022-06-28:
I played with the Adafruit_NeoMatrix library last night and got an example running.  I was thinking maybe have it write to a buffer for a subset of the pixels and then do a memcopy for that area after the memcopy I listed above.  The https://github.com/pixelmatix/SmartMatrix library for HUB75 panels uses layers which are then flattened down before output, I might have to think about that as well.

closed 2022-11-10 by moose4lord