kitesurfer1404/WS2812FX #317: While creating hotspot it getting reset and switching off

rahulbaruri, 2022-07-31:
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

#include <WS2812FX.h>

extern const char index_html[];
extern const char main_js[];

#ifndef APSSID
#define APSSID "ESPap"
#define APPSK  "thereisnospoon"
#endif

/* Set these to your desired credentials. */
const char *ssid = APSSID;
const char *password = APPSK;

// QUICKFIX...See https://github.com/esp8266/Arduino/issues/263
#define min(a,b) ((a)<(b)?(a):(b))
#define max(a,b) ((a)>(b)?(a):(b))

#define LED_PIN 2                       // 0 = GPIO0, 2=GPIO2
#define LED_COUNT 300

unsigned long auto_last_change = 0;
unsigned long last_wifi_check_time = 0;

ESP8266WebServer server(80);

/* Just a little test message.  Go to http://192.168.4.1 in a web browser
   connected to this access point to see it.
*/

String modes = "";
uint8_t myModes[] = {}; // *** optionally create a custom list of effect/mode numbers
bool auto_cycle = false;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  Serial.println("\n\nStarting...");

  modes.reserve(5000);
  modes_setup();

  Serial.println("WS2812FX setup");
  ws2812fx.init();
  ws2812fx.setMode(FX_MODE_STATIC);
  ws2812fx.setColor(0xFF5900);
  ws2812fx.setSpeed(1000);
  ws2812fx.setBrightness(128);
  ws2812fx.start();
  
  Serial.print("Configuring access point...");
  /* You can remove the password parameter if you want the AP to be open. */
  WiFi.softAP(ssid, password);

  IPAddress myIP = WiFi.softAPIP();
   Serial.println("HTTP server setup");
  server.on("/", srv_handle_index_html);
  server.on("/main.js", srv_handle_main_js);
  server.on("/modes", srv_handle_modes);
  server.on("/set", srv_handle_set);
  server.onNotFound(srv_handle_not_found);
  server.begin();
  Serial.println("HTTP server started.");

  Serial.println("ready!");
}

void loop() {
   unsigned long now = millis();

  server.handleClient();
  ws2812fx.service();

  if(auto_cycle && (now - auto_last_change > 10000)) { // cycle effect mode every 10 seconds
    uint8_t next_mode = (ws2812fx.getMode() + 1) % ws2812fx.getModeCount();
    if(sizeof(myModes) > 0) { // if custom list of modes exists
      for(uint8_t i=0; i < sizeof(myModes); i++) {
        if(myModes[i] == ws2812fx.getMode()) {
          next_mode = ((i + 1) < sizeof(myModes)) ? myModes[i + 1] : myModes[0];
          break;
        }
      }
    }
    ws2812fx.setMode(next_mode);
    Serial.print("mode is "); Serial.println(ws2812fx.getModeName(ws2812fx.getMode()));
    auto_last_change = now;
  }

}

/*
 * Build <li> string for all modes.
 */
void modes_setup() {
  modes = "";
  uint8_t num_modes = sizeof(myModes) > 0 ? sizeof(myModes) : ws2812fx.getModeCount();
  for(uint8_t i=0; i < num_modes; i++) {
    uint8_t m = sizeof(myModes) > 0 ? myModes[i] : i;
    modes += "<li><a href='#'>";
    modes += ws2812fx.getModeName(m);
    modes += "</a></li>";
  }
}

/* #####################################################
#  Webserver Functions
##################################################### */

void srv_handle_not_found() {
  server.send(404, "text/plain", "File Not Found");
}

void srv_handle_index_html() {
  server.send_P(200,"text/html", index_html);
}

void srv_handle_main_js() {
  server.send_P(200,"application/javascript", main_js);
}

void srv_handle_modes() {
  server.send(200,"text/plain", modes);
}

void srv_handle_set() {
  for (uint8_t i=0; i < server.args(); i++){
    if(server.argName(i) == "c") {
      uint32_t tmp = (uint32_t) strtol(server.arg(i).c_str(), NULL, 10);
      if(tmp <= 0xFFFFFF) {
        ws2812fx.setColor(tmp);
      }
    }

    if(server.argName(i) == "m") {
      uint8_t tmp = (uint8_t) strtol(server.arg(i).c_str(), NULL, 10);
      uint8_t new_mode = sizeof(myModes) > 0 ? myModes[tmp % sizeof(myModes)] : tmp % ws2812fx.getModeCount();
      ws2812fx.setMode(new_mode);
      auto_cycle = false;
      Serial.print("mode is "); Serial.println(ws2812fx.getModeName(ws2812fx.getMode()));
    }

    if(server.argName(i) == "b") {
      if(server.arg(i)[0] == '-') {
        ws2812fx.setBrightness(ws2812fx.getBrightness() * 0.8);
      } else if(server.arg(i)[0] == ' ') {
        ws2812fx.setBrightness(min(max(ws2812fx.getBrightness(), 5) * 1.2, 255));
      } else { // set brightness directly
        uint8_t tmp = (uint8_t) strtol(server.arg(i).c_str(), NULL, 10);
        ws2812fx.setBrightness(tmp);
      }
      Serial.print("brightness is "); Serial.println(ws2812fx.getBrightness());
    }

    if(server.argName(i) == "s") {
      if(server.arg(i)[0] == '-') {
        ws2812fx.setSpeed(max(ws2812fx.getSpeed(), 5) * 1.2);
      } else if(server.arg(i)[0] == ' ') {
        ws2812fx.setSpeed(ws2812fx.getSpeed() * 0.8);
      } else {
        uint16_t tmp = (uint16_t) strtol(server.arg(i).c_str(), NULL, 10);
        ws2812fx.setSpeed(tmp);
      }
      Serial.print("speed is "); Serial.println(ws2812fx.getSpeed());
    }

    if(server.argName(i) == "a") {
      if(server.arg(i)[0] == '-') {
        auto_cycle = false;
      } else {
        auto_cycle = true;
        auto_last_change = 0;
      }
    }
  }
  server.send(200, "text/plain", "OK");
}

moose4lord, 2022-07-31:
I could not reproduce the problem you are seeing. The AP mode seemed to work OK for me.

300 LEDs requires a lot of power. Are you sure your power supply can handle the load?

Attached is the test sketch that worked with my ESP8266 setup.
[sketch_jul31a.zip](https://github.com/kitesurfer1404/WS2812FX/files/9228940/sketch_jul31a.zip)

moose4lord, 2022-07-31:
> Hi @moos4lord Please reaplace code the following function with the function in the "esp8266_webinterface.ino"example WS2812FX library and edit LED_COUNT = 600
> 
> void wifi_setup() {
> 
>   WiFi.softAP("myAP", "12345678");
> 
>   IPAddress myIP = WiFi.softAPIP();
> 
>   Serial.print("myAP IP address: ");
> 
>   Serial.println(myIP);
> 
>   Serial.println();
> 
> }
> 
> and remove in loop code
> 
> if(now - last_wifi_check_time > WIFI_TIMEOUT) { Serial.print("Checking WiFi... "); if(WiFi.status() != WL_CONNECTED) { Serial.println("WiFi connection lost. Reconnecting..."); wifi_setup(); } else { Serial.println("OK"); } last_wifi_check_time = now; }
> 
> After 5 second it is switched off
> 

I made the edits you mention, but I still do not see any problem. The AP mode is working fine for me.

rahulbaruri, 2022-07-31:
> I made the edits you mention, but I still do not see any problem. The AP mode is working fine for me.

I used old version of library to run the code

moose4lord, 2022-07-31:
So the latest version of the library is working for you too?

rahulbaruri, 2022-08-01:
> So the latest version of the library is working for you too?

Nop... Using old Libaries.

moose4lord, 2022-08-01:
Is there some reason you're using old libraries?

rahulbaruri, 2022-08-01:
> Is there some reason you're using old libraries?

The new Libaries of ap code doesn't support esp01  ... After 5 second

moose4lord, 2022-08-02:
I see. Unfortunately, I only have ESP-12E modules in my test setup. If it's only an issue with the older ESP-01 modules, I won't be able to debug it. Sorry.

rahulbaruri, 2022-08-02:
> I see. Unfortunately, I only have ESP-12E modules in my test setup. If it's only an issue with the older ESP-01 modules, I won't be able to debug it. Sorry.

It's ok... I have tested in old library.. It worked

rahulbaruri, 2022-08-11:
> I see. Unfortunately, I only have ESP-12E modules in my test setup. If it's only an issue with the older ESP-01 modules, I won't be able to debug it. Sorry.

Still having problem
[image]

moose4lord, 2022-08-11:
Sorry, I thought you had gotten it working.

I think it would be helpful to know exactly what software/library versions you are using. Can you turn on compiler verbose output in the Arduino IDE preferences, compile you code, and then paste the output in this thread? That'll give me more info about what your development environment is like. Also can you tell me about your hardware setup? Like what brand of ESP-01 you're using and a description of your circuit.

[image]

rahulbaruri, 2022-08-11:
> Sorry, I thought you had gotten it working.
>
> I think it would be helpful to know exactly what software/library versions you are using. Can you turn on compiler verbose output in the Arduino IDE [...]

Using board 'generic' from platform in folder: C:\Users\Angel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2
Using core 'esp8266' from platform in folder: C:\Users\Angel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2
Detecting libraries used...
"C:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/include" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/lwip2/include" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Angel\\AppData\\Local\\Temp\\arduino-sketch-41F7C395D0AB897649448C0AFD3A19F4/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=1 -DESP8266 "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\cores\\esp8266" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\variants\\generic" "C:\\Users\\Angel\\AppData\\Local\\Temp\\arduino-sketch-41F7C395D0AB897649448C0AFD3A19F4\\sketch\\esp8266_webinterface_FINAL_AP.ino.cpp" -o nul
Alternatives for ESP8266WiFi.h: [ESP8266WiFi@1.0]
ResolveLibrary(ESP8266WiFi.h)
  -> candidates: [ESP8266WiFi@1.0]
"C:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/include" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/lwip2/include" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Angel\\AppData\\Local\\Temp\\arduino-sketch-41F7C395D0AB897649448C0AFD3A19F4/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=1 -DESP8266 "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\cores\\esp8266" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\variants\\generic" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\libraries\\ESP8266WiFi\\src" "C:\\Users\\Angel\\AppData\\Local\\Temp\\arduino-sketch-41F7C395D0AB897649448C0AFD3A19F4\\sketch\\esp8266_webinterface_FINAL_AP.ino.cpp" -o nul
Alternatives for ESP8266WebServer.h: [ESP8266WebServer@1.0]
ResolveLibrary(ESP8266WebServer.h)
  -> candidates: [ESP8266WebServer@1.0]
"C:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/include" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/lwip2/include" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Angel\\AppData\\Local\\Temp\\arduino-sketch-41F7C395D0AB897649448C0AFD3A19F4/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=1 -DESP8266 "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\cores\\esp8266" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\variants\\generic" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\libraries\\ESP8266WiFi\\src" "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\libraries\\ESP8266WebServer\\src" "C:\\Users\\Angel\\AppData\\Local\\Temp\\arduino-sketch-41F7C395D0AB897649448C0AFD3A19F4\\sketch\\esp8266_webinterface_FINAL_AP.ino.cpp" -o nul
Alternatives for WS2812FX.h: [WS2812FX-master@1.3.5]
ResolveLibrary(WS2812FX.h)
  -> candidates: [WS2812FX-master@1.3.5]
"C:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Angel\\AppData\\Local\\Arduino15\\packages\\es [... cut ...]
        ws2812fx.setSpeed(ws2812fx.getSpeed() * 0.8);
      } else {
        uint16_t tmp = (uint16_t) strtol(server.arg(i).c_str(), NULL, 10);
        ws2812fx.setSpeed(tmp);
      }
      Serial.print("speed is "); Serial.println(ws2812fx.getSpeed());
    }

    if(server.argName(i) == "a") {
      if(server.arg(i)[0] == '-') {
        auto_cycle = false;
      } else {
        auto_cycle = true;
        auto_last_change = 0;
      }
    }
  }
  server.send(200, "text/plain", "OK");
}

arduino : 2.0.0 rc3  
[image]

moose4lord, 2022-08-11:
Hmmm...that's strange. The "Tools" screen shot you showed looks like you're using the latest version (V3.0.2) of the Arduino ESP8266 core package, but the compile log shows that you're using a very old version (V2.4.2) of the core package. I think somehow you IDE installation has gotten corrupted.

I tried the old, v2.4.2 version of the core package and had weird problems during the sketch upload, so I think that's definitely NOT a good version to use (it's four years old).

At this point I think it would be wise for you to start from scratch and uninstall the Arduino IDE, then reinstall it. The version of the IDE you're using is quite old anyway, so it's probably a good idea to get caught up. Then use the boards manager to install the latest version of the Arduino ESP8266 core package.

Is that doable?

rahulbaruri, 2022-08-12:
Let me uninstall all... Then start from begining

moose4lord, 2022-08-12:
Download v1.8.19 of the IDE from here: [https://www.arduino.cc/en/software](https://www.arduino.cc/en/software)
Then startup the IDE and in the Preferences -> Additional Boards Manager URLs field enter:
```c++
https://arduino.esp8266.com/stable/package_esp8266com_index.json
```
Then in the IDE's Tools -> Boards Manager, search for "ESP8266" and install v3.0.2 of core ESP8266 package.

That will give you the latest versions of the IDE and ESP8266 core package.

From there you can use the IDE's Tools -> Manage Libraries feature to install the latest versions of the Adafruit NeoPixel (v1.10.5) and WS2812FX (v1.4.1) libraries.

Let me know how it goes.

rahulbaruri, 2022-08-12:
> Download v1.8.19 of the IDE from here: https://www.arduino.cc/en/software Then startup the IDE and in the Preferences -> Additional Boards Manager URLs field enter:
>
> ```c++ [...]

Ok... What will be the setting of the following
[image]

moose4lord, 2022-08-12:
That can be a hard question to answer, since it depends entirely on the manufacturer of the ESP-01 you happen to have. For instance, the amount of flash memory can be 512k or 1M or 4M depending on the ESP-01 manufacturer.  Can you take a picture of your ESP-01 and send it? That might help identify the manufacturer. 

As a starting point, I would use the values in your screen shot, except change "Reset Method" = "no dtr (aka ck)". If that doesn't work, we may have to dig a little deeper using the _esptool.py_ tool. Are you using Windows, Mac or Linux as a development system?

rahulbaruri, 2022-08-12:
[image]

moose4lord, 2022-08-12:
Hmmm...not a lot of markings on your ESP-01.

Were you able to successfully upload one of the simpler example sketches (like ws2812fx_segments) to your ESP-01? During the upload, is there a message like this displayed in the output window:
```
Auto-detected Flash size: 1MB
```
I'm just trying to make sure your ESP has enough memory to run the esp8266_webinterface sketch.

rahulbaruri, 2022-08-12:
let me see

rahulbaruri, 2022-08-12:
> Hmmm...not a lot of markings on your ESP-01.
>
> Were you able to successfully upload one of the simpler example sketches (like ws2812fx_segments) to your ESP-01? During the upload, is there a message [...]

Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 295688          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27541   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 
DATA   : 2808  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1312  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 26216 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 327349 bytes (34%) of program storage space. Maximum is 958448 bytes.
Global variables use 30336 bytes (37%) of dynamic memory, leaving 51584 bytes for local variables. Maximum is 81920 bytes.
esptool.py v3.0
Serial port COM9
Connecting.....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: e8:db:84:d8:e5:c1
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 331504 bytes to 236618...
Writing at 0x00000000... (6 %)
Writing at 0x00004000... (13 %)
Writing at 0x00008000... (20 %)
Writing at 0x0000c000... (26 %)
Writing at 0x00010000... (33 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (46 %)
Writing at 0x0001c000... (53 %)
Writing at 0x00020000... (60 %)
Writing at 0x00024000... (66 %)
Writing at 0x00028000... (73 %)
Writing at 0x0002c000... (80 %)
Writing at 0x00030000... (86 %)
Writing at 0x00034000... (93 %)
Writing at 0x00038000... (100 %)
Wrote 331504 bytes (236618 compressed) at 0x00000000 in 20.8 seconds (effective 127.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

serial ourput:
22:23:28.856 -> 
22:23:28.856 -> 
22:23:28.856 -> Starting...
22:23:28.856 -> modes length:2941
22:23:28.856 -> WS2812FX setup
22:23:28.856 -> Configuring access point...AP IP address: 192.168.4.1
22:23:29.746 -> HTTP server setup
22:23:29.746 -> HTTP server started.
22:23:29.746 -> ready!
22:23:57.589 -> mode is Single Dynamic
22:23:57.777 -> mode is Multi Dynamic
22:23:58.011 -> mode is Rainbow Cycle
22:23:58.152 -> mode is Dual Scan
22:23:58.339 -> mode is Fade
22:23:58.527 -> mode is Scan
22:23:58.714 -> mode is Rainbow Cycle
22:23:58.855 -> mode is Single Dynamic
22:23:59.042 -> mode is Random Color
22:23:59.277 -> mode is Color Wipe Reverse Inverse
22:23:59.370 -> mode is Color Wipe Reverse Inverse
22:23:59.605 -> mode is Single Dynamic
22:23:59.745 -> mode is Rainbow
22:24:00.308 -> mode is Theater Chase
22:24:00.448 -> mode is Theater Chase Rainbow
22:24:00.683 -> mode is Theater Chase Rainbow
22:24:00.823 -> mode is Fade
22:24:01.011 -> mode is Dual Scan
22:24:01.198 -> mode is Rainbow Cycle
22:24:01.902 -> mode is Color Wipe Reverse Inverse
22:24:02.089 -> mode is Color Wipe Random
22:24:02.370 -> mode is Multi Dynamic
22:24:02.464 -> mode is Scan
22:24:02.652 -> mode is Fade
22:24:02.839 -> mode is Theater Chase Rainbow
22:24:03.027 -> mode is Running Lights
22:24:03.214 -> mode is Twinkle
22:24:03.355 -> mode is Running Lights
22:24:03.917 -> mode is Dual Scan
22:24:04.058 -> mode is Dual Scan
22:24:04.292 -> mode is Rainbow
22:24:04.433 -> mode is Multi Dynamic
22:24:04.620 -> mode is Random Color
22:24:05.370 -> mode is Color Wipe Inverse

then it stopped  

[image]

moose4lord, 2022-08-12:
I think you may be running into an old problem with the ESP8266's WiFi stack. If your sketch is driving a lot of LEDs (more than a few dozen), there is a tendency for the ESP8266 CPU to spend too much time servicing the LEDs, and not enough time properly servicing the WiFi subroutines. This will cause the WiFi to hang unexpectedly. This all stems from the fact that the Adafruit_Neopixel ESP8266 driver uses a software technique ("bit banging" ) to deliver a properly timed signal to the LEDs.

One workaround is to use the ESP8266's hardware (DMA) to drive the LEDs. This offloads some of the CPU workload from the LEDs, and allows the CPU more time to spend on WiFi tasks.  The **ws2812fx_dma** example sketch shows a working example of using this technique. Note: the DMA hardware is connected to GPIO3, so you MUST use GPIO3 to drive the LEDs, not GPIO2. GPIO3 is usually labeled "RX" on an ESP-01. Also note: GPIO3/RX is used to upload software to the ESP-01, so you should connect the pin to your FTDI USB to TTL module while uploading the sketch, then connect the pin to your LEDs while the sketch is running.

To use DMA you'll need to use the Library Manager to install the "NeoPixelBus by Makuna" library, which contains a DMA driver for ESP8266.  Then add these elements to your sketch (see **ws2812fx_dma** for reference):

1) include the NeoPixelBus lib at the beginning of your sketch:
```
#include <NeoPixelBus.h>
```
2) create a NeoPixelBus object called strip:
```
NeoPixelBus<NeoGrbFeature, NeoEsp8266Dma800KbpsMethod> strip(LED_COUNT);
```
3) Initialize the strip object in the setup() function (after ws2812fx.init() ):
```
strip.Begin();
strip.Show();
```
4) setup the custom show function in the setup() function:
```
ws2812fx.setCustomShow(myCustomShow);   // set the custom show function
```
5) add this myCustomShow() function at the bottom of your sketch:
```
void myCustomShow(void) {
  if(strip.CanShow()) {
    memcpy(strip.Pixels(), ws2812fx.getPixels(), strip.PixelsSize());instance
    strip.Dirty();
    strip.Show();
  }
}
```

Sorry. I know this is very messy. It's a pain in the butt, but does make WiFi run a lot smoother.
Let me know if you have any success with it.

rahulbaruri, 2022-08-13:
> I think you may be running into an old problem with the ESP8266's WiFi stack. If your sketch is driving a lot of LEDs (more than a few dozen), there is a tendency for the ESP8266 CPU to spend too [...]

i am using esp 01 8266

and 300 rgb led.

how could i resolve this problem?

moose4lord, 2022-08-13:
I have modified your sketch to use the DMA technique I mentioned earlier (see attached). It uses GPIO3/RX to drive the LEDs (instead of GPIO2), so you'll have to change your wiring a little. It should fix your problem with WiFi crashing unexpectedly.

[sketch_aug13a.zip](https://github.com/kitesurfer1404/WS2812FX/files/9331381/sketch_aug13a.zip)

rahulbaruri, 2022-08-13:
> I have modified your sketch to use the DMA technique I mentioned earlier (see attached). It uses GPIO3/RX to drive the LEDs (instead of GPIO2), so you'll have to change your wiring a little. It should fix your problem with WiFi crashing unexpectedly.
> 
> [sketch_aug13a.zip](https://github.com/kitesurfer1404/WS2812FX/files/9331381/sketch_aug13a.zip)
> 
> It uses the NeoPixelBus library, so you'll need to use the Library Manager to install the "NeoPixelBus by Makuna" library.

> I have modified your sketch to use the DMA technique I mentioned earlier (see attached). It uses GPIO3/RX to drive the LEDs (instead of GPIO2), so you'll have to change your wiring a little. It should fix your problem with WiFi crashing unexpectedly.
> 
> [sketch_aug13a.zip](https://github.com/kitesurfer1404/WS2812FX/files/9331381/sketch_aug13a.zip)
> 
> It uses the NeoPixelBus library, so you'll need to use the Library Manager to install the "NeoPixelBus by Makuna" library.

21:24:25.393 -> Starting...
21:24:25.393 -> WS2812FX setup
21:24:25.393 -> Wifi setup
21:24:26.377 -> 192.168.4.1
21:24:26.377 -> HTTP server setup
21:24:26.377 -> HTTP server started.
21:24:26.377 -> ready!
21:24:26.471 -> 
21:24:26.471 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
21:24:26.471 -> 
21:24:26.471 -> Exception (4):
21:24:26.471 -> epc1=0x400005de epc2=0x00000000 epc3=0x00000000 excvaddr=0x000000e4 depc=0x00000000
21:24:26.471 -> 
21:24:26.471 -> >>>stack>>>
21:24:26.471 -> 
21:24:26.471 -> ctx: sys
21:24:26.471 -> sp: 3fffd8f0 end: 3fffffb0 offset: 0190
21:24:26.471 -> 3fffda80:  04000002 3feffe00 00000058 4000d770  
21:24:26.517 -> 3fffda90:  00000470 8001b00e 8d0c8c0d 8f0a8e0b  
21:24:26.517 -> 3fffdaa0:  00000009 3fffdab0 3fffeb30 401056ba  
21:24:26.517 -> 3fffdab0:  4010569c 3fffc100 00000000 00000000  
21:24:26.517 -> 3fffdac0:  00000004 400005cb 00000001 00000001  
21:24:26.517 -> 3fffdad0:  400005e1 00000000 00000000 00000000  
21:24:26.517 -> 3fffdae0:  4023fafe 00000033 00000010 00000000  
21:24:26.517 -> 3fffdaf0:  4023faf4 00000004 00000004 0000001e  
21:24:26.517 -> 3fffdb00:  fbf8ffff 40230000 3feffe00 000000b0  
21:24:26.517 -> 3fffdb10:  0000001a 00000018 04000102 00000000  
21:24:26.517 -> 3fffdb20:  000000e4 00000000 00000000 00000000  
21:24:26.564 -> 3fffdb30:  400005cb 00000030 00000010 ffffffff  
21:24:26.564 -> 3fffdb40:  4023faf4 00000033 4023fafe 0000001e  
21:24:26.564 -> 3fffdb50:  fbf8ffff 40230000 3feffe00 000000b0  
21:24:26.564 -> 3fffdb60:  0000001a 00000018 04000102 00000010  
21:24:26.564 -> 3fffdb70:  00000021 00000000 00000000 00000030  
21:24:26.564 -> 3fffdb80:  00000000 00000000 00000000 00000000  
21:24:26.564 -> 3fffdb90:  00000000 00000000 00000000 00000000  
21:24:26.564 -> 3fffdba0:  00000000 4023faf4 00000000 000000e4  
21:24:26.564 -> 3fffdbb0:  00000000 00000000 00000000 00000000  
21:24:26.564 -> 3fffdbc0:  40248090 00000004 00000001 fbf8ffff  
21:24:26.564 -> 3fffdbd0:  04000002 3feffe00 00000000 00000000  
21:24:26.611 -> 3fffdbe0:  00000000 00000000 00000000 00000000  
21:24:26.611 -> 3fffdbf0:  4023afa8 3ffe92e0 00000003 401056ba  
21:24:26.611 -> 3fffdc00:  4010569c 3fffc100 00000000 00000000  
21:24:26.611 -> 3fffdc10:  00000004 400005cb 00000000 00000000  
21:24:26.611 -> 3fffdc20:  400005e1 00000000 00000000 00000000  
21:24:26.611 -> 3fffdc30:  4023fafe 00000033 00000010 00000000  
21:24:26.611 -> 3fffdc40:  4023faf4 00000004 00000004 0000001e  
21:24:26.611 -> 3fffdc50:  fbf8ffff 40230000 3feffe00 000000b0  
21:24:26.611 -> 3fffdc60:  0000001a 00000018 04000102 00000000  
21:24:26.611 -> 3fffdc70:  000000e4 00000000 00000000 00000000  
21:24:26.658 -> 3fffdc80:  400005cb 00000030 00000010 ffffffff  
21:24:26.658 -> 3fffdc90:  c000288d 00000033 4023fae3 00000001  
21:24:26.658 -> 3fffdca0:  fbf8ffff 04000002 3feffe00 00000100  
21:24:26.658 -> 3fffdcb0:  0000001a 00000018 04000102 00000010  
21:24:26.658 -> 3fffdcc0:  00000021 00000000 0000001c 00000030  
21:24:26.658 -> 3fffdcd0:  4010f494 00000000 00000000 00000000  
21:24:26.658 -> 3fffdce0:  00000000 00000000 00000000 00000000  
21:24:26.658 -> 3fffdcf0:  00000000 4023faf4 00000000 000000e4  
21:24:26.658 -> 3fffdd00:  00000000 00000000 00000000 00000000  
[... 570 lines cut ...]
21:24:34.860 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
21:24:34.860 -> 
21:24:34.860 -> wdt reset
21:24:34.860 -> load 0x4010f000, len 3460, room 16 
21:24:34.906 -> tail 4
21:24:34.906 -> chksum 0xcc
21:24:34.906 -> load 0x3fff20b8, len 40, room 4 
21:24:34.906 -> tail 4
21:24:34.906 -> chksum 0xc9
21:24:34.906 -> csum 0xc9
21:24:34.906 -> v000514f0
21:24:34.906 -> ~ld
21:24:35.469 -> 
21:24:35.469 -> 
21:24:35.469 -> Starting...
21:24:35.469 -> WS2812FX setup
21:24:35.469 -> Wifi setup
21:24:36.453 -> 192.168.4.1
21:24:36.453 -> HTTP server setup
21:24:36.453 -> HTTP server started.
21:24:36.453 -> ready!
21:24:45.123 -> 
21:24:45.123 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,6)

this problem facing now

moose4lord, 2022-08-13:
Did you unplug the USB programmer when running the sketch? Since GPIO3/RX is shared between the LEDs and the FTDI programmer, you may have problems if both are connected at the same time.

rahulbaruri, 2022-08-13:
> Did you unplug the USB programmer when running the sketch? Since GPIO3/RX is shared between the LEDs and the FTDI programmer, you may have problems if both are connected at the same time.

I am using Arduino board by shorting gnd and rst

moose4lord, 2022-08-13:
Sorry, the stack trace you pasted isn't really telling me anything.

Unfortunately, without an ESP-01 to experiment with, my debug options are pretty limited.  My NodeMCU ESP-12E just doesn't seem to have the issues you're seeing.

So I'll order an ESP-01 from Amazon and see if I can reproduce your results with that. It should arrive in a couple days and I'll get back to you then.

moose4lord, 2022-08-15:
Well, I got my ESP-01 and FTDI programmer from Amazon this afternoon and have been playing with it for a couple hours. I'm sorry to say that the sketch I posted earlier ([sketch_aug13a.zip](https://github.com/kitesurfer1404/WS2812FX/files/9331381/sketch_aug13a.zip)) is working perfectly for me. I was hoping getting a real ESP-01 would reveal the problem you are seeing, but that's not the case.

Here is a picture of the ESP-01 and FTDI programmer I got from Amazon:
[image]
The ESP is labelled "ESP-01S", so maybe it's not quite the same as what you have.

At any rate, everything is working fine for me. I do not leave the ESP connected to the programmer during my testing. I only connect 3V3, GND and directly connect the RX pin to the Din pin of the string of WS2812b LEDs. I use my phone to connect to the Access Point at 192.168.4.1 and fire up Chrome to access the web server. The web page works fine and the WiFi seems solid.

Sorry. I'm not sure where to go from here.

rahulbaruri, 2022-08-16:
Okk... I am using 5volt power supply and 3.3 voltage regulator..then. I have to check current consuming... By 265 led while colour changing... Might drop downing the current at esp.

rahulbaruri, 2022-09-28:
> Well, I got my ESP-01 and FTDI programmer from Amazon this afternoon and have been playing with it for a couple hours. I'm sorry to say that the sketch I posted earlier [...]

After testing for a time period... The circuit is getting restarted after a moment of hours

rahulbaruri, 2022-10-11:
It is reseting after a couple of hours

moose4lord, 2023-06-03:
This seems to be an old conversation. I'm closing for now. If it's still an issue, please repost.

closed 2023-06-03 by moose4lord