kitesurfer1404/WS2812FX #325: Comblie error in version 1.4.1

Devilscave, 2022-12-27:
After update to v1.4.1 i became this errors in (IDE 1.8.19)
version v1.4.0 is running.

`In file included from G:\Arduino Zeug\Projekte\Zapfen_lichterkette\Zapfen_lichterkette.ino:21:
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:41:16: error: redefinition of 'struct Oscillator'
   41 | typedef struct Oscillator {
      |                ^~~~~~~~~~
In file included from G:\Arduino Zeug\Projekte\Zapfen_lichterkette\Zapfen_lichterkette.ino:6:
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/WS2812FX.h:514:8: note: previous definition of 'struct Oscillator'
  514 | struct Oscillator {
      |        ^~~~~~~~~~
In file included from G:\Arduino Zeug\Projekte\Zapfen_lichterkette\Zapfen_lichterkette.ino:21:
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h: In function 'uint16_t oscillate()':
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:56:3: error: cannot convert '<brace-enclosed initializer list>' to 'oscillator' {aka 'int'} in initialization
   56 |   };
      |   ^
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:56:3: error: cannot convert '<brace-enclosed initializer list>' to 'oscillator' {aka 'int'} in initialization
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:56:3: error: cannot convert '<brace-enclosed initializer list>' to 'oscillator' {aka 'int'} in initialization
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:59:20: error: request for member 'pos' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   59 |     oscillators[i].pos += oscillators[i].dir * oscillators[i].speed;
      |                    ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:59:42: error: request for member 'dir' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   59 |     oscillators[i].pos += oscillators[i].dir * oscillators[i].speed;
      |                                          ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:59:63: error: request for member 'speed' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   59 |     oscillators[i].pos += oscillators[i].dir * oscillators[i].speed;
      |                                                               ^~~~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:60:24: error: request for member 'dir' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   60 |     if((oscillators[i].dir == -1) && (oscillators[i].pos <= 0)) {
      |                        ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:60:54: error: request for member 'pos' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   60 |     if((oscillators[i].dir == -1) && (oscillators[i].pos <= 0)) {
      |                                                      ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:61:22: error: request for member 'pos' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   61 |       oscillators[i].pos = 0;
      |                      ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:62:22: error: request for member 'dir' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   62 |       oscillators[i].dir = 1;
      |                      ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:63:22: error: request for member 'speed' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   63 |       oscillators[i].speed = random(1, 3);
      |                      ^~~~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:66:24: error: request for member 'dir' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   66 |     if((oscillators[i].dir == 1) && (oscillators[i].pos >= (seglen - 1))) {
      |                        ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:66:53: error: request for member 'pos' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   66 |     if((oscillators[i].dir == 1) && (oscillators[i].pos >= (seglen - 1))) {
      |                                                     ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:67:22: error: request for member 'pos' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   67 |       oscillators[i].pos = seglen - 1;
      |                      ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:68:22: error: request for member 'dir' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   68 |       oscillators[i].dir = -1;
      |                      ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:69:22: error: request for member 'speed' in 'oscillators[((int)i)]', which is of non-class type 'oscillator' {aka 'int'}
   69 |       oscillators[i].speed = random(1, 3);
      |                      ^~~~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:77:30: error: request for member 'pos' in 'oscillators[((int)j)]', which is of non-class type 'oscillator' {aka 'int'}
   77 |       if(i >= oscillators[j].pos - oscillators[j].size && i <= oscillators[j].pos + oscillators[j].size) {
      |                              ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:77:51: error: request for member 'size' in 'oscillators[((int)j)]', which is of non-class type 'oscillator' {aka 'int'}
   77 |       if(i >= oscillators[j].pos - oscillators[j].size && i <= oscillators[j].pos + oscillators[j].size) {
      |                                                   ^~~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:77:79: error: request for member 'pos' in 'oscillators[((int)j)]', which is of non-class type 'oscillator' {aka 'int'}
   77 |       if(i >= oscillators[j].pos - oscillators[j].size && i <= oscillators[j].pos + oscillators[j].size) {
      |                                                                               ^~~
C:\Users\Blacky2\Documents\Arduino\libraries\WS2812FX\src/custom/Oscillate.h:77:100: error: request for member 'size' in 'oscillators[((int)j)]', which is of non-class type 'oscillator' {aka 'int'}
   77 |       if(i >= oscillators[j].pos - oscillators[j].size && i <= oscillators[j].pos + oscillators[j].size) {
      |                                                                                                    ^~~~
G:\Arduino Zeug\Projekte\Zapfen_lichterkette\Zapfen_lichterkette.ino: In function 'void setup()':
G:\Arduino Zeug\Projekte\Zapfen_lichterkette\Zapfen_lichterkette.ino:179:27: warning: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '450' to '194' [-Woverflow]
  179 |   cometbrush.setFadeSpeed(450);
      |                           ^~`

moose4lord, 2022-12-27:
It looks like you're trying to use the Oscillate custom pattern.

Originally the custom effects were created to allow adding new effects to WS2812FX without overwhelming the Arduino's limited memory. Users could pick and choose custom effects that they liked, and add them individually to their sketches without exhausting the Arduino's memory. This custom effect strategy kind of worked against users with ESP or RP2040 processors, which have more than enough memory to accommodate all of the custom effects.

One of the changes implemented in WS2812FX v1.4.1 was the separation of effects for Arduino and ESP/RP2040 microprocessors. On Arduino processors the custom effects work the same as before. However, on ESP and RP2040 processors (which I assume is what you're using) all of the custom effects are now compiled into the sketch as standard effects, and you no longer have to include them separately. So if you remove all the code that includes custom effects, your sketch should compile.

Devilscave, 2022-12-30:
Ok thanks for the info, unfortunately I hadn't found anything about that. Have now removed the old oscilator effect and it works again. The new changed effect doesn't look that great either.

closed 2022-12-30 by Devilscave