kitesurfer1404/WS2812FX #326: Nice Effekt BUG, Please LET IT!!!

Devilscave, 2022-12-30:
Hello, 
I integrated a few of the new effects into my project and found a funny bug with the effect of "Twinkelfox", which you can simply leave.
Description: If you choose the color black (RGB 0.0.0), it does not become dark but random colorful. Very NICE

And second I found two ways for the "Block Dissolve" effect to add really nice new functions.
1. If you change the color on the last LED in your Stripe befor it change to dark, it will be not dark, it loads the new color over the first.
2. If you change the color after the last it will be load to dark and after that it begins whit the new color (color change).

moose4lord, 2022-12-31:
Actually the TwinkleFox behavior when color[0]==BLACK isn't a bug, it's working as designed. Although, it's not documented anywhere. Surprise! :)

I'm not sure what you mean by the Block Dissolve feature. Are you changing the segment's colors[] array dynamically (as the animation if running)? Can you post an example sketch?

Devilscave, 2023-01-01:
Ahh, great surprise in my opinion.

I don't have an example sketch for Block Dissolve, but I try to explain it a little more closely.

1. For example, if you use 300 LEDs and the effect of Block Dissolve is called up in the color blue and I switch on LED 299 the color to red then the effect will not change too dark again, but puts red over all blue ones, then switch again at 299 Color he puts it back over the previous color over it without switching into the dark. Like a permanently Color Change.
I will try to write a code that simulates this effect and post it here.

2. The same system as with 1. would be easy to use if you do the color change instead of LED 299 at LED 301. Then "Block Dissolve" changes too dark again and then starts again with a different color, but it just makes the dark effect in between.
I will also try to simulate this effect.

Currently you can simulate it by hand via the web interface example.
Change the color befor it will be change to dark and after it change to dark you will see what i mean.

moose4lord, 2023-01-01:
Ah, I see what you mean about Block Dissolve. I haven't played with changing the colors on the fly like that. I looks good.
You can synchronize the effect by using the aux_param variable to trigger the color change. I made this little demo app:
```c++
#include <WS2812FX.h>

#define LED_PIN   12  // digital pin used to drive the LED strip
#define LED_COUNT 64  // number of LEDs on the strip

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

uint32_t myColors[] = { RED, YELLOW, GREEN, CYAN, BLUE, MAGENTA, ORANGE, PURPLE };
uint8_t colorIndex = 0;
uint8_t numColors = sizeof(myColors) / sizeof(myColors[0]);

void setup() {
  Serial.begin(115200);

  ws2812fx.init();
  ws2812fx.setBrightness(32);

  // parameters: index, start, stop, mode, color, speed
  ws2812fx.setSegment(0,  0,  LED_COUNT-1, FX_MODE_BLOCK_DISSOLVE, BLUE, 1000);  // start with blue

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();

  if(ws2812fx.getSegmentRuntime()->aux_param == 1) { // change color when all LEDs are the same color
//if(ws2812fx.getSegmentRuntime()->aux_param == 2) { // or change color when all LEDs are dark
    colorIndex = (colorIndex + 1) % numColors;
    ws2812fx.setColor(myColors[colorIndex]); // set the next color

    ws2812fx.getSegmentRuntime()->aux_param = 0;
  }
}
```

Thanks for the tip.

Devilscave, 2023-01-02:
YES man, you had it.
That is what i mean. I didn't get it yesterday. NICE
Can you make a random change of color ?

moose4lord, 2023-01-02:
That's a nice tweak. You can also use more random colors (not just those in the myColors[] array) like so:
```c++
  if(ws2812fx.getSegmentRuntime()->aux_param == 1) { // change color when all LEDs are the same color
//if(ws2812fx.getSegmentRuntime()->aux_param == 2) { // or change color when all LEDs are dark
//  colorIndex = (colorIndex + 1) % numColors;
//  ws2812fx.setColor(myColors[colorIndex]); // set the next color

    uint32_t randomColor = ws2812fx.color_wheel(ws2812fx.random8()); // select a new color at random
    ws2812fx.setColor(randomColor);

    ws2812fx.getSegmentRuntime()->aux_param = 0;
  }
```

Devilscave, 2023-01-02:
That's nice too, I immediately implemented it in my light chain.

moose4lord, 2023-06-03:
Just closing some old issues.

closed 2023-06-03 by moose4lord