kitesurfer1404/WS2812FX #327: Color changing on non changing effects?

Devilscave, 2023-01-02:
As the heading says, how can I change the colors of non -changeable effects except to change in modes.cpp. How do I determine which color gets which value.
The example of the main page:
`uint32_t colors[] = {RED, GREEN};
ws2812fx.setSegment(0, 0,           (LED_COUNT/2)-1, FX_MODE_BLINK, colors, 1000, false);
ws2812fx.setSegment(1, LED_COUNT/2, LED_COUNT-1,     FX_MODE_BLINK, COLORS(ORANGE, PURPLE), 1000, false);`

it does not work. It always comes the mistake `colors cannot be used as a funktion`

moose4lord, 2023-01-03:
I copied the three lines you mention into a test sketch and didn't see any compile issues. What environment are you using (Arduino IDE or PlatformIO) and what microprocessor?  If you can post your complete sketch,  I'll take a look further.

Devilscave, 2023-01-04:
I use a ESP8266F Board.
For legal reasons, I cannot post the entire code.
I will try again to complain in a different sketch.
Thanks

Devilscave, 2023-01-04:
Ok, i found a new way, but it doesn't do what it should 
I try to haven't led it with the help of individual sketch passages.
The complete Code is compilable and all other effekts worke fine.
But it does not hand over the color change to the effect even though I always recharge it again.

PARAMETERS:
`colChange = 0;
int COLOR1= 0xFF0000;
int COLOR2= 0x00FF00;
int COLOR3= 0x0000FF;
uint32_t colors[] = {COLOR1, COLOR2, COLOR3};`

SETUP:
`ws2812fx.setBrightness(80);
ws2812fx.init();`

LOOP: (I call the effects in a SWITCH/CASE)
`case 15:
        //Serial.println("Tricolor Chase 54"); now Theater Mode 16
        ws2812fx.resetSegments();
        ws2812fx.setBrightness(30);
        // Theater Mode 16 editiert in modes.cpp for three different colors
        ws2812fx.setSegment(0, 0, LED_COUNT - 1, 16, colors, 1500, NO_OPTIONS);        
        Serial.println("load effecct new");
        Serial.println(COLOR1);
        Serial.println(COLOR2);
        Serial.println(COLOR3);
        lastmenu = menu;                                               // stops the effekt reload for permanently restart
        break;`

LOWER IN THE LOOP: (i use the Web Interface to sepperate three colors in changing)

`for (uint8_t i = 0; i < server.args(); i++) {
    if (server.argName(i) == "c") {
      uint32_t tmp = (uint32_t) strtol(server.arg(i).c_str(), NULL, 10);
      if (tmp >= 0x000000 && tmp <= 0xFFFFFF) {
        if(lastmenu == 15){                                               // so that it only works on menu/case 15
          if(colChange == 0){
          COLOR1 = tmp;
          colChange++;
          lastmenu = 0;                                                      // that reload the Effekt so it loads new
          Serial.println("Color 1");
          }else if(colChange == 1){
          COLOR2 = tmp;
          colChange++;
          lastmenu = 0;
          Serial.println("Color 2");
          }else if(colChange == 2){
          COLOR3 = tmp;
          colChange = 0;
          lastmenu = 0;
          Serial.println("Color 3");
          }
        }else{                                                                        // this is for all other effects
        colChange = 0;
        Serial.println("only Color 1");        
        ws2812fx.setColor(tmp);
        }
      }
    }`
        
SERIAL MONITOR OUTPUT:       
`load effect new                                                               // that comes when I select the effect in the WEB Interface
16711680                                                                         // WHITE COLOR1 from Parameters
65280                                                                               // GREEN COLOR2 from Parameters
255                                                                                   // BLUE COLOR3 from Parameters
Farbe 1                                                                             // When I click once in the color selection
load effect new                                                                // it loads the effekt one times new in LOOP
11992832                                                                         // and you see it changes the COLOR1
65280                                                                               // same like before
255                                                                                   // same like before
Farbe 2                                                                             // When I click once again in the color selection
load effect new                                                                // reload again
11992832                                                                         // same like before
157183                                                                             // new value for COLOR2
255                                                                                   // same like before
Farbe 3                                                                             // When I click once again in the color selection
load effect new                                                                // reload again
11992832                                                                         // same like before
157183                                                                             // same like before
157183                                                                             // i click the same position so the same color for COLOR3`

you can see that the colors are changing, BUT he doesn't load them, they stay on red, green, blue like before.
But I don't understand why he doesn't take over the new colors although the values change.
I hope you understand my codes.

Devilscave, 2023-01-04:
its OK i am stupid

closed 2023-01-04 by Devilscave