kitesurfer1404/WS2812FX #328: potensio with ws2812b

anto168, 2023-01-18:
sorry to bother you, haven't visited you for a long time, this time I want to ask all my friends, I want to ask how to add potentio function to ws2812bfx, thank you all

moose4lord, 2023-01-18:
I'm not sure what a "potentio function" is. Can you elaborate?

anto168, 2023-01-18:
potentio to adjust the speed of the animation.

moose4lord, 2023-01-18:
Oh, you're talking about a potentiometer. What kind of microprocessor are you using?

anto168, 2023-01-18:
micro, arduino nano

anto168, 2023-01-18:
I want to make a decoration at home, with 1 led strip with 3 segments of fade animation, segment 1 lights up fades then moves to the next segment without turning off the previous segment.

anto168, 2023-01-18:
i am trying to use arduino nano.

anto168, 2023-01-18:
basic 3 segments

moose4lord, 2023-01-19:
Controlling the brightness of each segment independently isn't so easy with ws2812fx. You can use the setBrightness() function to control the global strip brightness, but there's no corresponding function for controlling an individual segment's brightness.

But I took it as a challenge and came up with this example sketch, which includes fading in each segment, one after the other, and some code to control the animation speed with a potentiometer:
```c++
#include <WS2812FX.h>

#define ANALOG_PIN A0 // analog pin connected to the potentiometer
#define LED_PIN     3 // digital pin used to drive the LED strip
#define LED_COUNT  24 // number of LEDs on the strip

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

unsigned long timer = 0;
unsigned long debugTimer = 0;

const uint32_t segColors[3] = {RED, GREEN, BLUE}; // initial color values of each of the three segments
int segBrights[3] = {0, 0, 0};                    // initial brightness values for each of the three segments

int potValue = 0;
uint16_t speedValue = 0;

void setup() {
  Serial.begin(115200);

  ws2812fx.init();
  ws2812fx.setBrightness(255);

  // parameters: index, start, stop, mode, color, speed, reverse
  ws2812fx.setSegment(0,  0,  7, FX_MODE_THEATER_CHASE, RED,   1000); // segment 0
  ws2812fx.setSegment(1,  8, 15, FX_MODE_SCAN,          GREEN, 1000); // segment 1
  ws2812fx.setSegment(2, 16, 23, FX_MODE_COMET,         BLUE,  1000); // segment 2

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();

  // update speed and brightness every 20ms
  if(millis() > timer + 20) {
    potValue = analogRead(ANALOG_PIN);             // get the potentiometer value (0 - 1023)
    speedValue = map(potValue, 0, 1023, 40, 2000); // scale the pot value to a speed value

    if(segBrights[0] < 255) { // increase each segment's brightness one after another
      segBrights[0]++;
    } else if(segBrights[1] < 255) {
      segBrights[1]++;
    } else if(segBrights[2] < 255) {
      segBrights[2]++;
    } else {
      segBrights[0] = 0; // when all segments reach full brightness, reset brightnesses
      segBrights[1] = 0;
      segBrights[2] = 0;
      ws2812fx.clear();
    }

    // update each segment
    for(int i=0; i<3; i++) {
      uint32_t newColor = ws2812fx.color_blend(BLACK, segColors[i], segBrights[i]); // calc blended color (initial color modified by current brightness)
      ws2812fx.setColor(i, newColor);   // assign modified color to each segment
      ws2812fx.setSpeed(i, speedValue); // assign speed to each segment
    }

    timer = millis();
  }

  if(millis() > debugTimer + 1000) { // debug message every 1 second
    Serial.print("brightnesses:"); Serial.print(segBrights[0]); Serial.print(", "); Serial.print(segBrights[1]); Serial.print(", "); Serial.println(segBrights[2]);
    Serial.print("potValue:"); Serial.println(potValue);
    Serial.print("speedValue:"); Serial.println(speedValue);
    debugTimer = millis();
  }
}
```
The sketch fades in each segment by using a segBrights[] array to keep track of each segment's brightness, and uses that to blend the segment's initial color with BLACK to yield a new color with the correct brightness. It doesn't work with all effects (notably the rainbow effects don't use a segment's assigned color, so they would be unaffected by the fade-in code), but it might be good enough for what you want to accomplish.

anto168, 2023-01-19:
thank you for helping me, I have made and modified your code from you.

anto168, 2023-01-19:
may I ask something else, about fastled, I tried to make a code like this, how do I adjust the brightness on case 4 so that it lights up from dim to bright.

anto168, 2023-01-19:
#include <FastLED.h>
#define NUM_LEDS 10
#define DATA_PIN 2
#define MAX_FRAMES 8

CRGB leds[NUM_LEDS];
int frameIndex = 0;

int nextFrameIndex() {
  if (frameIndex < MAX_FRAMES - 1) {
      return frameIndex + 1;
    } else {
      return 0;
    }
}

void setup() {
    FastLED.addLeds<WS2811, DATA_PIN, GRB>(leds, NUM_LEDS);
    FastLED.setBrightness(200);
    // Frame 1
  leds[0].setRGB(255, 255, 255);
  leds[1].setRGB(0, 0, 0);
  leds[2].setRGB(0, 0, 0);
  leds[3].setRGB(0, 0, 0);
  leds[4].setRGB(0, 0, 0);
  leds[5].setRGB(0, 0, 0);
  leds[6].setRGB(0, 0, 0);
  leds[7].setRGB(0, 0, 0);
  leds[8].setRGB(0, 0, 0);
  leds[9].setRGB(0, 0, 0);

}

void loop() {
    EVERY_N_MILLISECONDS(500) {
        frameIndex = nextFrameIndex();
        setColors();
    }
    FastLED.show();
}

void setColors() {
    switch (frameIndex) {
    case 0:
      leds[0].setRGB(255, 255, 255);
      leds[1].setRGB(0, 0, 0);
      leds[2].setRGB(0, 0, 0);
      leds[3].setRGB(0, 0, 0);
      leds[4].setRGB(0, 0, 0);
      leds[5].setRGB(0, 0, 0);
      leds[6].setRGB(0, 0, 0);
      leds[7].setRGB(0, 0, 0);
      leds[8].setRGB(0, 0, 0);
      leds[9].setRGB(0, 0, 0);
      break;
    case 1:
      leds[1].setRGB(255, 255, 255);
      break;
    case 2:
      leds[2].setRGB(255, 255, 255);
      leds[3].setRGB(255, 255, 255);
      break;
    case 3:
      leds[4].setRGB(255, 255, 255);
      break;
    case 4:
      leds[5].setRGB(255, 255, 255);
      leds[6].setRGB(255, 255, 255);
      leds[7].setRGB(255, 255, 255);
      leds[8].setRGB(255, 255, 255);
      leds[9].setRGB(255, 255, 255);
      break;
    case 5:
      leds[0].setRGB(0, 0, 0);
      leds[1].setRGB(0, 0, 0);
      leds[2].setRGB(0, 0, 0);
      leds[3].setRGB(0, 0, 0);
      leds[4].setRGB(0, 0, 0);
      leds[5].setRGB(0, 0, 0);
      leds[6].setRGB(0, 0, 0);
      leds[7].setRGB(0, 0, 0);
      leds[8].setRGB(0, 0, 0);
      leds[9].setRGB(0, 0, 0);
      break;
    case 6:
      leds[0].setRGB(255, 255, 255);
      leds[1].setRGB(255, 255, 255);
      leds[2].setRGB(255, 255, 255);
      leds[3].setRGB(255, 255, 255);
      leds[4].setRGB(255, 255, 255);
      leds[5].setRGB(255, 255, 255);
      leds[6].setRGB(255, 255, 255);
      leds[7].setRGB(255, 255, 255);
      leds[8].setRGB(255, 255, 255);
      leds[9].setRGB(255, 255, 255);
      break;
    case 7:
      leds[0].setRGB(0, 0, 0);
      leds[1].setRGB(0, 0, 0);
      leds[2].setRGB(0, 0, 0);
      leds[3].setRGB(0, 0, 0);
      leds[4].setRGB(0, 0, 0);
      leds[5].setRGB(0, 0, 0);
      leds[6].setRGB(0, 0, 0);
      leds[7].setRGB(0, 0, 0);
      leds[8].setRGB(0, 0, 0);
      leds[9].setRGB(0, 0, 0);
      break;

    }
}

moose4lord, 2023-01-19:
I'm not sure exactly what you're after. Do you just want to fade-in the 5 LEDs in case 4?

You can use the nscale8() function to adjust each individual LED's brightness. I came up with  this:
```c++
#include <FastLED.h>
#define NUM_LEDS 10
#define DATA_PIN 2
#define MAX_FRAMES 8

#define FRAME_DELAY 500                 // ms per frame
#define FADE_DELAY  (FRAME_DELAY / 256)  // ms per fade step

CRGB leds[NUM_LEDS];
int frameIndex = 0;

int nextFrameIndex() {
  if (frameIndex < MAX_FRAMES - 1) {
    return frameIndex + 1;
  } else {
    return 0;
  }
}

int fadeAmount = 0;

void setup() {
  FastLED.addLeds<WS2811, DATA_PIN, GRB>(leds, NUM_LEDS);
  FastLED.setBrightness(200);
  // Frame 1
  leds[0].setRGB(255, 255, 255);
  leds[1].setRGB(0, 0, 0);
  leds[2].setRGB(0, 0, 0);
  leds[3].setRGB(0, 0, 0);
  leds[4].setRGB(0, 0, 0);
  leds[5].setRGB(0, 0, 0);
  leds[6].setRGB(0, 0, 0);
  leds[7].setRGB(0, 0, 0);
  leds[8].setRGB(0, 0, 0);
  leds[9].setRGB(0, 0, 0);
}

void loop() {
  EVERY_N_MILLISECONDS( FRAME_DELAY ) { // this loop controls the frame rate
    frameIndex = nextFrameIndex();
    fadeAmount = 0; // reset the fadeAmount at the beginning of each frame
  }
  
  EVERY_N_MILLISECONDS( FADE_DELAY ) { // this loop controls the fade rate
    if(fadeAmount < 255) fadeAmount++; // increment the fadeAmount for each call to show()
    setColors();
    FastLED.show();
  }
}

void setColors() {
  switch (frameIndex) {
    case 0:
      leds[0].setRGB(255, 255, 255);
      leds[1].setRGB(0, 0, 0);
      leds[2].setRGB(0, 0, 0);
      leds[3].setRGB(0, 0, 0);
      leds[4].setRGB(0, 0, 0);
      leds[5].setRGB(0, 0, 0);
      leds[6].setRGB(0, 0, 0);
      leds[7].setRGB(0, 0, 0);
      leds[8].setRGB(0, 0, 0);
      leds[9].setRGB(0, 0, 0);
      break;
    case 1:
      leds[1].setRGB(255, 255, 255);
      break;
    case 2:
      leds[2].setRGB(255, 255, 255);
      leds[3].setRGB(255, 255, 255);
      break;
    case 3:
      leds[4].setRGB(255, 255, 255);
      break;
    case 4:
      leds[5].setRGB(255, 255, 255); leds[5].nscale8(fadeAmount);
      leds[6].setRGB(255, 255, 255); leds[6].nscale8(fadeAmount);
      leds[7].setRGB(255, 255, 255); leds[7].nscale8(fadeAmount);
      leds[8].setRGB(255, 255, 255); leds[8].nscale8(fadeAmount);
      leds[9].setRGB(255, 255, 255); leds[9].nscale8(fadeAmount);
      break;
    case 5:
      leds[0].setRGB(0, 0, 0);
      leds[1].setRGB(0, 0, 0);
      leds[2].setRGB(0, 0, 0);
      leds[3].setRGB(0, 0, 0);
      leds[4].setRGB(0, 0, 0);
      leds[5].setRGB(0, 0, 0);
      leds[6].setRGB(0, 0, 0);
      leds[7].setRGB(0, 0, 0);
      leds[8].setRGB(0, 0, 0);
      leds[9].setRGB(0, 0, 0);
      break;
    case 6:
      leds[0].setRGB(255, 255, 255);
      leds[1].setRGB(255, 255, 255);
      leds[2].setRGB(255, 255, 255);
      leds[3].setRGB(255, 255, 255);
      leds[4].setRGB(255, 255, 255);
      leds[5].setRGB(255, 255, 255);
      leds[6].setRGB(255, 255, 255);
      leds[7].setRGB(255, 255, 255);
      leds[8].setRGB(255, 255, 255);
      leds[9].setRGB(255, 255, 255);
      break;
    case 7:
      leds[0].setRGB(0, 0, 0);
      leds[1].setRGB(0, 0, 0);
      leds[2].setRGB(0, 0, 0);
      leds[3].setRGB(0, 0, 0);
      leds[4].setRGB(0, 0, 0);
      leds[5].setRGB(0, 0, 0);
      leds[6].setRGB(0, 0, 0);
      leds[7].setRGB(0, 0, 0);
      leds[8].setRGB(0, 0, 0);
      leds[9].setRGB(0, 0, 0);
      break;
  }
}
```
I'm not a FastLED expert, so there may be a better way to do this, but maybe this'll get you started.

anto168, 2023-01-20:
so when you enter case 4, on case 4 it doesn't light up immediately, but from dim to bright then moves to case 5.

anto168, 2023-01-20:
like fade in, from dim to bright.

anto168, 2023-01-20:
#define FRAME_DELAY 500                 // ms per frame
#define FADE_DELAY  (FRAME_DELAY / 256)

Can this value setting be changed, using a potentiometer?

moose4lord, 2023-01-21:
Sure. See my earlier sketch. Make FRAME_DELAY and FADE_DELAY variables. Then read the pot value and scale the pot value to a suitable range.
```c++
int potValue = analogRead(ANALOG_PIN);              // get the potentiometer value (0 - 1023)
int frameDelay = map(potValue, 0, 1023, 500, 5000); // scale the pot value to a suitable frameDelay
int fadeDelay = frameDelay / 256;
```

anto168, 2023-01-21:
#include <FastLED.h>
#define NUM_LEDS 10
#define DATA_PIN 2
#define MAX_FRAMES 8

#define FRAME_DELAY 500     
#define FADE_DELAY  

CRGB leds[NUM_LEDS];
int frameIndex = 0;

int potValue = analogRead(A0);        
int frameDelay = map(potValue, 0, 1023, 500, 5000); 
int fadeDelay = frameDelay / 256;

int nextFrameIndex() {
  if (frameIndex < MAX_FRAMES - 1) {
    return frameIndex + 1;
  } else {
    return 0;
  }
}

int fadeAmount = 0;

void setup() {
  FastLED.addLeds<WS2811, DATA_PIN, GRB>(leds, NUM_LEDS);
  FastLED.setBrightness(200);
  // Frame 1
  leds[0].setRGB(255, 255, 255);
  leds[1].setRGB(0, 0, 0);
  leds[2].setRGB(0, 0, 0);
  leds[3].setRGB(0, 0, 0);
  leds[4].setRGB(0, 0, 0);
  leds[5].setRGB(0, 0, 0);
  leds[6].setRGB(0, 0, 0);
  leds[7].setRGB(0, 0, 0);
  leds[8].setRGB(0, 0, 0);
  leds[9].setRGB(0, 0, 0);
}

void loop() {
  EVERY_N_MILLISECONDS( FRAME_DELAY ) { // this loop controls the frame rate
    frameIndex = nextFrameIndex();
    fadeAmount = 0; // reset the fadeAmount at the beginning of each frame
  }
  
  EVERY_N_MILLISECONDS( FADE_DELAY ) { // this loop controls the fade rate
    if(fadeAmount < 255) fadeAmount++; // increment the fadeAmount for each call to show()
    setColors();
    FastLED.show();
  }
}

void setColors() {
  switch (frameIndex) {
    case 0:
      leds[0].setRGB(255, 255, 255);
      leds[1].setRGB(0, 0, 0);
      leds[2].setRGB(0, 0, 0);
      leds[3].setRGB(0, 0, 0);
      leds[4].setRGB(0, 0, 0);
      leds[5].setRGB(0, 0, 0);
      leds[6].setRGB(0, 0, 0);
      leds[7].setRGB(0, 0, 0);
      leds[8].setRGB(0, 0, 0);
      leds[9].setRGB(0, 0, 0);
      break;
    case 1:
      leds[1].setRGB(255, 255, 255);
      break;
    case 2:
      leds[2].setRGB(255, 255, 255);
      leds[3].setRGB(255, 255, 255);
      break;
    case 3:
      leds[4].setRGB(255, 255, 255);
      break;
    case 4:
      leds[5].setRGB(255, 255, 255); leds[5].nscale8(fadeAmount);
      leds[6].setRGB(255, 255, 255); leds[6].nscale8(fadeAmount);
      leds[7].setRGB(255, 255, 255); leds[7].nscale8(fadeAmount);
      leds[8].setRGB(255, 255, 255); leds[8].nscale8(fadeAmount);
      leds[9].setRGB(255, 255, 255); leds[9].nscale8(fadeAmount);
      break;
    case 5:
      leds[0].setRGB(0, 0, 0);
      leds[1].setRGB(0, 0, 0);
      leds[2].setRGB(0, 0, 0);
      leds[3].setRGB(0, 0, 0);
      leds[4].setRGB(0, 0, 0);
      leds[5].setRGB(0, 0, 0);
      leds[6].setRGB(0, 0, 0);
      leds[7].setRGB(0, 0, 0);
      leds[8].setRGB(0, 0, 0);
      leds[9].setRGB(0, 0, 0);
      break;
    case 6:
      leds[0].setRGB(255, 255, 255);
      leds[1].setRGB(255, 255, 255);
      leds[2].setRGB(255, 255, 255);
      leds[3].setRGB(255, 255, 255);
      leds[4].setRGB(255, 255, 255);
      leds[5].setRGB(255, 255, 255);
      leds[6].setRGB(255, 255, 255);
      leds[7].setRGB(255, 255, 255);
      leds[8].setRGB(255, 255, 255);
      leds[9].setRGB(255, 255, 255);
      break;
    case 7:
      leds[0].setRGB(0, 0, 0);
      leds[1].setRGB(0, 0, 0);
      leds[2].setRGB(0, 0, 0);
      leds[3].setRGB(0, 0, 0);
      leds[4].setRGB(0, 0, 0);
      leds[5].setRGB(0, 0, 0);
      leds[6].setRGB(0, 0, 0);
      leds[7].setRGB(0, 0, 0);
      leds[8].setRGB(0, 0, 0);
      leds[9].setRGB(0, 0, 0);
      break;
  }
}

sorry is it like this, .  I tried it doesn't work.

moose4lord, 2023-01-21:
Hmmm...you seem to be pretty new at this. You should probably review [Adafruit's Neopixel Uberguide](https://learn.adafruit.com/adafruit-neopixel-uberguide) to get familiar with programming wS2812 LEDs. The Adafruit Neopixel library's example sketches can be a good learning tool too.

To fix your sketch, move the new potentiometer code to the beginning of the loop() function, and change FRAME_DELAY and FADE_DELAY to frameDelay and fadeDelay, respectively.

anto168, 2023-01-21:
#include <FastLED.h>
#define NUM_LEDS 10
#define DATA_PIN 2
#define MAX_FRAMES 8

CRGB leds[NUM_LEDS];
int frameIndex = 0;

int potValue = analogRead(A0);
int frameDelay = map(potValue, 0, 1023, 500, 5000);
int fadeDelay = frameDelay / 256;

int nextFrameIndex() {
if (frameIndex < MAX_FRAMES - 1) {
return frameIndex + 1;
} else {
return 0;
}
}

int fadeAmount = 0;

void setup() {
FastLED.addLeds<WS2811, DATA_PIN, GRB>(leds, NUM_LEDS);
FastLED.setBrightness(200);
// Frame 1
leds[0].setRGB(255, 255, 255);
leds[1].setRGB(0, 0, 0);
leds[2].setRGB(0, 0, 0);
leds[3].setRGB(0, 0, 0);
leds[4].setRGB(0, 0, 0);
leds[5].setRGB(0, 0, 0);
leds[6].setRGB(0, 0, 0);
leds[7].setRGB(0, 0, 0);
leds[8].setRGB(0, 0, 0);
leds[9].setRGB(0, 0, 0);
}

void loop() {
EVERY_N_MILLISECONDS( frameDelay) { // this loop controls the frame rate
frameIndex = nextFrameIndex();
fadeAmount = 0; // reset the fadeAmount at the beginning of each frame
}

EVERY_N_MILLISECONDS( fadeDelay ) { // this loop controls the fade rate
if(fadeAmount < 255) fadeAmount++; // increment the fadeAmount for each call to show()
setColors();
FastLED.show();
}
}

void setColors() {
switch (frameIndex) {
case 0:
leds[0].setRGB(255, 255, 255);
leds[1].setRGB(0, 0, 0);
leds[2].setRGB(0, 0, 0);
leds[3].setRGB(0, 0, 0);
leds[4].setRGB(0, 0, 0);
leds[5].setRGB(0, 0, 0);
leds[6].setRGB(0, 0, 0);
leds[7].setRGB(0, 0, 0);
leds[8].setRGB(0, 0, 0);
leds[9].setRGB(0, 0, 0);
break;
case 1:
leds[1].setRGB(255, 255, 255);
break;
case 2:
leds[2].setRGB(255, 255, 255);
leds[3].setRGB(255, 255, 255);
break;
case 3:
leds[4].setRGB(255, 255, 255);
break;
case 4:
leds[5].setRGB(255, 255, 255); leds[5].nscale8(fadeAmount);
leds[6].setRGB(255, 255, 255); leds[6].nscale8(fadeAmount);
leds[7].setRGB(255, 255, 255); leds[7].nscale8(fadeAmount);
leds[8].setRGB(255, 255, 255); leds[8].nscale8(fadeAmount);
leds[9].setRGB(255, 255, 255); leds[9].nscale8(fadeAmount);
break;
case 5:
leds[0].setRGB(0, 0, 0);
leds[1].setRGB(0, 0, 0);
leds[2].setRGB(0, 0, 0);
leds[3].setRGB(0, 0, 0);
leds[4].setRGB(0, 0, 0);
leds[5].setRGB(0, 0, 0);
leds[6].setRGB(0, 0, 0);
leds[7].setRGB(0, 0, 0);
leds[8].setRGB(0, 0, 0);
leds[9].setRGB(0, 0, 0);
break;
case 6:
leds[0].setRGB(255, 255, 255);
leds[1].setRGB(255, 255, 255);
leds[2].setRGB(255, 255, 255);
leds[3].setRGB(255, 255, 255);
leds[4].setRGB(255, 255, 255);
leds[5].setRGB(255, 255, 255);
leds[6].setRGB(255, 255, 255);
leds[7].setRGB(255, 255, 255);
leds[8].setRGB(255, 255, 255);
leds[9].setRGB(255, 255, 255);
break;
case 7:
leds[0].setRGB(0, 0, 0);
leds[1].setRGB(0, 0, 0);
leds[2].setRGB(0, 0, 0);
leds[3].setRGB(0, 0, 0);
leds[4].setRGB(0, 0, 0);
leds[5].setRGB(0, 0, 0);
leds[6].setRGB(0, 0, 0);
leds[7].setRGB(0, 0, 0);
leds[8].setRGB(0, 0, 0);
leds[9].setRGB(0, 0, 0);
break;
}
}

anto168, 2023-01-21:
sorry is it like this, .

moose4lord, 2023-01-22:
No, I was thinking of this:
```c++
void loop() {
  int potValue = analogRead(A0);                      // get the potentiometer value (0 - 1023)
  int frameDelay = map(potValue, 0, 1023, 500, 5000); // scale the pot value to a suitable frameDelay
  int fadeDelay = frameDelay / 256;

  EVERY_N_MILLISECONDS( frameDelay ) { // this loop controls the frame rate
    frameIndex = nextFrameIndex();
    fadeAmount = 0; // reset the fadeAmount at the beginning of each frame
  }
  
  EVERY_N_MILLISECONDS( fadeDelay ) {  // this loop controls the fade rate
    if(fadeAmount < 255) fadeAmount++; // increment the fadeAmount for each call to show()
    setColors();
    FastLED.show();
  }
}
```

I haven't used EVERY_N_MILLISECONDS in this way before, so I'm not sure this will work. I'll try to test it tomorrow.

anto168, 2023-01-22:
I have tried to enter this program but can not,

moose4lord, 2023-01-23:
You're right, EVERY_N_MILLISECONDS doesn't work with a variable delay. But FastLED provides the EVERY_N_MILLISECONDS_I macro (note the "_I" at the end) that does work. Here's my test sketch:
```c++
#include <FastLED.h>
#define NUM_LEDS 10
#define DATA_PIN 2
#define MAX_FRAMES 8

//#define FRAME_DELAY 500                 // ms per frame
//#define FADE_DELAY  (FRAME_DELAY / 256)  // ms per fade step

CRGB leds[NUM_LEDS];
int frameIndex = 0;

int nextFrameIndex() {
  if (frameIndex < MAX_FRAMES - 1) {
    return frameIndex + 1;
  } else {
    return 0;
  }
}

int fadeAmount = 0;

void setup() {
  FastLED.addLeds<WS2811, DATA_PIN, GRB>(leds, NUM_LEDS);
  FastLED.setBrightness(200);
  // Frame 1
  leds[0].setRGB(255, 255, 255);
  leds[1].setRGB(0, 0, 0);
  leds[2].setRGB(0, 0, 0);
  leds[3].setRGB(0, 0, 0);
  leds[4].setRGB(0, 0, 0);
  leds[5].setRGB(0, 0, 0);
  leds[6].setRGB(0, 0, 0);
  leds[7].setRGB(0, 0, 0);
  leds[8].setRGB(0, 0, 0);
  leds[9].setRGB(0, 0, 0);

}

void loop() {
  int potValue = analogRead(A0);                      // get the potentiometer value (0 - 1023)
  int frameDelay = map(potValue, 0, 1023, 256, 2048); // scale the pot value to a suitable frameDelay

  EVERY_N_MILLISECONDS_I( frameTimer, 512) { // this loop controls the frame rate
    frameTimer.setPeriod( frameDelay ); // update the frame rate based on potentiometer setting
    frameIndex = nextFrameIndex();
    fadeAmount = 0; // reset the fadeAmount at the beginning of each frame
  }

  EVERY_N_MILLISECONDS_I( fadeTimer, 2) { // this loop controls the fade rate
    fadeTimer.setPeriod( frameDelay / 256 ); // update the fade rate based on potentiometer setting
    if(fadeAmount < 255) fadeAmount++; // increment the fadeAmount for each call to show()
    setColors();
    FastLED.show();
  }
}

void setColors() {
  switch (frameIndex) {
    case 0:
      leds[0].setRGB(255, 255, 255);
      leds[1].setRGB(0, 0, 0);
      leds[2].setRGB(0, 0, 0);
      leds[3].setRGB(0, 0, 0);
      leds[4].setRGB(0, 0, 0);
      leds[5].setRGB(0, 0, 0);
      leds[6].setRGB(0, 0, 0);
      leds[7].setRGB(0, 0, 0);
      leds[8].setRGB(0, 0, 0);
      leds[9].setRGB(0, 0, 0);
      break;
    case 1:
      leds[1].setRGB(255, 255, 255);
      break;
    case 2:
      leds[2].setRGB(255, 255, 255);
      leds[3].setRGB(255, 255, 255);
      break;
    case 3:
      leds[4].setRGB(255, 255, 255);
      break;
    case 4:
      leds[5].setRGB(255, 255, 255); leds[5].nscale8(fadeAmount);
      leds[6].setRGB(255, 255, 255); leds[6].nscale8(fadeAmount);
      leds[7].setRGB(255, 255, 255); leds[7].nscale8(fadeAmount);
      leds[8].setRGB(255, 255, 255); leds[8].nscale8(fadeAmount);
      leds[9].setRGB(255, 255, 255); leds[9].nscale8(fadeAmount);
      break;
    case 5:
      fill_solid(leds, NUM_LEDS, CRGB::Black);
//      leds[0].setRGB(0, 0, 0);
//      leds[1].setRGB(0, 0, 0);
//      leds[2].setRGB(0, 0, 0);
//      leds[3].setRGB(0, 0, 0);
//      leds[4].setRGB(0, 0, 0);
//      leds[5].setRGB(0, 0, 0);
//      leds[6].setRGB(0, 0, 0);
//      leds[7].setRGB(0, 0, 0);
//      leds[8].setRGB(0, 0, 0);
//      leds[9].setRGB(0, 0, 0);
      break;
    case 6:
      fill_solid(leds, NUM_LEDS, CRGB::White);
//      leds[0].setRGB(255, 255, 255);
//      leds[1].setRGB(255, 255, 255);
//      leds[2].setRGB(255, 255, 255);
//      leds[3].setRGB(255, 255, 255);
//      leds[4].setRGB(255, 255, 255);
//      leds[5].setRGB(255, 255, 255);
//      leds[6].setRGB(255, 255, 255);
//      leds[7].setRGB(255, 255, 255);
//      leds[8].setRGB(255, 255, 255);
//      leds[9].setRGB(255, 255, 255);
      break;
    case 7:
      fill_solid(leds, NUM_LEDS, CRGB::Black);
//      leds[0].setRGB(0, 0, 0);
//      leds[1].setRGB(0, 0, 0);
//      leds[2].setRGB(0, 0, 0);
//      leds[3].setRGB(0, 0, 0);
//      leds[4].setRGB(0, 0, 0);
//      leds[5].setRGB(0, 0, 0);
//      leds[6].setRGB(0, 0, 0);
//      leds[7].setRGB(0, 0, 0);
//      leds[8].setRGB(0, 0, 0);
//      leds[9].setRGB(0, 0, 0);
      break;
  }
}
```

anto168, 2023-01-24:
sketch can run well, thanks a lot for helping and teaching me, I got new experiences about sketch...

anto168, 2023-01-26:
hello, I want to ask again, on the fade delay, when I use more LEDs, why does the fadeAmount have a different contrast, the results are dimmer.

moose4lord, 2023-01-26:
Hmmmm, I'm not sure. Adding for LEDs would cause the loop() to run slower, but I think you'd have to be driving hundreds of LEDs for that to cause enough delay to be a problem. How many LEDs are you trying to use?

anto168, 2023-01-27:
i tried to use 100 pixels, the led which is in the fade position, it becomes dimmer not bright like the others, but if i use 30 pixels, the contrast of all the leds is the same.

anto168, 2023-01-27:
I use your sketch, I only add it to 100 pixels.  for fade the contrast is not full, but the others are normal full contrast.

anto168, 2023-01-27:
sketch

#include <FastLED.h>
#define NUM_LEDS 290
#define DATA_PIN 2
#define MAX_FRAMES 207

#define FRAME_DELAY 500
#define FADE_DELAY (FRAME_DELAY / 256)

CRGB leds[NUM_LEDS];
int frameIndex = 0;
int nextFrameIndex() {
  if (frameIndex < MAX_FRAMES - 1) {
      return frameIndex + 1;
    } else {
      return 0;
    }
}
int fadeAmount = 0;
void setup() {
    FastLED.addLeds<WS2811, DATA_PIN, GRB>(leds, NUM_LEDS);
    FastLED.setBrightness(255);
    // Frame 1
  leds[0].setRGB(0, 0, 0);
  leds[1].setRGB(0, 0, 0);
  leds[2].setRGB(0, 0, 0);
  leds[3].setRGB(0, 0, 0);
  leds[4].setRGB(0, 0, 0);
  leds[5].setRGB(0, 0, 0);
  leds[6].setRGB(0, 0, 0);
  leds[7].setRGB(0, 0, 0);
  leds[8].setRGB(0, 0, 0);
  leds[9].setRGB(0, 0, 0);
  leds[10].setRGB(0, 0, 0);
  leds[11].setRGB(0, 0, 0);
  leds[12].setRGB(0, 0, 0);
  leds[13].setRGB(0, 0, 0);
  leds[14].setRGB(0, 0, 0);
  leds[15].setRGB(0, 0, 0);
  leds[16].setRGB(0, 0, 0);
  leds[17].setRGB(0, 0, 0);
  leds[18].setRGB(0, 0, 0);
  leds[19].setRGB(0, 0, 0);
  leds[20].setRGB(0, 0, 0);
  leds[21].setRGB(0, 0, 0);
  leds[22].setRGB(0, 0, 0);
  leds[23].setRGB(0, 0, 0);
  leds[24].setRGB(0, 0, 0);
  leds[25].setRGB(0, 0, 0);
  leds[26].setRGB(0, 0, 0);
  leds[27].setRGB(0, 0, 0);
  leds[28].setRGB(0, 0, 0);
  leds[29].setRGB(0, 0, 0);
  leds[30].setRGB(0, 0, 0);
  
  leds[31].setRGB(0, 0, 0);
  leds[32].setRGB(0, 0, 0);
  leds[33].setRGB(0, 0, 0);
  leds[34].setRGB(0, 0, 0);
  leds[35].setRGB(0, 0, 0);
  leds[36].setRGB(0, 0, 0);
  leds[37].setRGB(0, 0, 0);
  leds[38].setRGB(0, 0, 0);
  leds[39].setRGB(0, 0, 0);
  leds[40].setRGB(0, 0, 0);
  leds[41].setRGB(0, 0, 0);
  leds[42].setRGB(0, 0, 0);
  leds[43].setRGB(0, 0, 0);
  leds[44].setRGB(0, 0, 0);
[... 1256 lines cut ...]
  break;
  case 201:
  leds[256].setRGB(255, 255, 255);
  break;
  case 202:
  leds[257].setRGB(255, 255, 255);
  break;
  case 203:
  leds[258].setRGB(255, 255, 255);
  break;
  case 204:
  leds[259].setRGB(255, 255, 255);
  break;
  case 205:
  leds[260].setRGB(255, 255, 255);
  break;
  case 206:
  leds[261].setRGB(255, 255, 255);
  break;
  case 207:
  leds[262].setRGB(255, 255, 255);
  break;

    }
}

anto168, 2023-01-27:
I tried the one without the pot

anto168, 2023-01-27:
[image]

moose4lord, 2023-01-27:
First off, if you're using WS2812B LEDs, you should change your addLeds statement to reflect that:
```c++
FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);
```

As for the dimming issue you're having, it's because the number of LEDs you're trying to drive (300!) slows down the fade function. Updating each LED takes 30 microseconds, so updating 300 takes 9 milliseconds. That's just too long and is causing the fade-in function to end prematurely and leave the LEDs in a dimmed state. I think the easiest solution is to introduce another constant that determines the amount of "fade" that is applied each time the show() is executed (and is a function of the number of LEDs being driven). I came up with this:
```c++
#define FRAME_DELAY 512 // should a power of 2
#define FADE_STEP  ((NUM_LEDS / 32) + 1)
#define FADE_DELAY (FRAME_DELAY / (256 / FADE_STEP))
```
And the   EVERY_N_MILLISECONDS(FADE_DELAY) loop changes to this:
```c++
EVERY_N_MILLISECONDS(FADE_DELAY) {
    if (fadeAmount < (255  - FADE_STEP)) fadeAmount+=FADE_STEP;
    setColors();
    FastLED.show();
  }
```
That seemed to work better in my limited testing. Give it a try.

I'll also mention that 300 LEDs is A LOT! It's not uncommon for a single WS2812B to draw 40-50mA when set to WHITE at full brightness. 300 WS2812Bs would draw over 12 Amps! This is not trivial and can be dangerous. You should definitely read the [Neopixel Uberguide](https://learn.adafruit.com/adafruit-neopixel-uberguide) especially the section on power requirements and power distribution.

anto168, 2023-01-28:
for my power is not a problem, I understand.  .  I will try your sketch later.

anto168, 2023-01-28:
sorry I want to ask, can my small project run using the WS2812FX format?

anto168, 2023-01-28:
#include <WS2812FX.h>

#define LED_PIN    2  // digital pin used to drive the LED strip
#define LED_COUNT 290  // number of LEDs on the strip

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);

  ws2812fx.init();
  ws2812fx.setBrightness(255);

   

  ws2812fx.start();
}

void loop() {
   ws2812fx.setSegment(0,  0,  62, 30, WHITE, 1000, false);
   ws2812fx.setSegment(1, 63, 78 , 15, WHITE, 500, false);
   ws2812fx.setSegment(2, 63, 78 , 15, WHITE, 500, false);
   ws2812fx.setSegment(3, 79, 90 , 15, WHITE, 500, false);
   ws2812fx.setSegment(4, 91, 108 , 15, WHITE, 500, false);
   ws2812fx.setSegment(5, 109, 121 , 15, WHITE, 500, false);
   ws2812fx.setSegment(6, 121, 263 , 30, WHITE, 500, false);
   ws2812fx.setSegment(7, 264, 290 , 0, WHITE, 1000, false);

  ws2812fx.service();
}

I tried to make it like this but, no looping, the results run concurrently not sequentially for each segment.

hahahaha

moose4lord, 2023-01-28:
Yeah, unfortunately WS2812FX doesn't have a good API to manipulate LEDs sequentially. FastLED's EVERY_N_MILLISECONDS() feature makes it a little easier, but it's still not straightforward. Actually, for this type of application I don't think you need anything more than the basic Adafruit_NeoPixel library.

This is a test sketch I made, starting from your sketch and modifying it to use Adafruit_Neopixel:
```c++
#include <Adafruit_NeoPixel.h>

#define NUM_LEDS 290
#define DATA_PIN 2
#define MAX_FRAMES 207

#define FRAME_DELAY 512 // should a power of 2
#define FADE_STEP  ((NUM_LEDS / 32) + 1)
#define FADE_DELAY (FRAME_DELAY / (256 / FADE_STEP))

Adafruit_NeoPixel leds(NUM_LEDS, DATA_PIN, NEO_GRB + NEO_KHZ800);

int frameIndex = 0;
int fadeAmount = 0;

uint32_t color   = leds.Color(0, 255, 0); // foreground color (green)
uint32_t bgColor = leds.Color(0,   0, 0); // background color (black)

int nextFrameIndex() {
  if (frameIndex < MAX_FRAMES - 1) {
    return frameIndex + 1;
  } else {
    return 0;
  }
}

void fill(int start, int end, uint32_t color) {
  leds.fill(color, start, end - start + 1);
}

void fillFade(int start, int end, uint32_t color) {
  uint8_t r = (color >> 16) * fadeAmount / 256;
  uint8_t g = ((color >> 8) & 0xff) * fadeAmount / 256;
  uint8_t b = (color & 0xff) * fadeAmount / 256;
  uint32_t fadedColor = leds.Color(r, g, b);
  leds.fill(fadedColor, start, end - start + 1);
}

void setup() {
  leds.begin();
  leds.setBrightness(255);

  // Frame 1
  leds.clear();
  leds.show();
}

void loop() {
  static uint64_t frameTimer = 0, fadeTimer = 0;
  uint64_t now = millis();

  if(now > frameTimer + FRAME_DELAY) { // EVERY_N_MILLISECONDS(FRAME_DELAY)
    frameIndex = nextFrameIndex();
    fadeAmount = 0;
    frameTimer = now;
  }

  if(now > fadeTimer + FADE_DELAY) { // EVERY_N_MILLISECONDS(FADE_DELAY)
    if (fadeAmount < (255  - FADE_STEP)) fadeAmount+=FADE_STEP;
    setColors();
    leds.show();
    fadeTimer =  now;
  }
}

void setColors() {
  switch (frameIndex) {
    case 0:
      fill(  0, 262, bgColor); // leds[0] - leds[262]
      fill(263, 290, color);   // leds[263] - leds[290]
      break;

    //////////////////// 43 cm/chip
    case 1 ... 63:
      leds.setPixelColor(frameIndex - 1, color); // leds[0] - leds[62]
      break;

    //////////////////// 15 cm/chip
    case 64:
      fillFade(63, 77, color); // leds[63] - leds[77]
      break;

    //////////////////// 12 cm/chip
    case 65:
      fillFade(78, 89, color); // leds[78] - leds[89]
      break;

    //////////////////// 18 cm/chip
    case 66:
      fillFade(90, 107, color); // leds[90] - leds[107]
      break;

    //////////////////// 13 cm/chip
    case 67:
      fillFade(108, 120, color); // leds[108] - leds[120]
      break;

    //////////////////// 20 cm/chip
    case 68 ... 207:
      leds.setPixelColor(frameIndex - 68 + 121, color); // leds[121] - leds[260]
      break;  
  }
}
```

anto168, 2023-01-29:
the animation is good, but if I change the frame speed, the fade animation becomes fast too, how do I make the fade animation smooth, and between cases the fade has a delay,

moose4lord, 2023-01-29:
The code tries to complete the fade before the next frame starts, so if you reduce FRAME_RATE to make the animation run faster it has to make the fade run faster too. I'm not sure how to get around that. You could have more than one fadeAmount variable and time the fade separately for each frame, and maybe have the fades overlap, but that would get quite complicated.

Not sure what you mean by "make the fade animation smooth". The large number of LEDs in your setup puts constraints on how fast the fades can run and what fraction of brightness gets applied for each fade step. The fades would run smoother if you reduced the number of LEDs, if that's an option.

Also "between cases the fade has a delay" is confusing to me. Do you want to start the fade a short time after the case statement is reached? I think that would make the fade smoothness worse.

anto168, 2023-01-29:
the delay I mean is the delay in , case 64- case 67. but I have tried that by adding the delay line and running.  only the smoothness of the fadding has not been found, I have tried various methods from the internet but the fade function still doesn't work.

anto168, 2023-01-30:
#include <Adafruit_NeoPixel.h>

#define NUM_LEDS 292
#define DATA_PIN 2
#define MAX_FRAMES 250

#define FRAME_DELAY 10 // should a power of 2
#define FADE_STEP  ((NUM_LEDS / 512) + 1)
#define FADE_DELAY (FRAME_DELAY / (256 / FADE_STEP))

Adafruit_NeoPixel leds(NUM_LEDS, DATA_PIN, NEO_GRB + NEO_KHZ800);

int frameIndex = 0;
int fadeAmount = 0;

uint32_t color   = leds.Color(255, 255, 255); // foreground color (green)
uint32_t bgColor = leds.Color(0,   0, 0); // background color (black)

int nextFrameIndex() {
  if (frameIndex < MAX_FRAMES - 1) {
    return frameIndex + 1;
  } else {
    return 0;
  }
}

void fill(int start, int end, uint32_t color) {
  leds.fill(color, start, end - start + 1);
}

void fillFade(int start, int end, uint32_t color) {
  uint8_t r = (color >> 16) * fadeAmount / 256;
  uint8_t g = ((color >> 8) & 0xff) * fadeAmount / 256;
  uint8_t b = (color & 0xff) * fadeAmount / 256;
  uint32_t fadedColor = leds.Color(r, g, b);
  leds.fill(fadedColor, start, end - start + 1);
}

void setup() {
  leds.begin();
  leds.setBrightness(255);

  // Frame 1
  leds.clear();
  leds.show();
}

void loop() {
  static uint64_t frameTimer = 0, fadeTimer = 0;
  uint64_t now = millis();

  if(now > frameTimer + FRAME_DELAY) { // EVERY_N_MILLISECONDS(FRAME_DELAY)
    frameIndex = nextFrameIndex();
    fadeAmount = 0;
    frameTimer = now;
  }

  if(now > fadeTimer + FADE_DELAY) { // EVERY_N_MILLISECONDS(FADE_DELAY)
    if (fadeAmount < (256  - FADE_STEP)) fadeAmount=-FADE_STEP;
    setColors();
    leds.show();
    fadeTimer = 0;
  }
}

void setColors() {
  switch (frameIndex) {
    case 0:
      fill(  0, 262, bgColor); // leds[0] - leds[262]
      fill(263, 290, color);   // leds[263] - leds[290]
      break;

    //////////////////// 43 cm/chip
    case 1 ... 63:
      leds.setPixelColor(frameIndex - 1, color); // leds[0] - leds[62]
      break;

    //////////////////// 15 cm/chip
    case 64:
    
      fillFade(63, 77, color); // leds[63] - leds[77]
      break;

    //////////////////// 12 cm/chip
    case 65:
      fillFade(78, 89, color); // leds[78] - leds[89]
      delay(1000);

    //////////////////// 18 cm/chip
    case 66:
      fillFade(90, 108, color); // leds[90] - leds[107]
      leds.show();
      delay(500);

    //////////////////// 13 cm/chip
   
      fillFade(109, 121, color); // leds[108] - leds[120]
      delay(1000);

    //////////////////// 20 cm/chip
    case 68 ... 207:
      leds.setPixelColor(frameIndex - 68 + 123, color); // leds[121] - leds[260]
      break;  
  }
}

anto168, 2023-01-30:
I only made a few changes, what I haven't been able to get is the fading effect which isn't perfect yet.

moose4lord, 2023-01-30:
You've set FRAME_DELAY = 10ms, but you can't possibly fade-in the LEDs in 10ms. It sounds like you want two different frame rates. A fast frame rate when you're turning on one LED at a time (for case 1 ... 63), and a slow frame rate when you're fading in groups of LEDs (for case 64 ... 67). You want the program to linger for case 64 ... 67, so you can implement a slow fade. Does that sound right?

anto168, 2023-01-30:
yes right

anto168, 2023-01-30:
in case 64 to 67, still on pause as I did there is a delay of 1000 ms.

anto168, 2023-01-30:
according to the way it works, what I'm making, the led runs from case 1 to case 63, then when case 64 lights up it fades then moves to the next case.  and back again running on case 68..

moose4lord, 2023-06-03:
This conversation seems to have run its course, so I'm closing.

closed 2023-06-03 by moose4lord

anto168, 2023-06-03:
the problem has been solved, thank you very much for helping me from the
basic code that you have made, thank you.