kitesurfer1404/WS2812FX #329: How to reverse the led order?

biccius, 2023-01-21:
Hello everyone.
This library is great! Its simplicity and its many presets save a lot of time and everything works great.
My question is simple:  Is there any way to easily reverse the order of the LEDs?

My init is something like that:

`      ws2812fx.init();
	ws2812fx.setMode(counter);
	ws2812fx.setBrightness(10);
	ws2812fx.setColor(arrayRGB[color_index].r,
					  arrayRGB[color_index].g,
					  arrayRGB[color_index].b);
	ws2812fx.setSpeed(200);
	ws2812fx.start();
`

Best regards

moose4lord, 2023-01-21:
You can add the REVERSE option at the end of the segSegment() statement.
```c++
ws2812fx.setSegment(0, 0, 9, FX_MODE_FADE, colors, 2000, REVERSE);
```
This option, and others, is covered in the User Guide.

biccius, 2023-01-23:
Thanks, with some adaptations in my code it worked.
I thought there was a method at init that would do that but setSegment worked fine.

This issue can be closed.

closed 2023-01-23 by biccius