kitesurfer1404/WS2812FX #333: Won't compile (for me) for any ESP32 board but compiles fine on other boards.

dlveach, 2023-04-22:
Won't compile for ESP32 but compiles fine for any other board I've selected, Nano BLE 33, Nano, Uno, etc.

I'd really like to use one of the ESP 32 boards with on board wifi, bt, etc.   I don't have one and I am investigating options to buy.

So I tried the "most simple form" code example from the main page and compiling in the latest IDE (2.1.0) with various ESP boards selected.  All generate this error while I get clean compiles selecting non-ESP32 boards.   I didn't try every single available board option of course but there seems to be a pattern.  (other examples from the WS2812FX lib fail same way)

I confirmed the latest ESP32 board manager (v2.0.8) is installed.  Maybe I'm missing something obvious but not getting it?

Any help is much appreciated.

The compile error is:

In file included from /Users/dveach/Library/Arduino15/packages/esp32/hardware/esp32/2.0.8/cores/esp32/Arduino.h:176,
                 from /Users/dveach/github/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.h:41,
                 from /Users/dveach/github/Arduino/libraries/WS2812FX/src/WS2812FX.h:43,
                 from /Users/dveach/github/Arduino/libraries/WS2812FX/src/WS2812FX.cpp:55:
/Users/dveach/github/Arduino/libraries/WS2812FX/src/WS2812FX.cpp: In member function 'const __FlashStringHelper* WS2812FX::getModeName(uint8_t)':
/Users/dveach/Library/Arduino15/packages/esp32/hardware/esp32/2.0.8/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
/Users/dveach/github/Arduino/libraries/WS2812FX/src/WS2812FX.cpp:404:12: note: in expansion of macro 'F'
     return F("");
            ^

exit status 1

Compilation error: exit status 1

moose4lord, 2023-04-23:
Unfortunately, a bug was introduced in v2.08 of the Arduino-ESP32 core software that creates this problem. An issue on GitHub was just opened to address this:
[https://github.com/espressif/arduino-esp32/issues/8108](https://github.com/espressif/arduino-esp32/issues/8108)

V2.0.7 of the core software works fine, so until they fix the bug, I would suggest dropping back to that version.

dlveach, 2023-04-23:
Ahh good to know.   I will do that.   Thanks!   Updated the description here to reflect the root cause.  Since not an issue in the this library I'll close this but maybe it'll help others who might bump into it.   Cheers.

closed 2023-04-23 by dlveach