kitesurfer1404/WS2812FX #340: Multiple instances

pheuts, 2023-10-07:
Is it possible to run multiple instances so i can drive multiple ws2812 strips on multiple pin's?

moose4lord, 2023-10-07:
Yes, you can. You just need to create multiple ws2812fx objects, and do the normal programming for each one. Something like this:

```c++
#include <WS2812FX.h>

#define LED_PIN      2
#define LED_PIN2     3

#define LED_COUNT   10
#define LED_COUNT2  20

WS2812FX ws2812fx  = WS2812FX(LED_COUNT,  LED_PIN,  NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx2 = WS2812FX(LED_COUNT2, LED_PIN2, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(16);
  ws2812fx.setSegment(0, 0, LED_COUNT - 1, FX_MODE_COMET, RED, 1000);
  ws2812fx.start();

  ws2812fx2.init();
  ws2812fx2.setBrightness(32);
  ws2812fx2.setSegment(0, 0, LED_COUNT2 - 1, FX_MODE_BLINK, BLUE, 1000);
  ws2812fx2.start();
}

void loop() {
  ws2812fx.service();
  ws2812fx2.service();
}
```

pheuts, 2023-10-07:
Thanks for the quick response. I'll have a crack at it.

moose4lord, 2023-12-07:
Stale issue, so closing.

closed 2023-12-07 by moose4lord