kitesurfer1404/WS2812FX #341: ESP8266 refresh rate issue

test1025, 2023-11-05:
Hello! I am not good at English, so I use a translator. If I don't understand it, I'm really sorry~!
When I was using ws2812 to flash into the demo on GitHub, although the light turned on correctly, it looked very "flickering" and felt like the refresh rate was very low? There is no difference in the appearance after adjusting the speed and replacing the light bar, but there is no problem with using NeoPixel's Rainbow. May I ask where the problem is? I didn't add any other code at this time, and all modes have this issue, and the power supply is also normal
Thank you! Wishing you success in your work
I wish you all the best

moose4lord, 2023-11-05:
What hardware are you using? It would also be helpful to cut/paste the sketch you're using.

test1025, 2023-11-07:
I used the NodeMCU ESP8266 development version for development, and I tried two ESP8266 development boards The light strips were also replaced with different ones and the same results were obtained. This is my code, only the demo code of the project will be tested
It looks like a flickering light, it looks strange, just like a low refresh rate?
Have you ever encountered such a situation? I once encountered this issue in the Arduino support library of a 51 microcontroller, and it was not resolved at the time
Maybe I should use ESP32 for testing next week..?

Wish your family and beauty, brilliant career, happiness and health!

test1025, 2023-11-07:
#include <WS2812FX.h>

#define LED_COUNT 25
#define LED_PIN 5

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(100);
  ws2812fx.setSpeed(200);
  ws2812fx.setMode(FX_MODE_RAINBOW_CYCLE);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}

test1025, 2023-11-07:
I have also tried different ws2812fx.setSpeed（）counts, there is no difference

moose4lord, 2023-11-07:
Ah, I see. Setting speed=200 is very fast for the RAINBOW_CYCLE effect with only 25 LEDs, which is causing a strobe-like effect. Try setting speed=5000 and I think you will have better luck.
```
ws2812fx.setSpeed(5000);
```

Remember increasing the speed parameter makes the LED animation run **slower**, which is probably the opposite of what you expected. A common mistake.

test1025, 2023-11-26:
我很抱歉，GitHub没有邮件提醒我回复，今天才刚刚看到。也许不是速度的问题，不过我焊接了125颗灯，似乎看起来影响不大。
谢谢您！抱歉很长时间没回复！
感谢您ws2812fx库的开发，让我便捷的开发LightCube

test1025, 2023-11-26:
I'm sorry, GitHub didn't send me an email reminder to reply. I just saw it today. Perhaps it's not a matter of speed, but I soldered 125 lights, which seems to have little impact.
Thank you! Sorry for not replying for a long time!
Thank you for developing the ws2812fx library, which has made it convenient for me to develop LightCube

closed 2023-11-26 by test1025

moose4lord, 2023-11-26:
You're building a light cube? I'd like to see it when you're done. I'm always curious to see what creative people are doing with ws2812fx.

I've built an LED cube too, although it uses the fastLED library, not ws2812fx. You can see it in action here:
[https://www.champlainsystems.com/ledcube/](https://www.champlainsystems.com/ledcube/)

test1025, 2023-12-02:
yes!Use the ESP8266 microcontroller to connect to WiFi and control the color of the lights through Xiaoai, a voice assistant from China
It was given as a birthday gift to my classmates
[image]
[image]
[image]

moose4lord, 2023-12-02:
That's very cool. Thanks for sharing.